cortex-lab/phy #1113: Error in Vertex Shader (similar to #1034)

zoebsinger, 2021-09-23:
I'm having a similar/ same issues as issue [#1034](https://github.com/cortex-lab/phy/issues/1034) . Specifically, I get multiple "Error in Vertex shader" messages when trying to open the GUI. The GUI opens but is empty. In the original issue the user stated this was due to an incompatibility between pyqt and opengl and suggested reinstalling pyqt through conda. I did this (for pyqt and pytq5) and still get the error. Any tips?

rtraghavan, 2021-10-04:
I'm having the same issue on a new macbook. Reinstalling via a new conda environment didn't help. Systematically installing each required package independently is not helping either.

zoebsinger, 2021-10-04:
What worked for me was downgrading the versions of a few packages, specifically I found I had pyqt5.15.4 and pyqtwebengine 5.15.4. I uninstalled these and re-installed versions less than 5.13, which is what an error message I got when upgrading phy told me to do. 
[image]

afergu, 2021-10-05:
@zoebsinger It worked! Thanks for finding the solution!!

jjtukker, 2021-12-06:
> What worked for me was downgrading the versions of a few packages, specifically I found I had pyqt5.15.4 and pyqtwebengine 5.15.4. I uninstalled these and re-installed versions less than 5.13, which is what an error message I got when upgrading phy told me to do. [image]

I have this exact issue with a new installation of Phy on MacOS Big Sur. Could you explain exactly how you "manually downgraded" pyqt5 and pyqtwebengine to v5.12? When I tried to do this for pyqt5, using pip, I got an error related to PyQt5_sip:

  Successfully uninstalled PyQt5-5.15.6
(spikesorting) user-adm@BR-M-3016296dzne Kilosort_2020-08-03_140330 % pip install pyqt5==5.12
Collecting pyqt5==5.12
  Using cached PyQt5-5.12-5.12.1_a-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl (40.3 MB)
ERROR: Could not find a version that satisfies the requirement PyQt5_sip<4.20,>=4.19.14 (from pyqt5) (from versions: 12.7.0, 12.7.1, 12.7.2, 12.8.0, 12.8.1, 12.9.0)
ERROR: No matching distribution found for PyQt5_sip<4.20,>=4.19.14`