cortex-lab/phy #1114: Displaying messages from Logger in --debug mode

rhayman, 2021-09-28:
I've written a plugin for phy which works but would like to add some debug information which would display when phy is called in debug mode. I've copied what's been done in other places in the code base and tried:

```python
import logging
logger = logging.getLogger(__name__)

class MyPlugin(ManualClusteringView):
   def __init__(self, features=None):
        # blah
        logger.debug("Debug msg")
```

But no messages appear when I call phy like so:

```
phy template-gui params.py --debug
```

Am I missing something obvious?

rossant, 2021-09-29:
I wonder if using:

```
logger = logging.getLogger("phy")
```

at line 2 would work?

Another possible solution might be to add a custom logging handler for your plugin.

rhayman, 2021-09-29:
Happy to confirm that adding

```python
logger = logging.getLogger("phy")
```

worked a charm 

Thanks for the quick response

closed 2021-09-29 by rhayman