cortex-lab/phy #1115: Error while running

farazmoradi, 2021-09-28:
Hi, I received this error while running Phy on some of the sorted data!

QWidget: Must construct a QApplication before a QWidget
zsh: abort      phy template-gui params.py

guitchounts, 2021-12-17:
I'm seeing a similar error after installing phy using the Developer instructions in the [Readme](https://github.com/cortex-lab/phy). Has anyone seen anything like this and solved it?

`(phy) [gg121@compute-e-16-229 phylib]$ phy template-gui  /n/groups/datta/guitchounts/data/gmou48/gmou48_2021-12-11_14-24-38_odor/tmp_MS4/phy/params.py
16:41:57.386 [W] model:1325           File /n/groups/datta/guitchounts/data/gmou48/gmou48_2021-12-11_14-24-38_odor/tmp_MS4/phy/None does not exist.
16:41:59.642 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/gg121/anaconda3/envs/phy/bin/phy", line 33, in <module>
    sys.exit(load_entry_point('phy', 'console_scripts', 'phy')())
  File "/home/gg121/anaconda3/envs/phy/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/gg121/anaconda3/envs/phy/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/gg121/anaconda3/envs/phy/lib/python3.8/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/gg121/anaconda3/envs/phy/lib/python3.8/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/gg121/anaconda3/envs/phy/lib/python3.8/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/gg121/anaconda3/envs/phy/lib/python3.8/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/gg121/code/phy/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/gg121/code/phy/phy/apps/template/gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "/home/gg121/code/phylib/phylib/io/model.py", line 1350, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "/home/gg121/code/phylib/phylib/io/model.py", line 339, in __init__
    self._load_data()
  File "/home/gg121/code/phylib/phylib/io/model.py", line 358, in _load_data
    assert self.amplitudes.shape == (ns,)
AssertionError

QWidget: Must construct a QApplication before a QWidget
Aborted`

schloegl, 2022-04-08:
I saw the same issue. In my case, it was cased by an incorrect params.py, with an entry for  n_channels_dat that did not match the data. 

Part of the problem here is, the program aborts without a meaningful error message. In order to address this, I've applied this patch 

```
diff --git a/phylib/io/model.py b/phylib/io/model.py
index ad008f5..e755bc9 100644
--- a/phylib/io/model.py
+++ b/phylib/io/model.py
@@ -380,6 +380,9 @@ class TemplateModel(object):
         self.channel_mapping = self._load_channel_map()
         self.n_channels = nc = self.channel_mapping.shape[0]
         if self.n_channels_dat:
+            if not np.all(self.channel_mapping <= self.n_channels_dat - 1):
+                print("channel miss-match, check n_channels_dat in your params file %d,%d\n",self.channel_mapping,self.n_channels_dat)
+                return
             assert np.all(self.channel_mapping <= self.n_channels_dat - 1)
 
         # Channel positions.
```

fredbarthelemy, 2023-03-14:
Hi,
I face a similar issue when I try to run phy.

Traceback (most recent call last):
  File "D:\Labo\Anaconda\envs\phy_environment\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "D:\Labo\Anaconda\envs\phy_environment\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\Labo\Anaconda\envs\phy_environment\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\phylib\io\model.py", line 1433, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\phylib\io\model.py", line 339, in __init__
    self._load_data()
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\phylib\io\model.py", line 404, in _load_data
    self.sparse_templates = self._load_templates()
  File "D:\Labo\Anaconda\envs\phy_environment\lib\site-packages\phylib\io\model.py", line 701, in _load_templates
    assert data.dtype in (np.float32, np.float64)
AssertionError

I tried the patch proposed by @schloegl but it doesn't change anything in my case.

Regards

fredbarthelemy, 2023-03-15:
Hi again,

I solved the issue. It was due to an incorrect data format using spike interface pipeline to manage the sorting.
Now I scale the data and convert from int16 to float32 and it's just fine.

Best

jessiburd, 2024-01-23:
I got the same error. When I type  phy template-gui params.py --debug, I get the following output. Does anyone know what the problem is?

14:42:33.663 [D] __init__:68          Start capturing exceptions.                                                                                                 14:42:33.740 [W] model:603            Unreferenced clusters found in templates (generally not a problem)                                                          14:42:33.779 [D] model:619            Loading spike clusters.                                                                                                     14:42:33.820 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)                                                     14:42:33.867 [E] __init__:62          An error has occurred (AssertionError):                                                                                     Traceback (most recent call last):                                                                                                                                  File "<frozen runpy>", line 198, in _run_module_as_main                                                                                                           File "<frozen runpy>", line 88, in _run_code                                                                                                                      File "C:\Users\12023\anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>                                                                          sys.exit(phycli())                                                                                                                                                         ^^^^^^^^                                                                                                                                               File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1157, in __call__                                                                   return self.main(*args, **kwargs)                                                                                                                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                               File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1078, in main                                                                       rv = self.invoke(ctx)                                                                                                                                                  ^^^^^^^^^^^^^^^^                                                                                                                                           File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1688, in invoke                                                                     return _process_result(sub_ctx.command.invoke(sub_ctx))                                                                                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                          File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1434, in invoke                                                                     return ctx.invoke(self.callback, **ctx.params)                                                                                                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                  File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 783, in invoke                                                                      return __callback(*args, **kwargs)                                                                                                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                              File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\click\decorators.py", line 33, in new_func                                                               return f(get_current_context(), *args, **kwargs)                                                                                                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                                                File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui                                                     template_gui(params_path, **kwargs)                                                                                                                             File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui                                                     model = load_model(params_path)                                                                                                                                           ^^^^ [... cut ...]     self._load_data()                                                                                                                                               File "C:\Users\12023\anaconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 383, in _load_data                                                             assert np.all(self.channel_mapping <= self.n_channels_dat - 1)                                                                                                AssertionError

LordKelv, 2024-03-28:
> Hi again,
> 
> I solved the issue. It was due to an incorrect data format using spike interface pipeline to manage the sorting. Now I scale the data and convert from int16 to float32 and it's just fine.
> 
> Best

Hey there Fred!
How did you scale the data and convert from int16 to float32?
Thanks,
Kelvin