cortex-lab/phy #1117: GLError - clusterview OK but waveforms or other views are not displayed

YuNangoChen, 2021-09-30:
Hi phy team,

I was going to do the soring with phy on my ''klustaed'' spikes but found whenever I chose the clusters, all the ''view'' windows are blank (see attached picture). Therefore I reinstalled the whole environment and the problem still occurred. I am using a computer with AMD graphic card for this.
I also checked the phy.log, and found this error looping:
[90m15:57:37.855 [D] base:702             Exception in paintGL: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glClearColor,
	cArguments = (0, 0, 0, 10)
)[0m
[90m15:57:37.855 [D] base:675             Exception in initializetGL: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glClearColor,
	cArguments = (0, 0, 0, 0)

Could you have a look and tell me what actions should be taken? Thanks!

YuNangoChen, 2021-09-30:
[image]

vincisLab, 2021-11-17:
Hello, 
we are experiecing the exact same issue with physiology and Mac with macOS Big Sur.
The cluster view and similarity view are blank. You can click on them with the mouse and "blindly" change cluster, but obviously it is an issue. Any suggestion?

jjtukker, 2022-01-04:
I have the same issue, getting either everything except ClusterView and SimilarityView, OR only those two views.. Also on a Mac, OS Big Sur 11.6 (Graphics AMD Radeon Pro 5500M 8 GB and Intel UHD Graphics 630 1536 MB). I'm actually running phy on an Ubuntu server and using x2go and XQuartz, but the fact that others can run phy from this same server makes me think this is a local issue related to my Mac. 

Has anyone made any progress solving this issue?

Here is the terminal output I get (currently only getting the ClusterView and SimilarityView):
(phy) john@alz1:~/spikesorting/Kilosort_2020-08-03_140330$ phy template-gui params.py
09:28:55.189 [W] model:591            Unreferenced clusters found in templates (generally not a problem)
09:28:55.231 [W] model:613            Unreferenced clusters found in spike_clusters (generally not a problem)
09:28:55.376 [W] model:655            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
qt.qpa.xcb: X server does not support XInput 2
qt.qpa.xcb: QXcbConnection: XCB error: 1 (BadRequest), sequence: 169, resource id: 213, major code: 130 (Unknown), minor code: 47

In phy.log I get similar output as described above, lots of "Exception in PaintGL: Attempt to retrieve context when no valid context".

jjtukker, 2022-01-06:
I still do not have a solution for this problem, but I would just like to report that updating the OS to Monterey (12.1) did NOT solve this problem. 
However, when I run phy on an Ubuntu server and use VNC viewer to connect with my Mac, it works! I am able to see all the windows. So to summarize:
- when I run phy on an Ubuntu server and connect to Mac via X2go, I see ONLY the ClusterView and SimilarityView.
- when I run phy locally on my Mac, I see all windows EXCEPT the ClusterView and SimilarityView.
- when I run phy on an Ubuntu server and connect to Mac via VNC Viewer, I see all windows and Phy works (except for an annoying inability to rescale the overall phy window below a minimum size that is still larger than my screen, but that is a separate issue I can live with for now).

I hope this helps someone figure out what is going on! Would be nice to be able to run phy locally on a Mac, particularly for those who do not have the option to use another machine.

farazmoradi, 2022-02-15:
Hi everyone, I'm facing the same issue! Have you reached any solution?
@jjtukker

JiilKim, 2022-02-17:
I am having the exact same issue on my MacBook

jlassaortiz, 2022-02-17:
I am having the same issue on my MacBook running macOS 12.2.1 (Monterey)

breadnbeer, 2022-03-17:
I have the same issue on macOS Big Sur. Any solution yet?

wanghgogo, 2022-07-25:
> I just posted a potential solution (it worked for me, MacBook Pro running MacOS 12.2.1 (Monterey)) here: #1113
> 
> I have copied the response below. Hope this helps!!
> 
> **Short Version** I upgraded to python 3.9, installed phy as usual, then manually uninstalled/reinstalled pyqt and pyqtengine as per #1122 .
> 
> **Details/Code**
> 
> ```
> # create env
> conda create -n spikeSorting python=3.9
> 
> # install phy
> pip install phy --pre --upgrade
> 
> # uninstall/reinstall relevant libraries
> pip uninstall pyqt5
> pip uninstall pyqtwebengine
> pip install PyQtWebEngine==5.15
> pip install PyQt5==5.15
> ```

I used annaconda environment for phy and this solution solved my blank clusterview/similarityview issue!

zjm199502, 2022-11-01:
> I just posted a potential solution (it worked for me, MacBook Pro running MacOS 12.2.1 (Monterey)) here: #1113
>
> I have copied the response below. Hope this helps!!
>
> **Short Version** I upgraded to [...]

I use a MacbookPro MacOS 13.0. This method make everything well!

Thank your so much for sharing this.

StringXD, 2024-01-02:
I have the problem that the windows is exactly the same as @YuNangoChen, but not the same error, does anyone have clues?
11:37:56.520 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
11:37:56.763 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
11:37:56.854 [E] model:389            Some channels are on the same position, please check the channel positions file.
11:37:56.997 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.