cortex-lab/phy #1121: Windows didn't install Phy environment

crewl2, 2021-10-19:
I'm not sure were I went wrong here but I tried to install Phy, which I've done before, and it isn't installing. Unfortunately, I closed down the command prompt so I don't have the error.

crewl2, 2021-10-19:
[image]

I've also ran into these errors. I'm not sure where I went wrong.

antimo-buonocore, 2021-11-05:
Hi, any update on this?

I tried a couple of things and if you manually create a new environment in anaconda and then install phy it works. I think in the previous version of the instructions they were explicitly specifying this step but now they removed anything related to anacoda and the command to create the new environment (you can check the history of the readme file).

So, for some reason, if you try to install phy on the base environment you would get conflicts.

Hope it makes sense (and that it's correct this way!).

crewl2, 2021-11-05:
I tried that, but when I try to install phy again it just says requirements already satisfied. Is it too late? Should I reinstall windows?

antimo-buonocore, 2021-11-06:
I don't think you need to reinstall windows. Uninstall phy with with pip uninstall, then delete the environment so that you can start from scratch. Create a new environment in anaconda, install pip, install phy. This should work.

crewl2, 2021-11-08:
Okay, so I uninstall phy. Then I create the phy environment, then do I activate environment and install pip and phy in the activated environment??

antimo-buonocore, 2021-11-08:
This worked for me. I believe you need a fresh environment (different from anaconda "base" environment ) to install phy. Before you were installing phy in anaconda "base" environment and this was not working.

crewl2, 2021-11-09:
It worked. I uninstalled phy, created a phy environment, then installed pip and phy in the activated phy environment. Installing pip was a challenge for me because python commands were not installed in the environment, but I used the anaconda navigator to install everything in the phy environment. 

Thanks so much for the help. It worked!

antimo-buonocore, 2021-11-10:
Great that it worked!

If I'm not wrong, you can install pip just with "conda install pip".

antimo-buonocore, 2021-11-11:
To be precise, when you do "conda install pip" it will also install other packages needed by pip and python is one of them. So maybe to be precise, one should first create a new environment specifying the python version and then install pip and phy.

eliezyer, 2021-12-13:
I was having the exact same problem, using xubuntu 18.04, and didn't understand why I couldn't install phy anymore. It would be good for a update on the installation instructions

closed 2024-06-07 by zm711