cortex-lab/phy #1128: Output data of phy

Acnowlogia, 2021-11-22:
Hi, 

I would like to further analysis electrophysiological data after manually inspect by phy. Could you please tell me how can I output channel with largest amplitude and their trace of each cluster?

cafalchio, 2022-03-16:
I am also trying to use the data after passing througly Phy, would be nice to have it on the Docs.

cafalchio, 2022-04-12:
Hi @Acnowlogia , did you find a way? I am trying to input it back in python to plot waveforms.

Acnowlogia, 2022-04-18:
> Hi @Acnowlogia , did you find a way? I am trying to input it back in python to plot waveforms.

I have choose other methods to process data. 

If you want to plot waveforms. You may use readNPY in matlab (or directly use python) to read 'spike_templates.npy' which is generated by kilosort. Then you can find waveform of templates. Combining with 'spike_times.npy', 'spike_clusters.npy' and other .TSV file generated by Phy, you can draw the waveforms.