cortex-lab/phy #1133: Strange Splitting

cpiette95, 2022-01-30:
I have been using Kilosort3 on Neuropixel recordings, and downloaded the last version of phy. I need to do the phy extract-waveforms command to be able to look at the FeatureView. 

On two different datasets, I have the same issue (with also another issue: I don't see anything on the "AmplitudeView" plot):
 
When I try to clean a cluster to get a better refractory period for example, once I have selected the "core" of the cluster, I am left only with the minority of the initial number of spikes (for example 485 against 5000 initially). 
I did some tests, and in fact, when I select all but one displayed points in the Feature Fiew, I always end up with a new cluster containing 499 spikes. And the last point (also a single trace on WaveformView - further away from the core of the cluster - contains all the remaining spiking events... 

Yet, if I then re-iterate the phy extract-waveforms command on the new dataset (with the two new clusters), I can see more than just a single trace, and new points on the FeatureView, but the same problem re-emerge when I want to split again the cluster containing a high number of spikes. 

Do you have any idea of what I should change to clean properly and easily these datasets? 
Thank you!

DaniMagyar, 2022-04-15:
In AmplitudeView menu click on "Next amplitude type", then select a new cluster in ClusterView to update. This may solve the AmplitudeView problem. 

When you clean a cluster in FeatureView, try to select those points that you want to remove instead of the "core" of the cluster. Or, you can try to draw something like this: 

[image]

Hope this helps.

o-zhu, 2022-06-09:
I'm getting the same issue as cpiette95, on many datasets. I've only encountered this problem with KS3; previously I was using KS2 and had minimal issues. At its most extreme, I circle one point in FeatureView (which corresponds to one spike) from a cluster, press k to split, and that one point would produce a cluster that has 10,000 spikes, and all the other points on the scatter plot would be labeled as another cluster that has one spike. The mean waveform for the template would also look like noise for the cluster with 10,000 spikes, and would look like the original cluster for the neuron with one spike. 

It seems like the color and labels of the clusters have been switched. Any ideas on how to fix this?