cortex-lab/phy #1136: good units grayed-out while mua units appear green

xy103, 2022-02-08:
[image]
[image]

Hi I'm seeing a strange issue in the clusters. Some good units appear gray and if I don't manually click on them they get skipped when I advance to the next unit. At the same time mua units look green for some units. Once I change their labels the correct colors are displayed (green-good, gray-mua). Has anyone seen this problem? Is this a bug in Phy or perhaps there's some other issue? Thanks for any help in advance!

rossant, 2022-02-09:
What is the name of the column with "good" and "mua"? It might be the case that this column shows a classification of the clusters that would be independent from the native one in phy. The color would then be based on phy's classification, while this column may show something else (computed by the spike sorting program?)

xy103, 2022-02-09:
The column is "KSLabel." I ran spike sorting with kilosort 2.5 inside ecephys pipeline. Could it be ecephys changes the label based on some other quality metrics it computed?

xy103, 2022-02-09:
I figured out what happened. Basically ecephys has a module to classify noise templates from signal, and then writes a file called cluster_group, designating units as good or not. When opened in phy, the ecephys cluster_group dictates the color, while KSLabel is displayed as a column in text.