cortex-lab/phy #1138: Conda install and Error: No such command 'template-gui'.

mauricev, 2022-02-22:
I’m trying to install with conda using environment.yml. At the very least, sklearn should be scikit-learn and traitlets is missing. However, I still get the error 

```Error: No such command 'template-gui'

upon running phy template-gui params.py

So I assume something is still missing.

grahamfindlay, 2022-03-04:
`pip` needs to install the pre-release version phy, but `environment.yml` as it currently exists will only grab the latest (non-pre) release. I just spent 15 minutes trying to figure out how to pass the pip `--pre` option in `environment.yml` but nothing I've tried seems to work. So for now, you can do the following. 

1. Create your conda environment using the following environment.yml:
```
name: phy2
dependencies:
  - python=3.7
  - pip
  - numpy
  - matplotlib
  - scipy
  - h5py
  - pyqt
  - dask
  - cython
  - pillow
  - scikit-learn
  - traitlets
```
`> conda env create -f environment.yml`

2. Install `phy` using `pip` from the command line:
```
> conda activate phy2
> pip install phy --pre --upgrade
```

This fixed the issue for me.

grahamfindlay, 2022-03-08:
By the way -- we have had issues with the installation of certain QT Framework plugins when using either `pip` or the default conda channel. I recommend using conda-forge, and have edited the example environment.yml above to reflect that.