cortex-lab/phy #1143: Blank ClusterView and SimilarityView

AntonioST, 2022-03-11:
Hi,

I got some problem to start with a template-gui on my kilosort result. The ClusterView and SimilarityView didn't show anything. No mater how long I wait, it didn't change the blank content. If I tried to close the window, a timeout dialog pop up.

[image]

Here are the part of packages installed in my conda environment.

```
# Name                    Version                   Build  Channel
pyqt5                     5.15.5                   pypi_0    pypi
pyqt5-qt                  5.15.2                   pypi_0    pypi
pyqt5-qt5                 5.15.2                   pypi_0    pypi
pyqt5-sip                 12.9.1                   pypi_0    pypi
pyqtwebengine             5.15.5                   pypi_0    pypi
pyqtwebengine-qt          5.15.2                   pypi_0    pypi
pyqtwebengine-qt5         5.15.2                   pypi_0    pypi
pytest-qt                 3.2.2                    pypi_0    pypi
qtconsole                 4.5.5                    pypi_0    pypi
phy                       2.0a1                    pypi_0    pypi
phylib                    2.4.1                    pypi_0    pypi
```

I guess it is something relate to the QtWebEngine which js wasn't run.

Thanks

AliyaAbl, 2022-03-18:
I have the same problem. Have you found a solution to this?

AntonioST, 2022-03-18:
Hi @AliyaAbl 

I don't the real reason why qtwebengine is unable to show html content, but this command works on my linux compute 
`env QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui .../params.py`.

[image]

 I just find out this way this morning. I tried a simple QWebEngineView example with conda Qt (pyqtwebengine-qt 5.15.2 from conda pypi chanel), and my OS Qt (python-pyqt5-webengine-5.15.5-4 from archlinux). I have no such issue with my OS Qt.

abhijithkrishnan93, 2022-03-21:
I have the same problem and this  @AntonioST solution works.

arthurz323, 2022-05-23:
had the same issue, and solution from @AntonioST works for me too, thanks!

alejoe91, 2022-09-04:
> Hi @AliyaAbl
> 
> I don't the real reason why qtwebengine is unable to show html content, but this command works on my linux computer `env QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui .../params.py`.
> 
> [image]
> 
> I just find out this way this morning. I tried a simple QWebEngineView example with conda Qt (pyqtwebengine-qt 5.15.2 from conda pypi chanel), and my OS Qt (python-pyqt5-webengine-5.15.5-4 from archlinux). I have no such issue with my OS Qt.

@rossant could you add this to the Troubleshoot section?

KBillel, 2022-10-24:
Worked for me as well thank you !

Datamance, 2023-04-03:
@AntonioST You are a lifesaver! Thank you.

closed 2023-06-15 by rossant

Coglet00, 2023-12-15:
hey i am a bit inexperienced but where should i write the env QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui .../params.py. command exactly?

rory-bedford, 2023-12-15:
Hey @Coglet00, I'm guessing you're on linux?

In bash, you can define an environment variable for the execution of a specific command by putting it before the command. So in this case, we're setting QTWEBENGINE_CHROMIUM_FLAGS to "--single-process", then just running the usual command 'phy template-gui params.py'. The env bit at the start shouldn't matter much (it just runs the command in a clean environment). So in your terminal, navigate to the phy data directory, make sure it has params.py in your current directory, then run the command (don't actually put the .../ at the start).

rory-bedford, 2023-12-18:
Just to clarify is that on the same line? So your full command should be:

```
QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui params.py
```

If you run them on different lines the environment variable QTWEBENGINE_CHROMIUM_FLAGS won't get saved for when you run the next command.

Sorry if this is pedantic just checking though!

PR #1167 (2023-06-15): Set cluster views from URL of local file
This PR addresses and may resolve #1143, resolve #1151, resolve #1156, resolve #1159, and resolve #1165.

As @FranciscoNaveros and others have [observed](https://github.com/cortex-lab/phy/issues/1159#issue-1239068482), the issue of blank cluster views occurs when the number of clusters or columns exceeds a certain limit. This seems to be due to the method `setHtml` of `QWebEnginePage` having a [size limit of 2 MB](https://doc.qt.io/qtforpython-5/PySide2/QtWebEngineWidgets/QWebEnginePage.html?highlight=qwebenginepage#PySide2.QtWebEngineWidgets.PySide2.Qt [... cut]