cortex-lab/phy #1149: Finding Spike Channel IDs

argunsah, 2022-04-08:
Hello,
I sorted the data using SpykingCircus and exported to phy. When I run phy I can see channel ids for each cluster but I cannot find this data in the export folder. How can I access this information? (Ideally in Matlab).
Best,
Ali

GrimmSnark, 2022-04-08:
You need to press the save icon in phy then look in the cluster_info.tsv file. One of the headings is the best channel for the cluster. 

Best,

Michael

argunsah, 2022-04-08:
Thank you!
Related: is it possible to do it automatically without running phy and pressing the save icon? (Lets assume all clusters were correct)
Best,
Ali

GrimmSnark, 2022-04-08:
I do not know for sure but if you are comfortable writing a plugin this may do the trick

https://phy.readthedocs.io/en/latest/plugins/#saving-cluster-metadata-in-a-tsv-file

Best,

Michael

argunsah, 2022-04-08:
Thanks again. I will try.