cortex-lab/phy #1151: cluster view and similarity view not shown

sbyoo, 2022-04-10:
Hello

I am trying to use phy in the Mac mini with Big Sur as OS. I installed identical dependencies to the current Mac in another Mac, in which phy runs perfectly. However, in this Mac, all other function works except the cluster view and similarity view (screenshot attached). Your help will be greatly appreciated. Thank you.

Best,
[image]

Mousakr, 2022-04-20:
Hi,

I unfortunately have the same issue. I use fedora linux version 35, with a newly installed python environment (same issue arises for python 3.7 , 3.8 and 3.10) and newly installed phy. When I attempt to close the GUI, I get the following error (I am unsure if it is related...):

17:53:42.598 [E] qt:178               Timeout in _block().

kissadamfkut, 2022-05-09:
Same issue here. It's been happening since the last OS upgrade (Manjaro). The following packages have been upgraded when it turned broken:

[packages.txt](https://github.com/cortex-lab/phy/files/8651164/packages.txt)

arthurz323, 2022-05-20:
same issue! running on os fedora 36, kernel 5.17.8.

brandonStell, 2022-06-05:
This response https://github.com/cortex-lab/phy/issues/1143#issuecomment-1072420729 solved the problem for me on Ubuntu 22.04.

PR #1167 (2023-06-15): Set cluster views from URL of local file
This PR addresses and may resolve #1143, resolve #1151, resolve #1156, resolve #1159, and resolve #1165.

As @FranciscoNaveros and others have [observed](https://github.com/cortex-lab/phy/issues/1159#issue-1239068482), the issue of blank cluster views occurs when the number of clusters or columns exceeds a certain limit. This seems to be due to the method `setHtml` of `QWebEnginePage` having a [size limit of 2 MB](https://doc.qt.io/qtforpython-5/PySide2/QtWebEngineWidgets/QWebEnginePage.html?highlight=qwebenginepage#PySide2.QtWebEngineWidgets.PySide2.Qt [... cut]
closed 2023-06-15 by rossant