cortex-lab/phy #1155: Clusters in Waveform View Unaligned

pcpollak, 2022-04-21:
I identified all of the good clusters in a data set, partially by using the WaveformView to to identify clusters with clean consistent waveforms. However, just as I started merging clusters, all of clusters in WaveformView spontaneously unaligned themselves.  Instead of seeing checkmark shaped curves of spikes, I see disordered lines with individual peaks not allied to each other. When I press 'M' to see the average waveform, they still have normal waveforms shaped like they were before. The correlagrams and other window views still indicate that the clusters are good. 

I do not know why Phy spontaneously started displaying waveforms like this. Right before I was having a similar problem but only when I merged two clusters. I overlaid two clustered in WaveformView and they looked similar. But when I merged them none of their peaks were aligned and the waveform looked terrible. Even though the correlagrams continued to show a good cluster. Pressing 'M' to see the average merged waveform shows a noisy poorly shaped waveform. 

Does anyone know why the WaveformView is not alining the spikes within clusters?

himahuja, 2024-03-27:
I have the same issue, and don't know why this would happen
[image]

chiyu1203, 2024-04-23:
I have bumped into similiar issue (although two clusters were detected by Kilosort4, the spikes of both clusters has similar waveform, and are from the same channel. They are slightly misaligned by time so when merging them together, the new waveform looks terrible. Anyone has a idea if this is a feature of the extracellular recording or a bug in kilosorts?