cortex-lab/phy #1156: ClusterView problem

churanj, 2022-05-11:
Hello,
we have a similar problem like issue #1151 but with some interesting differences. We use a neuropixels probe and kilosort3. When we run phy on the output of kilosort , the ClusterView window remains blank, however, the SimilarityView window seems intact (see screenshot). Importantly, this issue only occurs in a part (~50%) of our recordings while other work nicely. When it occurs, it is independent of the machine we run phy at - currently we tested it at two Windows11 and two MacOS systems. When I compare the Kilosort outputs between the ‘working’ and ’not working’ measurements I also don’t see any obvious major differences. So far the mystery… Any suggestions or speculations?
Best!
Jan 
[image]

churanj, 2022-05-13:
An update on this issue: Apparently the cause of the problem is a file named 'metrics.csv' that is generated in the post-processing stages on the neuropixels processing pipeline. When we delete this file from the kilosort results directory, everything seems to work work fine. The content of the file looks like a table of the clusters and their properties. We are, however, not sure, what this file was supposed to do for phy and whether it is save to delete it. It would be great to get some insight into the functions of that file.
Best!
Jan

arthurz323, 2022-05-20:
Hello,

I have encountered the same issue. However, i do not have such `metrics.cvs` file in my output folder. Opened two sorted sessions, and they had the same problem.
Here is the screenshot of the error message:
[image]

Best,
Arthur

PR #1167 (2023-06-15): Set cluster views from URL of local file
This PR addresses and may resolve #1143, resolve #1151, resolve #1156, resolve #1159, and resolve #1165.

As @FranciscoNaveros and others have [observed](https://github.com/cortex-lab/phy/issues/1159#issue-1239068482), the issue of blank cluster views occurs when the number of clusters or columns exceeds a certain limit. This seems to be due to the method `setHtml` of `QWebEnginePage` having a [size limit of 2 MB](https://doc.qt.io/qtforpython-5/PySide2/QtWebEngineWidgets/QWebEnginePage.html?highlight=qwebenginepage#PySide2.QtWebEngineWidgets.PySide2.Qt [... cut]
closed 2023-06-15 by rossant