cortex-lab/phy #1166: FeatureView and TemplateFeatureView not working

lukebemish, 2022-07-14:
When opening Phy's template-gui with some data exported from Kilosort, the FeatureView and TemplateFeatureView windows cannot be opened; the console log prints `Could not create view FeatureView.` on startup and each time I try opening the view. I have included the log and a zip of data to reproduce.
[phy.log](https://github.com/cortex-lab/phy/files/9115533/phy.log)
[data.zip](https://github.com/cortex-lab/phy/files/9115537/data.zip)

manfredjoswald, 2022-08-03:
Hi, I have exactly the same problem and also get the 'Could not create view FeatureView' error message. The text log obtained from running the --debug command is attached. Should I reinstall Phy?

[Anaconda log.txt](https://github.com/cortex-lab/phy/files/9253340/Anaconda.log.txt)

GrimmSnark, 2022-08-31:
This is a known issue with Kilosort 3. See this for a somewhat workaround: https://github.com/MouseLand/Kilosort/issues/429

AHSAN-PA, 2024-05-07:
We can fix FeatureView. Please refer #1267 and #1145.