cortex-lab/phy #1168: ImportError: libQt5QmlModels.so.5: cannot open shared object file: No such file or directory

mfvd, 2022-07-25:
Hello,

I'm trying to open a dataset in phy and keep getting the error below. I'm running Ubuntu 20.04.4 LTS. I've tried reinstalling ```PyQtWebEngine```but it did not solve the problem. Any suggestion on how to solve this issue?

```python
Traceback (most recent call last):
  File "/home/neuroimaging/anaconda2/envs/si_openephys/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/home/neuroimaging/anaconda2/envs/si_openephys/lib/python3.9/site-packages/phy/gui/qt.py", line 41, in <module>
    from PyQt5.QtWebEngineWidgets import (QWebEngineView,  # noqa
ImportError: libQt5QmlModels.so.5: cannot open shared object file: No such file or directory
```

caitlinmallory, 2022-09-02:
having the same issue!

pderdeyn, 2023-02-14:
I got errors like this while trying to get an installation working on python 3.11. After quite a rabbit hole of trying to install and configure PyQt5 manually, I simply created a new python 3.7 environment, installed phy following the documentation, and my installation now works fine. I'm on Ubuntu 18.04 btw

rachaita20, 2023-10-13:
I am facing the same issue.