cortex-lab/phy #1170: Problem about the GUI loading

zjm199502, 2022-09-04:
Hi, i got a problem in launching the Phy GUI.  

While I follow the installation steps, after typing _phy template-gui params.py_. An error occurred as the following picture shown. Does anyone have idea on solving this issue?

Thank you very much!
[image]

rossant, 2022-09-04:
In your `params.py` file, you should put double quotes `"` around the path to the dat file

zjm199502, 2022-09-06:
Hi, Thanks for your suggestion and it do solve my previous problem.  

But another error occurred now. Following codes shown in the terminal  window:  

14:29:57.009 [W] model:655            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
Error in Vertex shader 776 (<string>)
 -> '' :  version '120' is not supported

 and then the PhyGUI appeared without any waveform displays as the following picture shows.
[image]

Does this phenomenon normal ?

Thank you !