cortex-lab/phy #1174: Error in Vertex shader 939

kla311, 2022-10-02:
Hi, 

When I run 'Phy template-gui params.py' in the anaconda terminal I get a long error beginning with:

Error in Vertex shader 939 (<string>)
 -> '' :  version '120' is not supported

The GUI pops up, but I can't see any of the clusters or waveforms.

tiburona, 2022-10-24:
I have the same issue, though I am running from a virtualenv environment rather than an anaconda terminal, and I'm running with the command `arch -x86_64 phy template-gui params.py` (I also installed phy with the `arch -x86_64` prefix) because that was the only way I could get the pyqt web engine to install on my M1 Mac.  @kla311 are you by any chance also on a Mac with an M1 chip?

kla311, 2022-10-24:
Hi @tiburona- No, my Mac does not have an M1 chip (I have an intel one). BUT- my issue is now fixed! I ran by the problem with a friend who is more familiar with python than I am, and we figured out that my version of pyqt was not the right one. So we uninstalled & reinstalled a a few packages with older versions than what I had. In the terminal, we did: 

conda uninstall pyqt
python
conda deactivate
conda remove --name phy2 --all
conda env list
conda create -n phy2 python=3.9
pip install phy --pre --upgrade
conda activate phy2
pip install phy --pre --upgrade
pip uninstall pyqt5
pip uninstall pyqtwebengine
pip install PyQtWebEngine==5.15
pip install PyQt5==5.15

Then it worked! I could see my phys data clusters and there were no errors in the terminal.

closed 2022-10-24 by kla311

tiburona, 2022-10-24:
Thanks @kla311 -- I had to use a slightly different series of commands because I'm using virtualenv, not conda, and I need to use the `arch -x86_64` prefix to activate Rosetta, but setting the version equal to 5.15 solved my issue too.