cortex-lab/phy #1180: Question in the 'spike sorting comparison' website : is phy a sorter or a GUI?

ZoeChen96, 2022-11-08:
Hi there,

Although I am not sure whether this is the right place for asking, I would like to post the question here. Recently I go into the website http://phy.cortexlab.net/data/sortingComparison/ and see a comparison among neuralpixel's dataset, with sorters such as JRclust, Kilosort, spykingcircus etc. I understand the above ones are 'sorters', but 'phy', as you give the introduction here, is a GUI for processing spike sorting. Does it have a sorter behind for the accuracy comparison? And what is the sorter behind (if exist)? 

[image]

zm711, 2023-02-28:
@ZoeChen96 
I'm not directly affiliated with this project, but that info is from 2016 and at that stage I believe one thought was that Phy would incorporate automatic and manual sorting (https://www.biorxiv.org/content/biorxiv/early/2015/02/16/015198.full.pdf) with a plug into klusta (https://github.com/kwikteam/klusta). But this was dropped and instead it became a final manual curation gui for use post kilosort (1, 2, 2.5, or 3) or spykingCircus. It works well for manual curation, but as far as I know it no longer does any of its own sorting. JRClust is also no longer actively supported.

ZoeChen96, 2023-03-03:
Hey Zach,
Thanks for the kind reply! Have a nice weekend!

closed 2023-03-03 by ZoeChen96