cortex-lab/phy #1184: chn 0 giving me no feature and amplitude value

fredchen00, 2022-11-27:
I was trying to write some plug-in for phy but I noticed the feature value and the amplitude for channel 0 is always 0. It even shows up as zero in the feature and amplitude view. I am wondering if this is a code bug or am I doing something wrong. I performed my spike sorting through Spyking Circus.  Thanks in advance
[image]

fredchen00, 2022-11-27:
Just want to make another comment, 0chn has feature values when I open its npy file. So somehow the value is lost in phy I think.

fredchen00, 2022-11-28:
Just another note, I was able to trace the data flow. Somehow the feature value was set to zero after         
features = from_sparse(features, cols, channel_ids)
in line 1042 inside the get_features function in phylib/io/model.py

alirza67, 2024-04-11:
> Just another note, I was able to trace the data flow. Somehow the feature value was set to zero after features = from_sparse(features, cols, channel_ids) in line 1042 inside the get_features function in phylib/io/model.py

Hi, is there any now update on this? did you manage to solve the problem?