cortex-lab/phy #1189: Error in Vertex shader:  version '120' is not supported

hemmingswu, 2022-12-31:
Hi,

Trying to install phy on my late 2013 macbook pro, and I got errors like "Error in Vertex shader 399 (<string>) -> '' :  version '120' is not supported"
phy opens, I saw numbers showing up in clusterview and similarityview, but no graphical data. I'm guessing it's opengl/glsl version issue.

Anyone had a similar issue? Any suggestion is highly appreciated.

MarkOrloff, 2023-07-03:
I'm having this issue as well... I'm running osx 13.4.1 on a 202 macbook

thejuc, 2023-07-04:
I'm also having this issue on 2019 macbook with osx 13.0

skim0119, 2023-08-22:
I got the same issue on my MacBook pro. I resolved it by downgrading `PyQt5`:
```sh
pip install PyQt5==5.15
pip install PyQtWebEngine==5.15
```