cortex-lab/phy #1190: All windows blank in Phy

elladulko, 2023-01-18:
After upgrading to Ubuntu 22 I can open Phy but all windows are black. 

(base) ella@fishfish:~$ cd /media/probeX/intanData/ela/analyzedData/ela_0023
(base) ella@fishfish:/media/probeX/intanData/ela/analyzedData/ela_0023$ conda activate phy
(phy) ella@fishfish:/media/probeX/intanData/ela/analyzedData/ela_0023$ phy template-gui params.py --debug
15:54:14.605 [D] __init__:68          Start capturing exceptions.
15:54:15.479 [W] model:592            Unreferenced clusters found in templates (generally not a problem)
15:54:15.881 [D] model:607            Loading spike clusters.
15:54:16.368 [W] model:614            Unreferenced clusters found in spike_clusters (generally not a problem)
15:54:16.790 [D] model:557            No channel shank file found.
15:54:16.792 [D] model:680            Loading templates.
15:54:16.971 [D] model:712            Templates are dense.
15:54:16.971 [W] model:656            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
15:54:16.971 [D] model:718            Loading the whitening matrix.
15:54:16.974 [D] model:725            Loading the inverse of the whitening matrix.
15:54:16.992 [D] model:754            Loading features.
15:54:16.994 [D] model:769            Features are sparse.
15:54:16.994 [D] model:791            Loading template features.
15:54:16.996 [D] model:801            Template features are sparse.
15:54:16.997 [D] model:492            Load `cluster_Amplitude.tsv`.
15:54:17.005 [D] model:492            Load `cluster_KSLabel.tsv`.
15:54:17.014 [D] model:492            Load `cluster_group.tsv`.
15:54:17.023 [D] model:492            Load `cluster_ContamPct.tsv`.
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
15:54:17.833 [D] context:100          Initialize joblib cache dir at `/media/probeX/intanData/ela/analyzedData/ela_0023/.phy`.
15:54:17.839 [D] context:101          Reducing the size of the cache if needed.
15:54:18.758 [D] base:102             Add filter `high_pass`.
15:54:18.758 [D] config:31            Load config file `/home/ella/.phy/phy_config.py`.
15:54:18.759 [D] plugin:145           Loading 0 plugins.
15:54:18.759 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
15:54:18.760 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
15:54:18.760 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
15:54:18.809 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
15:54:18.809 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
15:54:18.836 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
15:54:18.838 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
15:54:18.839 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
15:54:18.840 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
15:54:18.840 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
15:54:18.843 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
15:54:18.847 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
15:54:18.847 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
15:54:18.848 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
15:54:19.196 [D] context:185          Save data to `/media/probeX/intanData/ela/analyzedData/ela_0023/.phy/spikes_per_cluster.pkl`.
15:54:19.776 [D] gui:463              Creating GUI.
15:54:19.784 [D] state:46             Load /home/ella/.phy/TemplateGUI/state.json for GUIState.
15:54:19.785 [D] state:46             Load /media/probeX/intanData/ela/analyzedData/ela_0023/.phy/state.json for GUIState.
15:54:20.150 [D] gui:718              Add view ClusterView to GUI.
15:54:20.178 [D] gui:718              Add view SimilarityView to GUI.
15:54:20.236 [D] gui:718              Add view WaveformView to GUI.
15:54:20.244 [D] base:337             Set state for WaveformView.
15:54:20.264 [D] gui:718              Add view CorrelogramView to GUI.
15:54:20.266 [D] base:337             Set state for CorrelogramView.
15:54:20.304 [D] gui:718              Add view ISIView to GUI.
15:54:20.307 [D] base:337             Set state for ISIView.
15:54:20.342 [D] gui:718              Add view AmplitudeView to GUI.
15:54:20.345 [D] base:337             Set state for AmplitudeView.
15:54:20.381 [D] gui:718              Add view FiringRateView to GUI.
15:54:20.384 [D] base:337             Set state for FiringRateView.
15:54:21.232 [D] gui:718              Add view TraceView to GUI.
15:54:21.234 [D] base:337             Set state for TraceView.
15:54:21.235 [D] base:494             Set color scheme to random.
15:54:24.198 [D] base:1623            Adding default color schemes to TraceView.
15:54:27.957 [D] gui:718              Add view ProbeView to GUI.
15:54:27.961 [D] base:337             Set state for ProbeView.
15:54:27.987 [D] gui:718              Add view TemplateFeatureView to GUI.
15:54:27.989 [D] base:337             Set state for TemplateFeatureView.
15:54:28.022 [D] gui:718              Add view ClusterScatterView to GUI.
15:54:28.024 [D] base:337             Set state for ClusterScatterView.
15:54:28.024 [D] base:494             Set color scheme to random.
[... 35 lines cut ...]
15:54:28.896 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:28.897 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:28.898 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:28.997 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:28.999 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:29.000 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:29.000 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:29.001 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:29.002 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:29.003 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.882 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.885 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.886 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.887 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.888 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.889 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.890 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.891 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.892 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.893 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.894 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.895 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.896 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.898 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context
15:54:34.899 [D] base:702             Exception in paintGL: Attempt to retrieve context when no valid context

hyr2, 2023-03-14:
Same issue on linux

Using pyqt version 5.13.1 but tried 5.12.1 as well. Python version 3.7.0

MarkOrloff, 2023-07-03:
I believe I'm having a similar issue on mac 13.4.1, python 3.11.3. I am getting the following errors (excerpt below, but it just continues on with paintGL initializetGL errors):

phy template-gui params.py --debug
22:32:02.120 [D] __init__:68          Start capturing exceptions.
22:32:02.128 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
22:32:02.129 [D] model:619            Loading spike clusters.
22:32:02.131 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
22:32:02.136 [D] model:692            Loading templates.
22:32:02.162 [D] model:720            Templates are sparse.
22:32:02.162 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
22:32:02.162 [D] model:730            Loading the whitening matrix.
22:32:02.164 [D] model:737            Loading the inverse of the whitening matrix.
22:32:02.192 [D] model:766            Loading features.
22:32:02.266 [D] model:781            Features are sparse.
22:32:02.267 [D] model:803            Loading template features.
22:32:02.289 [D] model:813            Template features are sparse.
22:32:02.291 [D] model:504            Load `cluster_group.tsv`.
22:32:02.929 [D] context:100          Initialize joblib cache dir at `/Users/maorloff/Downloads/phy-data-master/template/.phy`.
22:32:02.930 [D] context:101          Reducing the size of the cache if needed.
22:32:02.959 [D] base:102             Add filter `high_pass`.
22:32:02.959 [D] config:31            Load config file `/Users/maorloff/.phy/phy_config.py`.
22:32:02.961 [D] plugin:145           Loading 0 plugins.
22:32:02.961 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
22:32:02.961 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
22:32:02.962 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
22:32:02.962 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
22:32:02.963 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
22:32:02.963 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
22:32:02.964 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
22:32:02.965 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
22:32:02.965 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
22:32:02.966 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
22:32:02.966 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
22:32:02.967 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
22:32:02.968 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
22:32:02.968 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
22:32:02.969 [D] context:209          The file `/Users/maorloff/Downloads/phy-data-master/template/.phy/new_cluster_id.pkl` doesn't exist.
22:32:02.975 [D] context:185          Save data to `/Users/maorloff/Downloads/phy-data-master/template/.phy/spikes_per_cluster.pkl`.
22:32:02.986 [D] gui:463              Creating GUI.
22:32:03.058 [D] state:46             Load /Users/maorloff/.phy/TemplateGUI/state.json for GUIState.
22:32:03.059 [D] state:46             Load /Users/maorloff/Downloads/phy-data-master/template/.phy/state.json for GUIState.
22:32:03.315 [D] gui:718              Add view ClusterView to GUI.
22:32:03.379 [D] gui:718              Add view SimilarityView to GUI.
22:32:03.445 [D] gui:718              Add view WaveformView to GUI.
22:32:03.506 [D] base:337             Set state for WaveformView.
22:32:03.522 [D] gui:718              Add view CorrelogramView to GUI.
22:32:03.525 [D] base:337             Set state for CorrelogramView.
22:32:03.554 [D] gui:718              Add view ISIView to GUI.
22:32:03.559 [D] base:337             Set state for ISIView.
22:32:03.579 [D] gui:718              Add view FeatureView to GUI.
22:32:03.582 [D] base:337             Set state for FeatureView.
22:32:03.612 [D] gui:718              Add view AmplitudeView to GUI.
22:32:03.616 [D] base:337             Set state for AmplitudeView.
22:32:03.644 [D] gui:718              Add view FiringRateView to GUI.
22:32:03.647 [D] base:337             Set state for FiringRateView.
22:32:03.741 [D] gui:718              Add view TraceView to GUI.
22:32:03.744 [D] base:337             Set state for TraceView.
22:32:03.745 [D] base:494             Set color scheme to random.
22:32:03.769 [D] base:1626            Adding default color schemes to TraceView.
22:32:03.808 [D] gui:718              Add view ProbeView to GUI.
22:32:03.811 [D] base:337             Set state for ProbeView.
22:32:03.837 [D] gui:718              Add view ClusterScatterView to GUI.
22:32:03.840 [D] base:337             Set state for ClusterScatterView.
22:32:03.840 [D] base:494             Set color scheme to random.
22:32:03.841 [D] base:1626            Adding default color schemes to ClusterScatterView.
22:32:04.007 [D] gui:519              Load the geometry state.
22:32:04.244 [D] base:675             Exception in initializetGL: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glEnable,
[... 41 lines cut ...]
[... cut ...]
 ...

Error in Vertex shader 716 (<string>)
 -> '' :  #version required and missing.

 000 #define _GLUMPY__VERTEX_SHADER__
 001 // --- Snippets code : start --- //
 002 // --- Snippets code : end --- //
 003 varying vec2 v_pos_tmp;
 ...

Error in Vertex shader 716 (<string>)
 -> 'varying' : syntax error: syntax error

 ...
 001 // --- Snippets code : start --- //
 002 // --- Snippets code : end --- //
 003 varying vec2 v_pos_tmp;
 004 
 005 uniform vec2 u_window_size;
 006 uniform vec2 u_pan;
 ...

chris-langfield, 2023-08-29:
Identical issue on Mac 13.0, Apple M2. 

In `phy.log` many repetitions of:

```
12:12:59.211 [D] base:675             Exception in initializetGL: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glEnable,
	cArguments = (GL_POINT_SPRITE,)
)
12:12:59.211 [D] base:675             Exception in initializetGL: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glEnable,
	cArguments = (GL_POINT_SPRITE,)
)

```

And in stoud while running the program:

```
Error in Vertex shader 210 (<string>)
 -> 'varying' : syntax error: syntax error

 ...
 001 // --- Snippets code : start --- //
 002 // --- Snippets code : end --- //
 003 varying vec2 v_pos_tmp;
 004 
 005 uniform vec2 u_window_size;
 ...

Error in Vertex shader 91 (<string>)
 -> '' :  version '120' is not supported

 000 #define _GLUMPY__VERTEX_SHADER__
 001 // --- Snippets code : start --- //
 002 // --- Snippets code : end --- //

```