cortex-lab/phy #1193: Attribute error on np.bool

Datamance, 2023-02-07:
Hi there,

I'm trying to use Phy in a virtual environment containing Python 3.11.1 and numpy 1.24.2. I get this error when I try to run the standard `phy template-gui params.py`

```
╰─(env) ○ phy template-gui params.py               
...
  File "/home/rico/.asdf/installs/python/3.11.1/lib/python3.11/site-packages/phy/plot/gloo/variable.py", line 86, in <module>
    gl.GL_BOOL: (1, gl.GL_BOOL, np.bool),
                                ^^^^^^^
  File "/home/rico/.asdf/installs/python/3.11.1/lib/python3.11/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'bool'.

```

What is the maximum numpy version to which I must downgrade in order to get this working? Thanks.

Datamance, 2023-02-07:
Because of how the entrypoint generator is written, 

```python
  18   │ def importlib_load_entry_point(spec, group, name):
  19   │     dist_name, _, _ = spec.partition('==')
  20   │     matches = (
  21   │         entry_point
  22   │         for entry_point in distribution(dist_name).entry_points
  23   │         if entry_point.group == group and entry_point.name == name
  24   │     )
  25   │     return next(matches).load()
```

This is going to break every time. I'm not sure how this function is generated but perhaps you can tell setup.py to actually use the version information rather than throwing it away...

Datamance, 2023-02-07:
I'm not sure what's going on with how you load modules, but issues seem to arise with the use of virtual environments.

Even if I do `pip install phy --pre` without `--upgrade`, ensuring that I can pin numpy to a version that contains `np.bool` (I chose 1.22.4 to roughly match the date of the last commit in this repo), I still get this numpy compilation error:
```
RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xf
```
upon running either `phy template-gui params.py` via the CLI or running it from a python script via
```python
from phy.apps.template import template_gui
template_gui("params.py")
```
To me, this suggests that `phy` is somehow trying to circumvent the `numpy` in the virtualenv and attempting to use the only updated one on my system - the one in my global python installation:
```
╭─rico at Five-Pebbles in ~/Science/Data/062122Ferret_Pen4_mseq1_g1/kilosort3 23-02-07 - 15:55:39
╰─○ python -c "import numpy as np; print(np.__version__); print(np.__file__)"
1.24.2
/home/rico/.asdf/installs/python/3.11.1/lib/python3.11/site-packages/numpy/__init__.py
╭─rico at Five-Pebbles in ~/Science/Data/062122Ferret_Pen4_mseq1_g1/kilosort3 23-02-07 - 15:55:56
╰─○ source env/bin/activate                                                  
(env) ╭─rico at Five-Pebbles in ~/Science/Data/062122Ferret_Pen4_mseq1_g1/kilosort3 23-02-07 - 15:56:04
╰─(env) ○ python -c "import numpy as np; print(np.__version__); print(np.__file__)"
1.22.4
/home/rico/Science/Data/062122Ferret_Pen4_mseq1_g1/kilosort3/env/lib/python3.11/site-packages/numpy/__init__.py
```

Have other people run into similar issues? What is the suggested workaround here?

Datamance, 2023-02-08:
OK, I hunted down the version of numpy (1.23.1) that is compiled against the same C API as what's being used by Phy, while also still including the deprecated `np.bool`. You have to pin this dependency and then phy will run.

*Please* fix this. All it takes is pinning your dependencies, which you should do anyway.

caniko, 2023-02-20:
Consider using poetry to lock dependencies, and errors like this will never happen again.

caniko, 2023-02-24:
If you want to make this solution portable, I recently created [`phy-poetry`](https://github.com/caniko/phy-poetry). You can `pip install phy-poetry`.

closed 2024-06-04 by zm711