cortex-lab/phy #1197: ImportError: /usr/lib/x86_64-linux-gnu/libgssapi_krb5.so.2: undefined symbol: krb5_ser_context_init, version krb5_3_MIT

GoktugAlkan, 2023-02-21:
Dear community,

I followed the description for the installation of phy. After running the command `pip install phy --pre --upgrade` in the terminal of my linux computer, I got the message that the package phy was successfully installed. 

However, when I try to import the packages for the GUI, i.e., when I run the command `from phy.apps.template import template_gui`, I see the following error:

`ImportError: /usr/lib/x86_64-linux-gnu/libgssapi_krb5.so.2: undefined symbol: krb5_ser_context_init, version krb5_3_MIT`

When I install the package phy on a virtual machine on my linux computer, which operates with Windows 10, I don't see that error message. However, the virtual machine cannot access the GPU causing empty fields in the GUI. Hence, I strongly wish to use the GUI on my linux computer.

Could you please help me to overcome this error?

Thanks in advance for your effort!

mani-venkatesh, 2023-03-03:
Any updates on this error? I have been experiencing the same

reinderdorman, 2023-03-22:
Having the same problem on Ubuntu 20.04.6 LTS. Installed latest version of phy2 today, checked if libgssapi_krb5 is up to date, but no luck. Switching to windows is not really a viable option at the moment (with other things on this PC).

Any update would be appreciated!

llobetv, 2023-08-22:
I have the same issue, it's probably related to incompatible version of library. 

Quick and dirty fix that works for me:

- Download this conda env and extract it: 
https://mega.nz/file/nwpEwJ4B#TDSDyWyzYIj1aqeDLhKlmT8q8irE9U-bZBqGjwoLIvg
- launch: conda create -n phy2 --clone path/to/file/extract
- In ~/miniconda3/envs/phy2/bin/phy change absolute for python pathway to the rightfull one ~/miniconda3/envs/phy2/bin/python

I would probably works to install the same version for each library but i didn't test it.

Hope it will help :)