cortex-lab/phy #1200: Include env or conda env in installation instructions

zm711, 2023-02-25:
In an old version of the Phy installation it walked through creating a conda env to install phy. Since not all scientists may be comfortable with Python and its env structure (especially if coming from Kilosort and Matlab), it would probably be nice to still include this in the readme, so people don't install phy directly into their base environment. Happy to be overruled on that, but something like:

```sh
conda create -n phy2 python
conda activate phy2
pip install phy --pre --upgrade
```

(and with the current NumPy issues a downgrade to 1.23.1)

Or equivalent with env or pipenv.

PaolaGith, 2023-05-13:
Thank you so much!
I am very new to python and I don't know why but I could not install phy by using pip install.

zm711, 2023-05-13:
Howdy @PaolaGith. Couple questions to start:
operating system (Linux, Windows, Mac)? Conda install or not? In a virtual environment or not?

Could you copy the exact error here so we can see it? Happy to help once I see exactly what is going on. Could be a couple things we can try to troubleshoot.

PaolaGith, 2023-05-18:
Hi @zm711, 
Sorry for the late reply.

When I run the 
`phy template-gui params.py `
I get this error

```
File "/home/paola/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/qt.py", line 29, in <module>
    from OpenGL import GL  # noqa
    ^^^^^^^^^^^^^^^^^^^^^
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/GL/__init__.py", line 4, in <module>
    from OpenGL.GL.VERSION.GL_1_1 import *
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/GL/VERSION/GL_1_1.py", line 14, in <module>
    from OpenGL.raw.GL.VERSION.GL_1_1 import *
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/raw/GL/VERSION/GL_1_1.py", line 7, in <module>
    from OpenGL.raw.GL import _errors
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/raw/GL/_errors.py", line 4, in <module>
    _error_checker = _ErrorChecker( _p, _p.GL.glGetError )
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/error.py", line 183, in __init__
    self._isValid = platform.CurrentContextIsValid
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/platform/baseplatform.py", line 15, in __get__
    value = self.fget( obj )
            ^^^^^^^^^^^^^^^^
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/platform/baseplatform.py", line 356, in CurrentContextIsValid
    return self.GetCurrentContext
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/platform/baseplatform.py", line 15, in __get__
    value = self.fget( obj )
            ^^^^^^^^^^^^^^^^
  File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/OpenGL/platform/egl.py", line 106, in GetCurrentContext
    return self.EGL.eglGetCurrentContext
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'eglGetCurrentContext'
```

zm711, 2023-05-18:
Howdy @PaolaGith,

Couple more things: 
This looks like Linux, but which distribution? Also you are running this on python 3.11 which this definitely hasn't been tested against. I believe the last tested was 3.8, but I have this working on 3.9 and maybe 3.10. So you may want to downgrade your python just to be safe. It looks like openGL might be part of the problem. Do you have that installed on your system? Is it the correct version for Phy?

PaolaGith, 2023-05-19:
Hi,
My python version is 3.9. I installed the version of phy using these commands:

```
conda create -n phy2 python
conda activate phy2
pip install phy --pre --upgrade
```

I am very new to python so I am not used to all these issues when installing a software.
If I checked correctly, openGL doesn't seem to be installed on my system. On my base environment It looks like I only have:
```
openjpeg                  2.4.0                h3ad879b_0  
openpyxl                  3.0.10           py39h5eee18b_0  
openssl    
```

While on my phy2 env I only have:

`openssl`

Should I install openGL?And how?

zm711, 2023-05-19:
So Python packing and environments can definitely be confusing when you first start. Your base python might be 3.8 (ie what came with your computer). But if you look here:
> File "/home/paola/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 22, in <module>

You've actually installed python3.11 into your virtual environment. To fix that you would actually need to specify the python in the command

```python
conda create -n phy2 python=3.8
```

I didn't include that in my instructions originally, because I was suggesting the maintainers update their docs with something based on that template, but not use exactly that. 

So the general idea with python is that you have a (base) which is what comes with certain computers (traditionally Macs and Linux, but not Windows) that is used by the OS for some things behind the scenes. If you then want an isolated scientific or other package it is best to not alter that base python and instead make a virtual environment. There are two major choices for that: pypi environments (venv, virtualenv, etc) or the conda ecosystem (conda env). The instructions above are for a conda environment generation.  Once you have the environment generated it should allow for a clean install of all the python packages you need without hurting anything else for your base python. (Just a mini primer on python environments).

So if you type 
```python
conda activate phy2
conda list
```
Could you copy all the output here?

As far as installing something like openGL that would be a computer specific process so which operating system are you using specifically? The easiest would like be something like: How do I install openGL on a .... (put in whatever operating system you have) into google and see what the instructions are.

PaolaGith, 2023-05-22:
Dear zm711,

I downgraded my python version. If I do

```
(phy2) paola@PaolaHP:/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M2/Spike_sorting_results_M2/wf_MS_all$ python -V
Python 3.8.0
```

Before I received your message, I was getting a lot of errors of missing packages etc. After one afternoon of trouble shooting, now I am able to open Phy, but I can only see the list of the neurons. The graphic is missing.

[image]

Nevertheless, I don't get any error:

```
(phy2) paola@PaolaHP:/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M2/Spike_sorting_results_M2/wf_MS_all/phy$ phy template-gui params.py
11:37:53.192 [W] model:1408           File /mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M2/wf_MS_all/phy/recording.dat does not exist.
11:37:53.378 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
11:37:53.391 [W] traces:462           File /mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M2/wf_MS_all/phy/recording.dat does not exist.
11:37:53.392 [W] traces:462           File /mnt/c/Users/Paola/Desktop/SPINAL_64_ARRAY/Array_M2/wf_MS_all/phy/recording.dat does not exist.
QFactoryLoader::QFactoryLoader() checking directory path "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms" ...
QFactoryLoader::QFactoryLoader() looking at "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqeglfs.so"
Found metadata in lib /home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqeglfs.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "eglfs"
        ]
    },
    "archreq": 0,
    "className": "QEglFSIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("eglfs")
QFactoryLoader::QFactoryLoader() looking at "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqlinuxfb.so"
Found metadata in lib /home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqlinuxfb.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "linuxfb"
        ]
    },
    "archreq": 0,
    "className": "QLinuxFbIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("linuxfb")
QFactoryLoader::QFactoryLoader() looking at "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqminimal.so"
Found metadata in lib /home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/platforms/libqminimal.so, metadata=
{
    "IID": "org.qt-project.Qt.QPA.QPlatformIntegrationFactoryInterface.5.3",
    "MetaData": {
        "Keys": [
            "minimal"
        ]
    },
    "archreq": 0,
    "className": "QMinimalIntegrationPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("minimal")
[... 446 lines cut ...]
            "image/webp"
        ]
    },
    "archreq": 0,
    "className": "QWebpPlugin",
    "debug": false,
    "version": 331520
}

Got keys from plugin meta data ("webp")
QFactoryLoader::QFactoryLoader() checking directory path "/home/paola/anaconda3/envs/phy2/bin/imageformats" ...
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqgif.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqicns.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqico.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqjpeg.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqsvg.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqtga.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqtiff.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqwbmp.so"
loaded library "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/imageformats/libqwebp.so"
QFactoryLoader::QFactoryLoader() checking directory path "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/accessible" ...
QFactoryLoader::QFactoryLoader() checking directory path "/home/paola/anaconda3/envs/phy2/bin/accessible" ...
QFactoryLoader::QFactoryLoader() checking directory path "/home/paola/anaconda3/envs/phy2/lib/python3.8/site-packages/PyQt5/Qt5/plugins/accessiblebridge" ...
QFactoryLoader::QFactoryLoader() checking directory path "/home/paola/anaconda3/envs/phy2/bin/accessiblebridge" ...
```

PaolaGith, 2023-05-22:
Please find here the list of packages of my environment:

```
(phy2) paola@PaolaHP:/mnt/d/Documents_D/MATLAB_D/SPINAL_64_ARRAY/Array_M2/Spike_sorting_results_M2/wf_MS_all/phy$ conda list
# packages in environment at /home/paola/anaconda3/envs/phy2:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
asttokens                 2.2.1                    pypi_0    pypi
backcall                  0.2.0                    pypi_0    pypi
bzip2                     1.0.8                h7b6447c_0  
ca-certificates           2023.01.10           h06a4308_0  
certifi                   2023.5.7                 pypi_0    pypi
charset-normalizer        3.1.0                    pypi_0    pypi
click                     8.1.3                    pypi_0    pypi
cloudpickle               2.2.1                    pypi_0    pypi
colorcet                  3.0.1                    pypi_0    pypi
comm                      0.1.3                    pypi_0    pypi
contourpy                 1.0.7                    pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
cython                    0.29.34                  pypi_0    pypi
dask                      2023.5.0                 pypi_0    pypi
debugpy                   1.6.7                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
executing                 1.2.0                    pypi_0    pypi
fonttools                 4.39.4                   pypi_0    pypi
fsspec                    2023.5.0                 pypi_0    pypi
ghp-import                2.1.0                    pypi_0    pypi
h5py                      3.8.0                    pypi_0    pypi
idna                      3.4                      pypi_0    pypi
importlib-metadata        6.6.0                    pypi_0    pypi
importlib-resources       5.12.0                   pypi_0    pypi
ipykernel                 6.23.1                   pypi_0    pypi
ipython                   8.12.2                   pypi_0    pypi
ipython-genutils          0.2.0                    pypi_0    pypi
jedi                      0.18.2                   pypi_0    pypi
jinja2                    3.1.2                    pypi_0    pypi
joblib                    1.2.0                    pypi_0    pypi
jupyter-client            8.2.0                    pypi_0    pypi
jupyter-core              5.3.0                    pypi_0    pypi
kiwisolver                1.4.4                    pypi_0    pypi
ld_impl_linux-64          2.38                 h1181459_1  
libedit                   3.1.20221030         h5eee18b_0  
libffi                    3.2.1             hf484d3e_1007  
libgcc-ng                 11.2.0               h1234567_1  
libgomp                   11.2.0               h1234567_1  
libstdcxx-ng              11.2.0               h1234567_1  
libuuid                   1.41.5               h5eee18b_0  
locket                    1.0.0                    pypi_0    pypi
markdown                  3.3.7                    pypi_0    pypi
markupsafe                2.1.2                    pypi_0    pypi
matplotlib                3.7.1                    pypi_0    pypi
matplotlib-inline         0.1.6                    pypi_0    pypi
mergedeep                 1.3.4                    pypi_0    pypi
mkdocs                    1.4.3                    pypi_0    pypi
mtscomp                   1.0.2                    pypi_0    pypi
ncurses                   6.4                  h6a678d5_0  
nest-asyncio              1.5.6                    pypi_0    pypi
numpy                     1.23.1                   pypi_0    pypi
opencv-python-headless    4.7.0.72                 pypi_0    pypi
openssl                   1.1.1t               h7f8727e_0  
packaging                 23.1                     pypi_0    pypi
param                     1.13.0                   pypi_0    pypi
parso                     0.8.3                    pypi_0    pypi
partd                     1.4.0                    pypi_0    pypi
pexpect                   4.8.0                    pypi_0    pypi
phy                       2.0b5                    pypi_0    pypi
phylib                    2.4.3                    pypi_0    pypi
pickleshare               0.7.5                    pypi_0    pypi
[... 20 lines cut ...]
pyzmq                     25.0.2                   pypi_0    pypi
qtconsole                 5.4.3                    pypi_0    pypi
qtpy                      2.3.1                    pypi_0    pypi
readline                  7.0                  h7b6447c_5  
requests                  2.30.0                   pypi_0    pypi
scipy                     1.10.1                   pypi_0    pypi
setuptools                66.0.0           py38h06a4308_0  
six                       1.16.0                   pypi_0    pypi
sqlite                    3.33.0               h62c20be_0  
stack-data                0.6.2                    pypi_0    pypi
tk                        8.6.12               h1ccaba5_0  
toolz                     0.12.0                   pypi_0    pypi
tornado                   6.3.2                    pypi_0    pypi
tqdm                      4.65.0                   pypi_0    pypi
traitlets                 5.9.0                    pypi_0    pypi
typing-extensions         4.5.0                    pypi_0    pypi
tzdata                    2023c                h04d1e81_0  
urllib3                   2.0.2                    pypi_0    pypi
watchdog                  3.0.0                    pypi_0    pypi
wcwidth                   0.2.6                    pypi_0    pypi
wheel                     0.38.4           py38h06a4308_0  
xz                        5.4.2                h5eee18b_0  
zipp                      3.15.0                   pypi_0    pypi
zlib                      1.2.13               h5eee18b_0 
```

zm711, 2023-05-22:
@PaolaGith  One thing you could try is follow instructions from #1143. For Linux there can be issues with the data updating and many people have found success with that. Could you try their solution and see if it helps yours set-up?

PaolaGith, 2023-05-25:
Hi, 
I tried by following the instructions in 1143 issue, but it did not work for me:

`env QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui params.py`

Still I get the window with the neurons' list but not the graphics...

PaolaGith, 2023-06-02:
@zm711 
Do you have any other thing to suggest? It is really weird I can't use it despite I don't get any errors...

zm711, 2023-06-02:
@PaolaGith 
Sorry for the delay. I'm a Mac and Windows user so I'm less familiar with Linux. Other people have had success trying to download phy on Windows. So if you have the option to do this on a Windows computer you could test it there. If it works there then it definitely tells you there's something with your specific Linux set-up (missing package or something like that). Since I'm less familiar with Linux it's a bit harder for me to know which package could be missing, but since it seems to be a display issue, it's likely a missing display package. Do you have access to a Windows computer to test? Otherwise, unfortunately I don't think I can specifically help other than saying try a different Linux distro and try to make sure you have all packages that Phy might use (ie you'll need to read through the source code and requirements files and confirm). Sorry!

zm711, 2023-06-07:
Hey @PaolaGith 

So I also don't know running through VSCode super well. I mainly use it just to write code and then doing my testing elsewhere but based on this:
`/mnt/d/` 
from your image above I wonder if your VSCode is running WSL under the hood. Windows OS doesn't use the `mnt` directory to indicate using the D drive, but WSL does. If your VSCode is running WSL then maybe try running Phy outside of VSCode on the native Windows OS?

PaolaGith, 2023-06-14:
@zm711 
Hi!
Same for me, same situation. I am using WSL and the mnt directory.
"try running Phy outside of VSCode on the native Windows OS" How can I do this?

Thank you

(it is a bit frustrating that I am not able to use phy despite everything is correctly installed...)

zm711, 2023-06-14:
@PaolaGith 

Totally agree. That is super frustrating.

So for Windows (with anaconda)

Go to your Start menu and search Anaconda Prompt. That should bring up a terminal looking window.
In that window you should type
```sh
conda activate phy2
cd path\to\the\file
phy template-gui params.py
```
An alternative way to do it would be 
```sh
conda activate phy2
phy template-gui path\to\params.py
```

Of note the `activate phy2` needs to be the name of your conda env. So if you named it something else put that there. (in my example above I named in phy2). Based on your code above it looks like you named it `phy2` as well. To be clear the `path\to\the\file` needs to be the windows path to your `params.py`. From above it looks like it should be something like:

`Documents_D\MATLAB_D\SPINAL_64_ARRAY\Array_M2\Spike_sorting_results_M2`

Or with the shell it would be

```sh
cd/d D:\Documents_D\MATLAB_D\SPINAL_64_ARRAY\Array_M2\Spike_sorting_results_M2
```

If your `params.py` is not located on your `C:` drive then you made need to type `cd/d` instead. That is the window command to allow you to switch drives when using the `cd` command.

Could you try that and let me know how that goes?

PaolaGith, 2023-06-21:
Dear @zm711 

I tried to activate my conda phy2 environment on the windows Anaconda prompt, but it looks like the env does not exist.

`EnvironmentNameNotFound:` Could not find conda environment: phy2'

So I tried to clone the environment by using 
`conda env export --from-history > environment.yml`

and then creting it on the Anaconda prompt in windows:

`conda env create -f environment.yml`

Then I activate the env phy2 and run the phy command, but I get error (of course...)

`(phy2) D:\Documents_D\MATLAB_D\SPINAL_64_ARRAY\Array_M2\Spike_sorting_results_M2>phy template-gui params.py
'phy' is not recognized as an internal or external command,
operable program or batch file.`

I really don't know how to proceed and I can't keep asking colleagues to lend me their personal computers...I really do not understand why, if there are no error, I get visualization issues in the window...

zm711, 2023-06-21:
@PaolaGith 

so my last attempt (sorry this really really stinks). The environment yml was just updated so could you copy that file 

environment.yml 

from the initial page of this repo. And once you have the file saved put that file into your documents folder 
Then from the anaconda prompt cd to documents (make sure you are in the folder with the file) and type

conda env create -f environment.yml (note this will try to automatically name it Phy2 so you should remove any old copies of Phy2 env you have.

PaolaGith, 2023-06-23:
@zm711 
It worked finally!!! Now I can also visualize the waveforms!! At least on Anaconda prompt in windows.
Do you suggest me to create also a new phy2 environment in vscode and reinstall phy with the new environment.yml?

Thanks a lot for your patience!

zm711, 2023-06-23:
Howdy @PaolaGith 

Congrats that’s so awesome! Happy to help when I can. It was a long road but we got there. 

So yeah you can try the exact same thing for vscode. If it works then you can do phy from there. If it doesn’t you should be able to just use the native window version. (Which is better than borrowing someone else’s computer). Definitely drop a comment here if it works so people can learn from this :).

closed 2023-07-05 by zm711

PaolaGith, 2023-07-10:
@zm711 
Hi, I confirm that by downloading the environment.yml file from github and creating the new environment in anaconda prompt in windows, using `conda env create -f environment.yml`, it works.

So I tried the same strategy in VS code but, again I can load the data but I can't see the waveforms.
Apparently this solution did not work in this case.

I hope it helps.

Thank you