cortex-lab/phy #1202: Error: No such command 'template-gui'.

Lildev0408, 2023-03-07:
Hello,
I'm trying to get the phy works, but having no lucks. 
I created conda environment using the environment.yml file. 
I had an issue with scikit-learn installing. But I fixed it by installing scikit-learn first and the others (replaced order of the packages in the environment file). And I needed to install the 'traitlets' in addition to the provided yml.
I'm using latest version of miniconda and python 3.11. In the macOS Ventura version 13 (M1 chip). 
When I try to run the phy, I keep get the error message of "Error: No such command 'template-gui'"
I saw a comment that it should be fixed after PyQt5. 
But I think at some point, it tries to use PyQt4 which I don't have. 
I would really appreciate it if someone can help me out!

zm711, 2023-03-08:
@Lildev0408 dev0408
what did you exactly type? did you activate your environment? the environment yaml definitely doesn't include python 3.11 so which dependencies did you include. For instance using numpy 1.24+ will fail because of #1193. 

My guess is first type
```bash
conda activate phy2
```

Then try typing
```bash
phy template-gui params.py
```

In general `template-gui` is not a command by itself. It is an argument to be given after the `phy` command.

brendonw1, 2023-04-15:
I have this same issue.
I am able to type "phy" and I get this:
_brendonw@LNX10027:~/analysis/Brendon/NicoletteEphys_ConcatenatedGroupDats$ phy
Usage: phy [OPTIONS] COMMAND [ARGS]...

  By default, the `phy` command does nothing. Add subcommands with plugins
  using `attach_to_cli()` and the `click` library.

Options:
  --version   Show the version and exit.
  -h, --help  Show this message and exit._

... so I think I'm able to access phy itself
But typing "phy template-gui params.py" (in the right folder, used auto-complete to make sure params. py was in there) gives:
_brendonw@LNX10027:~/analysis/Brendon/NicoletteEphys_ConcatenatedGroupDats$ phy template-gui params.py
Usage: phy [OPTIONS] COMMAND [ARGS]...

Error: No such command "template-gui"._

??

zm711, 2023-04-15:
Did you install in base or a virtual environment? And are you using conda or not? Could you copy the whole output of the error here (straight from the terminal).

brendonw1, 2023-04-15:
**Thanks so much! 
Installed in base I think.  I had someone do it for us, but I think these were the install commands:**
[root@LNX10027 ~]# pip install phy traitlets
Collecting phy
  Downloading phy-1.0.9-py2.py3-none-any.whl (204 kB)
     |████████████████████████████████| 204 kB 2.2 MB/s
Collecting traitlets
  Using cached traitlets-5.9.0-py3-none-any.whl (117 kB)
Installing collected packages: phy, traitlets
Successfully installed phy-1.0.9 traitlets-5.9.0
 [root@LNX10027 python3.8]# sed -i 's/np.float,/float,/; s/[np.int](http://np.int/),/int,/; s/np.bool/bool/' /usr/local/lib/python3.8/dist-packages/phy/utils/_types.py

**no conda or source activate here
And shown above is the full terminal output actually**

brendonw1, 2023-04-15:
Essentially we followed instructions on the front of this github repo webpage but had to compensate for some things not working.  including "pre" causing errors for one thing

zm711, 2023-04-15:
The official instructions are to install by typing:
pip install phy --pre --upgrade (your install is missing the --pre --upgrade). So I would actually start with that. 

Is there a reason why you don't want to use a virtual environment? Phy hasn't been updated for a while so it might be safer to keep it isolated from your base python install using a conda env or one of the python virtual envs. (I'm a conda user personally so it's easier for me to give conda recs vs python environment recs). 

In that case what error did pre cause?

brendonw1, 2023-04-15:
**
Again, appreciated
We just went by the front page instructions here: https://github.com/cortex-lab/phy#installation-instructions

I didn't see anything about environments, but are there better instructions elsewhere?  I'm on Ubuntu 20.04 (bc managed by my institution)

Using --pre produces tons of output but ends with the following:**

Collecting PyQtWebEngine-Qt5>=5.15.0
  Using cached PyQtWebEngine_Qt5-5.15.2-py3-none-manylinux2014_x86_64.whl (67.5 MB)
Collecting PyQt5>=5.15.4
  Using cached PyQt5-5.15.9.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 /tmp/tmpgm62_bkp prepare_metadata_for_build_wheel /tmp/tmp_z8sfes3
         cwd: /tmp/pip-install-_9o80yd1/PyQt5
    Complete output (31 lines):
    Traceback (most recent call last):
      File "/tmp/tmpgm62_bkp", line 126, in prepare_metadata_for_build_wheel
        hook = backend.prepare_metadata_for_build_wheel
    AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/tmp/tmpgm62_bkp", line 280, in <module>
        main()
      File "/tmp/tmpgm62_bkp", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/tmp/tmpgm62_bkp", line 130, in prepare_metadata_for_build_wheel
        return _get_wheel_metadata_from_wheel(backend, metadata_directory,
      File "/tmp/tmpgm62_bkp", line 159, in _get_wheel_metadata_from_wheel
        whl_basename = backend.build_wheel(metadata_directory, config_settings)
      File "/tmp/pip-build-env-k7a2g0e3/overlay/lib/python3.8/site-packages/sipbuild/api.py", line 46, in build_wheel
        project = AbstractProject.bootstrap('wheel',
      File "/tmp/pip-build-env-k7a2g0e3/overlay/lib/python3.8/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
        project.setup(pyproject, tool, tool_description)
      File "/tmp/pip-build-env-k7a2g0e3/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 586, in setup
        self.apply_user_defaults(tool)
      File "/tmp/pip-install-_9o80yd1/PyQt5/project.py", line 68, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-k7a2g0e3/overlay/lib/python3.8/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
        super().apply_user_defaults(tool)
      File "/tmp/pip-build-env-k7a2g0e3/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 237, in apply_user_defaults
        self.builder.apply_user_defaults(tool)
      File "/tmp/pip-build-env-k7a2g0e3/overlay/lib/python3.8/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
        raise PyProjectOptionException('qmake',
    sipbuild.pyproject.PyProjectOptionException
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmpgm62_bkp prepare_metadata_for_build_wheel /tmp/tmp_z8sfes3 Check the logs for full command output.

zm711, 2023-04-15:
>Again, appreciated
We just went by the front page instructions here: https://github.com/cortex-lab/phy#installation-instructions

It's more about python best practices. For most python packages it's best to store them in a virtual environment (for example if you have NumPy>=1.24 in your base environment it will break Phy. But if you have a virtual environment you can force that environment to have <1.24 so that you don't break Phy. If you don't want to use conda maybe read up a bit here:
https://docs.python.org/3/library/venv.html. They actually use to include instructions on how to put Phy2 into a virtual environment, but I think at some point they figured people would just do it themselves (which is a shame in my opinion).

>ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmpgm62_bkp prepare_metadata_for_build_wheel /tmp/tmp_z8sfes3 Check the logs for full command output.

So this is saying that the metadata isn't correct, so a couple of the packages aren't playing well together (ie the PyQt5 that is being downloaded either isn't happy with your pip, your python, or the 20.04)

You could start by trying to update pip (to make sure it's up-to-date):

```bash
pip install --upgrade pip
```

If updating `pip` doesn't fix things you'll probably have to read up on your ubuntu version and try to figure out which package is messing things up. 
To make the virtual env with the "correct" packages they do include the requirements.txt file 
https://github.com/cortex-lab/phy/blob/master/requirements.txt
But unfortunately it doesn't work super well, since they haven't pinned their version numbers (like NumPy that I mentioned above). 

Their recommendation for python in the `requirements.txt` is 3.7 so if changing pip doesn't work you could also try install 3.7 to see if that helps with version numbering between the packages. (again with a venv or conda you can easily specify which python version you want).

brendonw1, 2023-04-16:
Thank you.  pip install --upgrade pip was successful but didn't fix the problems either with installation or running template-gui :(

I guess we'll keep digging

zm711, 2023-08-05:
@brendonw1 

Just as an FYI, the package has been updated to allow for python 3.11 and numpy 1.24. You could try reinstalling on your ubuntu system if you wanted. Took a spell, but if you're still trying to work with `Phy` great chance to try again. You could use the below command to automatically do everything for you (if you use `conda`)

```python
conda create env -f environment.yml
```

brendonw1, 2023-08-05:
Oh thanks so much.
I think we started with a new box and it's working well since then.  Much
appreciated
B

Brendon Watson, MD-PhD
Assistant Professor in Psychiatry, Bioinformatics and Biomedical
Engineering

Biomedical Sciences Research Building, Room 5059
University of Michigan
109 Zina Pitcher Place
Ann Arbor, MI 48109-5720
Lab Website: http://watsonneurolab.org
Clinical phone: 734-764-0231

closed 2024-06-04 by zm711