cortex-lab/phy #1207: numpy broken and empty interface on Linux

rat-h, 2023-05-02:
Linux Ubuntu-like with KDE 
```
<ws>:~/.local/apps$ python --version
Python 3.10.6
<ws>:~/.local/apps$ qmake --version
QMake version 3.1
Using Qt version 5.15.3 in /usr/lib/x86_64-linux-gnu
```

Virtual environment
```
<ws>:~/.local/apps$ python -m venv phy
<ws>:~/.local/apps$ source phy/bin/activate
(phy) <ws>:~/.local/apps$ pip install phy --pre --upgrade
Collecting phy
  Using cached phy-2.0b5-py2.py3-none-any.whl
Collecting requests
  Using cached requests-2.29.0-py3-none-any.whl (62 kB)
Collecting scipy
  Using cached scipy-1.10.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (34.4 MB)
Collecting click
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting qtconsole
  Using cached qtconsole-5.4.2-py3-none-any.whl (121 kB)
Collecting mtscomp
  Using cached mtscomp-1.0.2-py2.py3-none-any.whl (16 kB)
Collecting PyQtWebEngine
  Using cached PyQtWebEngine-5.15.6-cp37-abi3-manylinux1_x86_64.whl (230 kB)
Collecting matplotlib
  Using cached matplotlib-3.7.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (11.6 MB)
Collecting dask
  Using cached dask-2023.4.1-py3-none-any.whl (1.2 MB)
Collecting PyQt5
  Using cached PyQt5-5.15.9-cp37-abi3-manylinux_2_17_x86_64.whl (8.4 MB)
Collecting pillow
  Using cached Pillow-9.5.0-cp310-cp310-manylinux_2_28_x86_64.whl (3.4 MB)
Collecting tqdm
  Using cached tqdm-4.65.0-py3-none-any.whl (77 kB)
Collecting h5py
  Using cached h5py-3.8.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.6 MB)
Collecting numpy
  Using cached numpy-1.24.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.3 MB)
Collecting colorcet
  Using cached colorcet-3.0.1-py2.py3-none-any.whl (1.7 MB)
Collecting joblib
  Using cached joblib-1.2.0-py3-none-any.whl (297 kB)
Collecting pyopengl
  Using cached PyOpenGL-3.1.6-py3-none-any.whl (2.4 MB)
Collecting mkdocs
  Using cached mkdocs-1.4.2-py3-none-any.whl (3.7 MB)
Collecting phylib
  Using cached phylib-2.4.3-py2.py3-none-any.whl (80 kB)
Collecting cython
  Using cached Cython-3.0.0b2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (3.6 MB)
Collecting pyct>=0.4.4
  Using cached pyct-0.5.0-py2.py3-none-any.whl (15 kB)
Collecting toolz>=0.10.0
  Using cached toolz-0.12.0-py3-none-any.whl (55 kB)
Collecting packaging>=20.0
  Using cached packaging-23.1-py3-none-any.whl (48 kB)
Collecting partd>=1.2.0
  Using cached partd-1.4.0-py3-none-any.whl (18 kB)
Collecting pyyaml>=5.3.1
  Using cached PyYAML-6.0-cp310-cp310-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_12_x86_64.manylinux2010_x86_64.whl (682 kB)
Collecting importlib-metadata>=4.13.0
  Using cached importlib_metadata-6.6.0-py3-none-any.whl (22 kB)
Collecting cloudpickle>=1.5.0
  Using cached cloudpickle-2.2.1-py3-none-any.whl (25 kB)
Collecting fsspec>=2021.09.0
  Using cached fsspec-2023.4.0-py3-none-any.whl (153 kB)
[... 187 lines cut ...]
15:44:42.226 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
15:44:42.228 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
15:44:42.232 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
15:44:42.233 [W] model:55             133/4489 values are nan in /home/rth/Rhythm_FPGA-100.0-ks2.5/similar_templates.npy, replacing by zero.
Data files              /home/rth/Rhythm_FPGA-100.0-ks2.5/temp_wh.dat
Directory               /home/rth/Rhythm_FPGA-100.0-ks2.5
Duration                1014.6s
Sample rate             30.0 kHz
Data type               int16
# of channels           128
# of channels (raw)     128
# of templates          67
# of spikes             46,850
(phy) <ws>:~/Rhythm_FPGA-100.0-ks2.5$ phy template-gui params.py 
15:45:39.332 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
15:45:39.333 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
15:45:39.337 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
15:45:39.338 [W] model:55             133/4489 values are nan in /home/rth/Rhythm_FPGA-100.0-ks2.5/similar_templates.npy, replacing by zero.
```
[image]

Then try to close

[image]
[image]

zm711, 2023-05-03:
@rat-h
Not completely sure, but first thing I would try are these two commands:

```sh
phy template-gui params.py --clear-state
phy template-gui params.py --clear-cache
```
Those are the commands to try to fix any GUI errors. They just clear the global and local gui state. here is the info from their
RTD: https://phy.readthedocs.io/en/latest/troubleshooting/

I would say worth a try and if it still doesn't work then maybe use a lower python version, but not sure with Linux issues.

rat-h, 2023-05-04:
I tried the last command before with the same result. 

I have run both and combination of two
```
phy template-gui params.py --clear-state --clear-cache
```
with no luck

zm711, 2023-05-04:
The numpy #1193 and linux issues #1202 have been reported, so I think the issue might be that there are multiple versions of packages no longer playing super well together. I know @caniko wrote a [phy-poetry](https://github.com/caniko/phy-poetry) with versions pinned, but I'm not sure which OS it works with/which python it needs (I also don't know poetry for dependency-management), but you might want to check out the repo and decide if you want to give that a try.

caniko, 2023-05-04:
It works with every OS supported by `phy` @zm711.

You can just `pip install phy-poetry`, and proceed as before (as if `pip install phy`). No need to worry about poetry that is strictly for the developer to worry about, not the end-user :relaxed:

zm711, 2023-05-04:
Absolutely agree that the numpy issue is for everything, but on Windows with conda one can spin up an env for phy with no problem (after reverting numpy), so I'm wondering if there is some other package that needed to be pinned for this to work on some of the Linux flavors @caniko. Most of the recent issues all seem to be linux-based even after reverting numpy <1.24. Either way if you try this (@rat-h) and it works could be a good way to solve issues for a lot of the linux users posting issues.

rat-h, 2023-05-04:
**Update**
I installed `Miniconda3-py39_23.3.1-0-Linux-x86_64` which required to reinstall `PyQt5` 
```
pip install pyqt5 pyqt5-tools pyqt5.sip
```
and the result is the same on older Python version.

zm711, 2023-05-04:
Spyking Circus has this documentation for trying to install phy. So it might be worth trying their instructions, but there are so many packages I really don't know how to figure out the mismatched one causing the issue.

[spiking-circus](https://spyking-circus.readthedocs.io/en/latest/introduction/install.html)

```sh
pip install colorcet pyopengl qtconsole requests traitlets tqdm joblib click mkdocs dask toolz mtscomp
pip install --upgrade https://github.com/cortex-lab/phy/archive/master.zip
pip install --upgrade https://github.com/cortex-lab/phylib/archive/master.zip
 ```

caniko, 2023-05-04:
Do you have QT installed @rat-h? You need QT to be installed

rat-h, 2023-05-04:
it is! I use KDE and qt5-dep :)
```
qmake --version
QMake version 3.1
Using Qt version 5.15.3 in /usr/lib/x86_64-linux-gnu
```

rat-h, 2023-05-04:
**Update**
I ran it with `--debug` flag. Here's an output
```
phy template-gui params.py --clear-state --clear-cache --debug
16:11:34.293 [D] __init__:68          Start capturing exceptions.
16:11:34.298 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
16:11:34.299 [D] model:619            Loading spike clusters.
16:11:34.300 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
16:11:34.302 [D] model:569            No channel shank file found.
16:11:34.302 [D] model:692            Loading templates.
16:11:34.304 [D] model:724            Templates are dense.
16:11:34.304 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
16:11:34.304 [D] model:730            Loading the whitening matrix.
16:11:34.304 [D] model:737            Loading the inverse of the whitening matrix.
16:11:34.305 [W] model:55             133/4489 values are nan in /home/rth/Rhythm_FPGA-100.0-ks2.5/similar_templates.npy, replacing by zero.
16:11:34.305 [D] model:766            Loading features.
16:11:34.306 [D] model:781            Features are sparse.
16:11:34.306 [D] model:803            Loading template features.
16:11:34.307 [D] model:813            Template features are sparse.
16:11:34.307 [D] model:504            Load `cluster_KSLabel.tsv`.
16:11:34.307 [D] model:504            Load `cluster_ContamPct.tsv`.
16:11:34.308 [D] model:504            Load `cluster_Amplitude.tsv`.
16:11:34.308 [D] model:504            Load `cluster_group.tsv`.
16:11:34.452 [W] base:941             Deleting the cache directory /home/rth/Rhythm_FPGA-100.0-ks2.5/.phy.
16:11:34.453 [D] context:80           Create cache directory `/home/rth/Rhythm_FPGA-100.0-ks2.5/.phy`.
16:11:34.470 [D] context:100          Initialize joblib cache dir at `/home/rth/Rhythm_FPGA-100.0-ks2.5/.phy`.
16:11:34.470 [D] context:101          Reducing the size of the cache if needed.
16:11:34.471 [D] base:102             Add filter `high_pass`.
16:11:34.471 [W] base:948             Deleting /home/rth/.phy/TemplateGUI/state.json.
16:11:34.471 [D] config:31            Load config file `/home/rth/.phy/phy_config.py`.
16:11:34.471 [D] plugin:145           Loading 0 plugins.
16:11:34.472 [D] context:209          The file `/home/rth/Rhythm_FPGA-100.0-ks2.5/.phy/new_cluster_id.pkl` doesn't exist.
16:11:34.473 [D] context:209          The file `/home/rth/Rhythm_FPGA-100.0-ks2.5/.phy/spikes_per_cluster.pkl` doesn't exist.
16:11:34.473 [D] clustering:237       Recompute spikes_per_cluster manually: this might take a while.
16:11:34.475 [D] context:185          Save data to `/home/rth/Rhythm_FPGA-100.0-ks2.5/.phy/spikes_per_cluster.pkl`.
16:11:34.483 [D] gui:463              Creating GUI.
16:11:34.488 [D] state:177            The GUI state file `/home/rth/.phy/TemplateGUI/state.json` doesn't exist, creating a default one...
16:11:34.488 [I] state:180            Copied /home/rth/.local/apps/phy/lib/python3.10/site-packages/phy/apps/template/static/state.json to /home/rth/.phy/TemplateGUI/state.json.
16:11:34.488 [D] state:46             Load /home/rth/.phy/TemplateGUI/state.json for GUIState.
16:11:34.791 [D] gui:718              Add view ClusterView to GUI.
16:11:34.834 [D] gui:718              Add view SimilarityView to GUI.
16:11:34.878 [D] gui:718              Add view WaveformView to GUI.
16:11:34.889 [D] base:337             Set state for WaveformView.
16:11:34.901 [D] gui:718              Add view CorrelogramView to GUI.
16:11:34.902 [D] base:337             Set state for CorrelogramView.
16:11:34.923 [D] gui:718              Add view ISIView to GUI.
16:11:34.925 [D] base:337             Set state for ISIView.
16:11:34.948 [D] gui:718              Add view FeatureView to GUI.
16:11:34.950 [D] base:337             Set state for FeatureView.
16:11:34.971 [D] gui:718              Add view AmplitudeView to GUI.
16:11:34.972 [D] base:337             Set state for AmplitudeView.
16:11:34.991 [D] gui:718              Add view FiringRateView to GUI.
16:11:34.992 [D] base:337             Set state for FiringRateView.
16:11:35.197 [D] gui:718              Add view TraceView to GUI.
16:11:35.198 [D] base:337             Set state for TraceView.
16:11:35.198 [D] base:494             Set color scheme to depth.
16:11:35.238 [D] base:1623            Adding default color schemes to TraceView.
16:11:35.382 [D] gui:718              Add view ProbeView to GUI.
16:11:35.384 [D] base:337             Set state for ProbeView.
16:11:35.397 [D] gui:718              Add view TemplateFeatureView to GUI.
16:11:35.398 [D] base:337             Set state for TemplateFeatureView.
16:11:35.415 [D] gui:718              Add view ClusterScatterView to GUI.
16:11:35.416 [D] base:337             Set state for ClusterScatterView.
16:11:35.416 [D] base:494             Set color scheme to firing_rate.
16:11:35.417 [D] base:1623            Adding default color schemes to ClusterScatterView.
16:11:35.505 [D] gui:519              Load the geometry state.
```

rat-h, 2023-05-04:
I naively inserted a few printing lines into `ClusterView._reset_table` function in `phy/cluster/supervisor.py`. It seems, it was called twice: the first time with absolutely correctly read data, but the second time with `data=None`. Can it be a source of the empty GUI? 
```
16:51:24.549 [D] state:46             Load /home/rth/.phy/TemplateGUI/state.json for GUIState.
------------------------------------------
data [{'id': 0, 'ch': '7', 'sh': 0, 'depth': 553.0, 'fr': 0.04632306560134567, 'amp': 39.31453, 'n_spikes': 47, 'Amplitude': 472.2, 'ContamPct': 0.0, 'KSLabel': 'mua', 'group': None, 'is_masked': False}, {'id': 1, ......, 'fr': 0.14882516820857863, 'amp': 43.894745, 'n_spikes': 151, 'Amplitude': 567.4, 'ContamPct': 0.0, 'KSLabel': 'mua', 'group': None, 'is_masked': False}, {'id': 66, 'ch': '117', 'sh': 0, 'depth': 231.0, 'fr': 0.00887037426408747, 'amp': 54.40136, 'n_spikes': 9, 'Amplitude': 622.4, 'ContamPct': 100.0, 'KSLabel': 'mua', 'group': None, 'is_masked': False}]
columns ['id', 'ch', 'sh', 'depth', 'fr', 'amp', 'n_spikes', 'KSLabel', 'ContamPct', 'Amplitude']
value_names ['id', 'ch', 'sh', 'depth', 'fr', 'amp', 'n_spikes', 'KSLabel', 'ContamPct', 'Amplitude', {'data': ['group']}]
sort ['n_spikes', 'desc']
------------------------------------------
------------------------------------------
data None
columns ['id', 'ch', 'sh', 'depth', 'fr', 'amp', 'n_spikes', 'KSLabel', 'ContamPct', 'Amplitude', 'similarity']
value_names ['id', 'ch', 'sh', 'depth', 'fr', 'amp', 'n_spikes', 'KSLabel', 'ContamPct', 'Amplitude', 'similarity', {'data': ['group']}]
sort ('similarity', 'desc')
------------------------------------------

```

zm711, 2023-05-04:
Based on the traceback it seems like it is "waiting" too long while trying to update the gui state causing the runtime error. I guess that raises two questions:
1. Even if the gui state itself isn't saved are you able to use the gui, close and reload the curated values? You should check to see if the `cluster_group.tsv` file is being changed since that should be your curation file. If that is being changed then the gui is still working, but just misbehaving at close. Annoying, but not the end of the world. Just remember to use the `tsv` for your post-curation analysis.
2. What size is your data? Is it taking too long to try to save the gui state because you're overwhelming it with the amount of data you have (in which case congrats, haha)? Maybe your computer specs aren't high enough for the gui/data can handle?

zm711, 2023-05-04:
One other question/idea: So looking back at `template-describe` above I see 
>/home/rth/Rhythm_FPGA-100.0-ks2.5/temp_wh.dat

You're using the intermediate data generated by matlab in a binary format, whereas my phy is using the `*.bin` file that I originally loaded into Kilosort and not the `temp_wh.dat` from Kilosort. Do you have the raw input binary you used? The kilosort example is given as:
```matlab
>> datI = int16(dat);
>> whos datI 
  Name        Size                  Bytes  Class    Attributes

  datI      385x30000            23100000  int16              

% This dataset is 385 channels by 30000 time samples.
>> fid = fopen('D:\my\path\myNewFile.bin', 'w'); 
>> fwrite(fid, datI, 'int16');
>> fclose(fid);
```

So my `phy` is reading the `myNewFile.bin'` in their example and not the `temp_wh`?

caniko, 2023-05-05:
Can you try to load another dataset?

zm711, 2023-05-05:
I guess then that still leaves the question why linux is doing this wrong, but windows isn't. Since this is at the wait/timeout block (or the traceback is wrong) there must be some issue there, which then causes the gui to freak out and try to reload with empty dataset. I think you figured out what is going wrong, but I honestly don't know *why* it is going wrong. Like I said, I'm not super familiar with all the packaging behind linux distros, so I probably won't be any help beyond this point. If you have a working windows computer and access to a server I guess you could do the meat of your analysis on your Linux machine, do phy on windows, and then do the final analysis back on your linux machine. Annoying, but at least it works.

caniko, 2023-05-05:
I wouldn't blame it on Linux. I have gotten `phy-poetry` working on Ubuntu, Fedora, and Arch. PopOS is downstream from Ubuntu, so it should be fine.

Both Fedora and Arch was on KDE. PopOS comes with a heavily modified version of GNOME, you said you were on KDE. I am assuming you followed the System76 guide to switch to KDE, it should be fine... Strange that you are experiencing this.

I think your install might be broken.

zm711, 2023-05-05:
@rat-h Definitely keep me  in the loop if you figure it out. I want to switch over to PopOS eventually, but I'm stuck on windows for work.

@caniko Sorry wasn't meaning to blame linux. I meant that getting the packages to play together can require some work so I was agreeing with you in that it could be a specific build issue. Some of the other issues lately have involved Ubuntu problems, so I was wondering if some of the recent updates to some subset of packages upstream at the Ubuntu level are causing issues.

rat-h, 2023-05-08:
I installed fresh Ubuntu, `sudo apt install kde-stadard`, then `sudo apt install python3-venv`, then 
```
python3 -m venv phy
source phy/bin/activate
phy template-gui params.py --debug
```
The same result
 
[image]

Debug log absolutely the same
```
13:52:06.594 [D] __init__:68          Start capturing exceptions.
13:52:06.610 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
13:52:06.614 [D] model:619            Loading spike clusters.
13:52:06.616 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
13:52:06.625 [D] model:569            No channel shank file found.
13:52:06.627 [D] model:692            Loading templates.
13:52:06.635 [D] model:724            Templates are dense.
13:52:06.635 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
13:52:06.635 [D] model:730            Loading the whitening matrix.
13:52:06.637 [D] model:737            Loading the inverse of the whitening matrix.
13:52:06.641 [W] model:55             133/4489 values are nan in /experiment/similar_templates.npy, replacing by zero.
13:52:06.644 [D] model:766            Loading features.
13:52:06.650 [D] model:781            Features are sparse.
13:52:06.651 [D] model:803            Loading template features.
13:52:06.658 [D] model:813            Template features are sparse.
13:52:06.663 [D] model:504            Load `cluster_KSLabel.tsv`.
13:52:06.667 [D] model:504            Load `cluster_ContamPct.tsv`.
13:52:06.670 [D] model:504            Load `cluster_Amplitude.tsv`.
13:52:06.672 [D] model:504            Load `cluster_group.tsv`.
13:52:07.492 [D] context:100          Initialize joblib cache dir at `/experiment/.phy`.
13:52:07.493 [D] context:101          Reducing the size of the cache if needed.
13:52:07.511 [D] base:102             Add filter `high_pass`.
13:52:07.512 [D] config:31            Load config file `/home/rth/.phy/phy_config.py`.
13:52:07.513 [D] plugin:145           Loading 0 plugins.
13:52:07.523 [D] context:209          The file `/experiment/.phy/new_cluster_id.pkl` doesn't exist.
13:52:07.542 [D] context:185          Save data to `/experiment/.phy/spikes_per_cluster.pkl`.
13:52:07.614 [D] gui:463              Creating GUI.
13:52:07.630 [D] state:46             Load /home/rth/.phy/TemplateGUI/state.json for GUIState.
13:52:07.933 [D] gui:718              Add view ClusterView to GUI.
13:52:07.963 [D] gui:718              Add view SimilarityView to GUI.
13:52:08.030 [D] gui:718              Add view WaveformView to GUI.
13:52:08.038 [D] base:337             Set state for WaveformView.
13:52:08.057 [D] gui:718              Add view CorrelogramView to GUI.
13:52:08.059 [D] base:337             Set state for CorrelogramView.
13:52:08.092 [D] gui:718              Add view ISIView to GUI.
13:52:08.095 [D] base:337             Set state for ISIView.
13:52:08.115 [D] gui:718              Add view FeatureView to GUI.
13:52:08.117 [D] base:337             Set state for FeatureView.
13:52:08.147 [D] gui:718              Add view AmplitudeView to GUI.
13:52:08.153 [D] base:337             Set state for AmplitudeView.
13:52:08.208 [D] gui:718              Add view FiringRateView to GUI.
13:52:08.210 [D] base:337             Set state for FiringRateView.
13:52:08.598 [D] gui:718              Add view TraceView to GUI.
13:52:08.601 [D] base:337             Set state for TraceView.
13:52:08.601 [D] base:494             Set color scheme to depth.
13:52:08.644 [D] base:1623            Adding default color schemes to TraceView.
13:52:08.799 [D] gui:718              Add view ProbeView to GUI.
13:52:08.801 [D] base:337             Set state for ProbeView.
13:52:08.832 [D] gui:718              Add view TemplateFeatureView to GUI.
13:52:08.834 [D] base:337             Set state for TemplateFeatureView.
13:52:08.871 [D] gui:718              Add view ClusterScatterView to GUI.
13:52:08.873 [D] base:337             Set state for ClusterScatterView.
13:52:08.874 [D] base:494             Set color scheme to firing_rate.
13:52:08.875 [D] base:1623            Adding default color schemes to ClusterScatterView.
13:52:08.996 [D] gui:519              Load the geometry state.
14:12:49.483 [D] base:276             Close view ClusterScatterView.
14:12:49.485 [D] state:171            Update GUI state for ClusterScatterView
14:13:13.147 [E] qt:178               Timeout in _block().
[... 92 lines cut ...]
  File "/home/rth/phy/lib/python3.10/site-packages/phy/gui/widgets.py", line 191, in wait
    _block(self.have_all_finished)
  File "/home/rth/phy/lib/python3.10/site-packages/phy/gui/qt.py", line 182, in _block
    raise RuntimeError("Timeout in _block().")
RuntimeError: Timeout in _block().

14:13:21.740 [E] __init__:62          An error has occurred (RuntimeError): Timeout in _block().
Traceback (most recent call last):
  File "/home/rth/phy/lib/python3.10/site-packages/phy/gui/gui.py", line 590, in closeEvent
    res = emit('close', self)
  File "/home/rth/phy/lib/python3.10/site-packages/phylib/utils/event.py", line 141, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/rth/phy/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 729, in _save_gui_state
    gui.state.update_view_state(self.cluster_view, self.cluster_view.state)
  File "/home/rth/phy/lib/python3.10/site-packages/phy/cluster/supervisor.py", line 333, in state
    b.wait()
  File "/home/rth/phy/lib/python3.10/site-packages/phy/gui/widgets.py", line 191, in wait
    _block(self.have_all_finished)
  File "/home/rth/phy/lib/python3.10/site-packages/phy/gui/qt.py", line 182, in _block
    raise RuntimeError("Timeout in _block().")
RuntimeError: Timeout in _block().

14:13:21.781 [D] __init__:87          Stop capturing exceptions.
14:13:21.781 [D] __init__:87          Stop capturing exceptions.
```

rat-h, 2023-05-08:
Well I figured out initialization problem - it isn't a problem just two views: `ClusterView` and `SimilarityView`. The last one with empty data.

zm711, 2023-05-08:
@ rat-h. Found this in the old issues. Might want to try? #1143

closed 2023-05-08 by rat-h