cortex-lab/phy #1209: OpenGL core dumped on launching phy on Ubuntu 23.04

larsrollik, 2023-06-09:
Same issue as #1196, but for Ubuntu 23.04.

Install
```
conda create -n phy python=3.9 -y
conda activate phy
pip install phy --pre --upgrade

```

Error
```
$ phy template-gui params.py

15:07:10.569 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
15:07:10.764 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
15:07:10.875 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
15:07:12.359 [W] gui:671              Could not create view FeatureView.
15:07:12.393 [W] gui:671              Could not create view TraceView.
15:07:12.450 [W] gui:671              Could not create view TemplateFeatureView.
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)

$
```

zm711, 2023-08-09:
@larsrollik 

Not sure if it matters any more for you, but there are new installation instructions to install from source rather than pypi, which has helped others to ensure all packages are coordinated. 

Easiest thing would be to download the environment.yaml and run:

```python
conda env create -f environment.yml
```

larsrollik, 2024-04-18:
Thank you, @zm711 !
I ended up containerizing it to avoid these spurious issues.

closed 2024-04-18 by larsrollik

MHRosenberg, 2024-10-22:
@larsrollik, would it be possible to share your container? Is it a docker image? What was your solution?

I have followed all suggestions in the prior issues I found: [#1196](https://github.com/cortex-lab/phy/issues/1196), [#1198](https://github.com/cortex-lab/phy/issues/1198)

ie I have tried both requirements files as well as verified that my versions match that of the new updated conda environment yaml file. For reference, here are my dependencies and my error (same as others, as far as I can tell...)

`conda list
# packages in environment at /home/mr/miniconda3/envs/phy2:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
arrow-cpp                 16.1.0               hc1eb8f0_0  
asttokens                 2.0.5              pyhd3eb1b0_0  
aws-c-auth                0.6.19               h5eee18b_0  
aws-c-cal                 0.5.20               hdbd6064_0  
aws-c-common              0.8.5                h5eee18b_0  
aws-c-compression         0.2.16               h5eee18b_0  
aws-c-event-stream        0.2.15               h6a678d5_0  
aws-c-http                0.6.25               h5eee18b_0  
aws-c-io                  0.13.10              h5eee18b_0  
aws-c-mqtt                0.7.13               h5eee18b_0  
aws-c-s3                  0.1.51               hdbd6064_0  
aws-c-sdkutils            0.1.6                h5eee18b_0  
aws-checksums             0.1.13               h5eee18b_0  
aws-crt-cpp               0.18.16              h6a678d5_0  
aws-sdk-cpp               1.10.55              h721c034_0  
blas                      1.0                         mkl  
bokeh                     3.6.0           py311h06a4308_0  
boost-cpp                 1.82.0               hdb19cb5_2  
bottleneck                1.3.7           py311hf4808d0_0  
brotli                    1.0.9                h5eee18b_8  
brotli-bin                1.0.9                h5eee18b_8  
brotli-python             1.0.9           py311h6a678d5_8  
bzip2                     1.0.8                h5eee18b_6  
c-ares                    1.19.1               h5eee18b_0  
ca-certificates           2024.9.24            h06a4308_0  
certifi                   2024.8.30       py311h06a4308_0  
charset-normalizer        3.3.2              pyhd3eb1b0_0  
click                     8.1.7           py311h06a4308_0  
cloudpickle               3.0.0           py311h06a4308_0  
colorcet                  3.1.0                    pypi_0    pypi
comm                      0.2.1           py311h06a4308_0  
contourpy                 1.2.0           py311hdb19cb5_0  
coverage                  7.6.4                    pypi_0    pypi
coveralls                 4.0.1                    pypi_0    pypi
cycler                    0.11.0             pyhd3eb1b0_0  
cyrus-sasl                2.1.28               h52b45da_1  
cython                    3.0.11          py311h5eee18b_0  
cytoolz                   0.12.2          py311h5eee18b_0  
dask                      2024.8.2        py311h06a4308_0  
dask-core                 2024.8.2        py311h06a4308_0  
dask-expr                 1.1.13          py311h06a4308_0  
dbus                      1.13.18              hb2f20db_0  
debugpy                   1.6.7           py311h6a678d5_0  
decorator                 5.1.1              pyhd3eb1b0_0  
distributed               2024.8.2        py311h06a4308_0  
docopt                    0.6.2                    pypi_0    pypi
executing                 0.8.3              pyhd3eb1b0_0  
expat                     2.6.3                h6a678d5_0  
flake8                    7.1.1                    pypi_0    pypi
fontconfig                2.14.1               h4c34cd2_2  
fonttools                 4.51.0          py311h5eee18b_0  
freetype                  2.12.1               h4a9f257_0  
fsspec                    2024.6.1        py311h06a4308_0  
gflags                    2.2.2                h6a678d5_1  
ghp-import                2.1.0                    pypi_0    pypi
glib                      2.78.4               h6a678d5_0  
glib-tools                2.78.4               h6a678d5_0  
glog                      0.5.0                h6a678d5_1  
[... 219 lines cut ...]
17:33:27.249 [D] gui:718              Add view WaveformView to GUI.
17:33:27.250 [D] base:337             Set state for WaveformView.
17:33:27.255 [D] gui:718              Add view CorrelogramView to GUI.
17:33:27.256 [D] base:337             Set state for CorrelogramView.
17:33:27.266 [D] gui:718              Add view ISIView to GUI.
17:33:27.267 [D] base:337             Set state for ISIView.
17:33:27.272 [D] gui:718              Add view FeatureView to GUI.
17:33:27.273 [D] base:337             Set state for FeatureView.
17:33:27.282 [D] gui:718              Add view AmplitudeView to GUI.
17:33:27.283 [D] base:337             Set state for AmplitudeView.
17:33:27.293 [D] gui:718              Add view FiringRateView to GUI.
17:33:27.293 [D] base:337             Set state for FiringRateView.
17:33:27.294 [W] gui:671              Could not create view TraceView.
17:33:27.327 [D] gui:718              Add view ProbeView to GUI.
17:33:27.327 [D] base:337             Set state for ProbeView.
17:33:27.328 [W] gui:671              Could not create view TemplateFeatureView.
17:33:27.337 [D] gui:718              Add view ClusterScatterView to GUI.
[... cut ...]
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)
(phy2) mr@mr-System-Product-Name:~/Documents/241022_phy_spike_visualization-testing/imec0_ks4$ 

`