cortex-lab/phy #1211: Installation error pyqt5:  metadata-generation-failed

mfvd, 2023-06-16:
I've been trying to install phy for a couple hours now and keep getting the same error (below). I'm following the  installation instructions on the repository. I also tried the previous method `pip install phy --pre --upgrade` but did not manage to install phy.

Any suggestion on how to overcome this issue?

```
Collecting PyQtWebEngine (from phy==2.0b5)
  Using cached PyQtWebEngine-5.15.6.tar.gz (48 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [20 lines of output]
      Traceback (most recent call last):
        File "/opt/anaconda3/envs/phy/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/opt/anaconda3/envs/phy/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/opt/anaconda3/envs/phy/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
        File "/private/var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/pip-build-env-jn5yiz34/overlay/lib/python3.8/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
        File "/private/var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/pip-build-env-jn5yiz34/overlay/lib/python3.8/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/pip-build-env-jn5yiz34/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 586, in setup
          self.apply_user_defaults(tool)
        File "/private/var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/pip-build-env-jn5yiz34/overlay/lib/python3.8/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/pip-build-env-jn5yiz34/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 237, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/1q/x87fsmf571x62kz_721lvzdc0000gn/T/pip-build-env-jn5yiz34/overlay/lib/python3.8/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details
```

mfvd, 2023-06-16:
@rossant thanks! I had not noticed that. I'll give it a try.

neurvanna, 2023-06-20:
Also getting the metadata-generation-failed error. I think I am using the correct Python version.

Collecting git+https://github.com/cortex-lab/phy.git
  Cloning https://github.com/cortex-lab/phy.git to /private/var/folders/sb/_j66yk253xbgj3kxwzk2f55w0000gn/T/pip-req-build-3oj0rihm
  Running command git clone --filter=blob:none --quiet https://github.com/cortex-lab/phy.git /private/var/folders/sb/_j66yk253xbgj3kxwzk2f55w0000gn/T/pip-req-build-3oj0rihm
  Resolved https://github.com/cortex-lab/phy.git to commit 642c875994b44618c3b3157d6e6f4425450813f8
  Preparing metadata (setup.py) ... done
Collecting phylib (from phy==2.0b5)
  Using cached phylib-2.4.3-py2.py3-none-any.whl (80 kB)
Collecting mtscomp (from phy==2.0b5)
  Using cached mtscomp-1.0.2-py2.py3-none-any.whl (16 kB)
Requirement already satisfied: numpy in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (1.24.3)
Requirement already satisfied: matplotlib in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (3.7.1)
Requirement already satisfied: scipy in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (1.10.1)
Requirement already satisfied: h5py in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (3.7.0)
Requirement already satisfied: dask in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (2023.5.1)
Requirement already satisfied: cython in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (0.29.35)
Requirement already satisfied: pillow in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (9.4.0)
Collecting colorcet (from phy==2.0b5)
  Using cached colorcet-3.0.1-py2.py3-none-any.whl (1.7 MB)
Requirement already satisfied: pyopengl in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (3.1.1a1)
Requirement already satisfied: requests in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (2.29.0)
Requirement already satisfied: qtconsole in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (5.4.2)
Collecting tqdm (from phy==2.0b5)
  Using cached tqdm-4.65.0-py3-none-any.whl (77 kB)
Requirement already satisfied: joblib in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (1.2.0)
Requirement already satisfied: click in ./anaconda3/envs/phy2/lib/python3.11/site-packages (from phy==2.0b5) (8.0.4)
Collecting mkdocs (from phy==2.0b5)
  Using cached mkdocs-1.4.3-py3-none-any.whl (3.7 MB)
Collecting PyQtWebEngine (from phy==2.0b5)
  Using cached PyQtWebEngine-5.15.6.tar.gz (48 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [12 lines of output]
      Querying qmake about your Qt installation...
      These bindings will be built: QtWebEngineCore, QtWebEngine, QtWebEngineWidgets.
      Generating the QtWebEngineCore bindings...
      Generating the QtWebEngine bindings...
      Generating the QtWebEngineWidgets bindings...
      Generating the .pro file for the QtWebEngineCore module...
      Generating the .pro file for the QtWebEngine module...
      Generating the .pro file for the QtWebEngineWidgets module...
      Generating the top-level .pro file...
      Generating the Makefiles...
      Compiling the project...
      _in_process.py: 'make' failed returning 2
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

neurvanna, 2023-06-29:
So looks like PyQt5 is incompatible with M1 processor, only PyQt6 is, and phy demands that PyQt5 is installed. Is this something that you could possibly look into @rossant? Thanks very much!

zm711, 2023-07-05:
@neurvanna urvanna It seems like this might require a bit of a work around--not sure if this will work (haven't tried it myself yet). But you could try a Rosetta terminal. Open terminal click on get info. Under general there should be a check box for rosetta. Click that (it will make you install if you don't have it.) Then close terminal. Open a new terminal session (which will now be in rosetta mode). And retry the installation. When done it should be okay to go back into settings and turn off rosetta terminal if you want the native terminal again. This should make it try to install everything Intel style, but once translated (if it works) should then be fine.
Another option would be using `homebrew`, but I'm not sure how well that will play with conda (ie you might need to make a `venv` instead).

zm711, 2024-06-07:
I'll close this for now. A bunch of people haven't been able to get phy working on an M-series mac (I think one person did my reverting to an old version of phy on python 3.7, but I wouldn't really recommend that).

closed 2024-06-07 by zm711

davidkhess, 2025-05-07:
This is long closed issue, but I ran into this exact same error of `_in_process.py: 'make' failed returning 2` at the same point.

For anybody else who finds this issue for the same reason, for my particular circumstance, I solved it by installing the python*-dev package. Make was failing because the python header files couldn't be found.