cortex-lab/phy #1217: Help with Singular Matrix Traceback Error

Ana0902, 2023-08-17:
Hi Everyone!

I am having a bit trouble when trying to open the phy template in spyking-circus. I have tried reinstalling spiking circus, but I am getting a lot of errors. I was wondering if anyone would know why this error is happening and how to solve. Your help would be greatly appreciated. 

Thank you! 

20:34:21.181 [E] __init__:62          An error has occurred (LinAlgError): Singular matrix
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\User\Anaconda3\envs\circus\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
             ^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\apps\template\gui.py", line 218, in template_gui
    gui = controller.create_gui()
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\apps\base.py", line 1697, in create_gui
    self.supervisor.attach(gui)
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\cluster\supervisor.py", line 946, in attach
    self._create_views(
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\cluster\supervisor.py", line 761, in _create_views
    gui, data=self.cluster_info, columns=self.columns, sort=sort)
              ^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\cluster\supervisor.py", line 921, in cluster_info
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\cluster\supervisor.py", line 921, in <listcomp>
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\cluster\supervisor.py", line 746, in get_cluster_info
    out[key] = func(cluster_id)
               ^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\phy\utils\context.py", line 154, in memcached
    out = f(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\.phy\plugins\IsoDStats.py", line 40, in IsoD
    dist=mahal_dist(this_features,all_other_features)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\.phy\plugins\custom_stats_functions.py", line 21, in mahal_dist
    cluster_inv_cov=inv(np.cov(cluster_features_t))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<__array_function__ internals>", line 180, in inv
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\numpy\linalg\linalg.py", line 552, in inv
    ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\Anaconda3\envs\circus\Lib\site-packages\numpy\linalg\linalg.py", line 89, in _raise_linalgerror_singular
    raise LinAlgError("Singular matrix")
numpy.linalg.LinAlgError: Singular matrix

zm711, 2023-08-17:
@Ana0902 

This doesn't appear to be a Phy problem. The error is being called in the plugins for Phy.

> File "C:\Users\User.phy\plugins\IsoDStats.py", line 40, in IsoD
> dist=mahal_dist(this_features,all_other_features)

Users can use their own plugins with Phy. It looks like this plugin is being called from your hard drive and not from the circus environment. Has someone in your lab written a custom plugin for Phy? 

The actual problem you're getting is that the mahalobnis distance requires the calculation of the covariance matrix which cannot occur if the matrix is singular. In this case you should guard for that in your plugin code by catching that error and returning a nan instead of the error. 

Could you paste the script that you find here:  "C:\Users\User.phy\plugins\IsoDStats.py" on your computer. If you think that plugin came from spyking-circus you should probably open the issue over there: https://github.com/spyking-circus/spyking-circus, but it really looks like someone else added the plugin to your computer and not from the creators of the programs.

Hope that helps :)

Ana0902, 2023-08-22:
Thank you so much! It helped a lot.  I was able to solve the issue!

closed 2023-08-22 by Ana0902