cortex-lab/phy #1219: WaveformView won't show up

Antho2422, 2023-09-04:
Hi ! 

I'm working on a project were I'm performing spike sorting using SpikeInterface (SpikingCircus2 as sorter) and I'm trying to use Phy on my data. I made it work fine a few weeks ago but now, my data is loading but the WaveformView panel don't show the traces. 

I though this was coming from some environment updates, so I tried to reinstall phy from this environment : https://github.com/cortex-lab/phy/blob/master/environment.yml

Unfortunately it did not fix the issue. 

This is the error I get :
```python
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\gui\qt.py", line 547, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\cluster\views\base.py", line 175, in _worker  
    self.on_select(cluster_ids=cluster_ids, **kwargs)
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\cluster\views\base.py", line 143, in on_select
    self.plot(**kwargs)
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\cluster\views\waveform.py", line 311, in plot 
    bunchs = self.get_clusters_data()
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\cluster\views\waveform.py", line 189, in get_clusters_data
    bunchs = [
             ^
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\cluster\views\waveform.py", line 190, in <listcomp>
    self.waveforms_types.get()(cluster_id) for cluster_id in self.cluster_ids]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\utils\context.py", line 154, in memcached
    out = f(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^
  File "C:\Users\user.name\Anaconda3\envs\phy2\Lib\site-packages\phy\apps\base.py", line 562, in _get_template_waveforms
    data[i][:, b.channel_ids] = b.template
``` 

And this is what the gui is showing up : 

[image]

Any help would be appreciated 😃 

Thanks, 
Anthony

Antho2422, 2023-09-05:
I also get the following error when I click on a cluster : 

```python
C:\Users\user.name\AppData\Local\pypoetry\Cache\virtualenvs\priam-5spsvreu-py3.9\lib\site-packages\joblib\_store_backends.py:207: CacheWarning: Unable to cache to disk. Possibly a race condition in the creation of the directory. Exception: [Errno 2] No such file or directory: '\\\\l2export\\iss02.xxx\\analyses\\user.name\\git_lustre\\priam\\data\\processed\\model_name\\03231\\preprocessed\\phy\\1\\.phy\\phy\\apps\\base\\TemplateMixin\\get_spike_template_amplitudes\\3cb821db3a3a89ad88d836c6936f64ae\\output.pkl.thread-1796794117808-pid-11172'.
  warnings.warn(
```

zm711, 2023-09-05:
@Antho2422 

I think you cutoff the error in the trace. Could you post the full error traceback?  I see the lines where the error occurred, but not the final error. It would be helpful to figure out if this is a spikeinterface problem or a phy problem.

Antho2422, 2023-09-06:
Thanks for your answer @zm711 . Sorry for cutting the error I didn't notice! 

Here is the last line : 
```python
AttributeError: 'NoneType' object has no attribute 'template'
``` 

I'm afraid that is the only line I forgot...

Here is also a preview of the content of the phy folder. The spike_templates.npy file is here. 
[image]

There is also the log file in case you need it : 
[phy.log](https://github.com/cortex-lab/phy/files/12536081/phy.log)

Cheers :)

zm711, 2023-09-06:
@Antho2422 

Could you run with `--debug`?

so : `phy template-gui params.py --debug`

So we can see all the debug info as well. I think I have an idea of what is going on. I may need you to share the `templates.npy` file after the debugging.

Best,
Zach

zm711, 2023-09-07:
@Antho2422 

could I get template_ind.npy too? Thanks Anthony!

Zach

zm711, 2023-09-07:
If the file is found the debugger should print whether the templates are sparse or dense:
https://github.com/cortex-lab/phylib/blob/2b567cd2e2c382bd5e8926e9abd9a2f9353bfd76/phylib/io/model.py#L720-L724

But your debug does not say this likely indicating that their was an IOError exit before hand.

zm711, 2023-09-07:
What is the `dat_path` specified in your `params.py`. And where is the folder containing all the files (C:\users\ap\documents or something like that?). Ideally we want to see the exact io error occurring to try to figure out which function in particular is failing. Basically we have the issue narrowed down to between 2 functions either the `_find_path` or the `_read_array`.

Antho2422, 2023-09-11:
The dat_path is the following : 
```python
dat_path = r'/network/lustre/iss02/charpier/analyses/anthony.pinto/git_lustre/priam/data/processed/golgi_v1.0/02711_2019-04-18_11-04/preprocessed/phy/1/recording.dat'
``` 

The folder containing all the files is at the same place I guess, are you talking about the phy folder ? The path is on a lustre server but I already tried it in local and I got the same issue.

zm711, 2023-09-11:
@Antho2422 

So there is an issue with np.load where if the name of a file is too long it can cause an `IOError`. Your file path is 152 characters which is a pretty long name (as far as Python/NumPy is concerned). Could you try to greatly shorten your file path as a test and rerun locally testing something like C:\users\name\data\recording.dat. To see if that could be causing it.

Antho2422, 2023-09-11:
Okay I will try that asap and keep you updated thank you !

zm711, 2023-09-25:
@Antho2422 

Just wanted to see if that worked. The other option is that there are some Path issues on the SpikeInterface side. So I can propagate that up the chain if it's not a python file name length issue.

Antho2422, 2023-09-26:
Hi @zm711,

Sorry, I was on holidays and I didn’t have time to test it yet ! I will do it asap and keep you updated of course ! 

Thank you !

Antho2422, 2023-09-28:
Hi @zm711 !

Sorry for the wait ! I tried with a shorter path and it worked ! Is there any way to avoid this error even when the paths are long ? 

Thank you !

zm711, 2023-09-28:
@Antho2422 

That's great!

But unfortunately no... I actually ran into this issue in a different program I was running so I opened issues myself and was told that it is just a computer+python thing. So I just empirically determined the allowable length of file and made sure not to cross that limit. (I want to say it was either 100 or 150 total characters including slashes--for my computer), but you would probably need to double check with yours and then just stay within that limit.

Happy to help!

Antho2422, 2023-09-28:
Okay ! Thank you so much for the help !

Cheers, 
Anthony

closed 2023-09-28 by Antho2422