cortex-lab/phy #1224: Error loading internal MIME data

martuser, 2023-10-19:
Hi!
I've checked issue 1212 and 1218 but I still have the same error. 
I've followed the exact instructions as its described in README.
My OS is Windows 11 Pro for Workstations and conda version 4.10.3.

Thanks,
Marta.

martuser, 2023-10-19:
I've uninstalled Anaconda and installed the latest version to check if that could be the issue. Now, with the Anaconda3 version 2023.07-2  and conda version 23.7.2 , when following the README instructions, I still encounter the same error. The problem is that even though the error appears, it does display the GUI. However, when clicking on any cluster, the GUI crashes.

martuser, 2023-11-05:
Thanks, Zach, for responding. Here is the information from 'conda list':
```# packages in environment at C:\ProgramData\anaconda3:
#
# Name                    Version                   Build  Channel
_anaconda_depends         2023.07                 py311_1
abseil-cpp                20211102.0           hd77b12b_0
aiobotocore               2.4.2           py311haa95532_0
aiofiles                  22.1.0          py311haa95532_0
aiohttp                   3.8.3           py311h2bbff1b_0
aioitertools              0.7.1              pyhd3eb1b0_0
aiosignal                 1.2.0              pyhd3eb1b0_0
aiosqlite                 0.18.0          py311haa95532_0
alabaster                 0.7.12             pyhd3eb1b0_0
anaconda-catalogs         0.2.0           py311haa95532_0
anaconda-client           1.12.0          py311haa95532_0
anaconda-navigator        2.4.2           py311haa95532_0
anaconda-project          0.11.1          py311haa95532_0
anyio                     3.5.0           py311haa95532_0
appdirs                   1.4.4              pyhd3eb1b0_0
argon2-cffi               21.3.0             pyhd3eb1b0_0
argon2-cffi-bindings      21.2.0          py311h2bbff1b_0
arrow                     1.2.3           py311haa95532_1
arrow-cpp                 11.0.0          py311h308b814_0
astroid                   2.14.2          py311haa95532_0
astropy                   5.1             py311h5bb9823_0
asttokens                 2.0.5              pyhd3eb1b0_0
async-timeout             4.0.2           py311haa95532_0
atomicwrites              1.4.0                      py_0
attrs                     22.1.0          py311haa95532_0
automat                   20.2.0                     py_0
autopep8                  1.6.0              pyhd3eb1b0_1
aws-c-common              0.4.57               ha925a31_1
aws-c-event-stream        0.1.6                hd77b12b_5
aws-checksums             0.1.9                ha925a31_0
aws-sdk-cpp               1.8.185              hd77b12b_0
babel                     2.11.0          py311haa95532_0
backcall                  0.2.0              pyhd3eb1b0_0
backports                 1.1                pyhd3eb1b0_0
backports.functools_lru_cache 1.6.4              pyhd3eb1b0_0
backports.tempfile        1.0                pyhd3eb1b0_1
backports.weakref         1.0.post1                  py_1
bcrypt                    3.2.0           py311h2bbff1b_1
beautifulsoup4            4.12.2          py311haa95532_0
binaryornot               0.4.4              pyhd3eb1b0_1
black                     23.3.0          py311haa95532_0
blas                      1.0                         mkl
bleach                    4.1.0              pyhd3eb1b0_0
blosc                     1.21.3               h6c2663c_0
bokeh                     3.2.1           py311h746a85d_0
boltons                   23.0.0          py311haa95532_0
boost-cpp                 1.73.0              h2bbff1b_12
boto3                     1.24.28         py311haa95532_0
botocore                  1.27.59         py311haa95532_0
bottleneck                1.3.5           py311h5bb9823_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py311h2bbff1b_1002
bzip2                     1.0.8                he774522_0
c-ares                    1.19.0               h2bbff1b_0
c-blosc2                  2.8.0                hd77b12b_0
ca-certificates           2023.05.30           haa95532_0
certifi                   2023.7.22       py311haa95532_0
cffi                      1.15.1          py311h2bbff1b_3
cfitsio                   3.470                h2bbff1b_7
chardet                   4.0.0           py311haa95532_1003
charls                    2.2.0                h6c2663c_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4           py311haa95532_0
cloudpickle               2.2.1           py311haa95532_0
clyent                    1.2.2           py311haa95532_1
[... 386 lines cut ...]
wheel                     0.38.4          py311haa95532_0
widgetsnbextension        4.0.5           py311haa95532_0
win_inet_pton             1.1.0           py311haa95532_0
winpty                    0.4.3                         4
wrapt                     1.14.1          py311h2bbff1b_0
xarray                    2023.6.0        py311haa95532_0
xlwings                   0.29.1          py311haa95532_0
xyzservices               2022.9.0        py311haa95532_1
xz                        5.4.2                h8cc25b3_0
y-py                      0.5.9           py311hb6bf4ef_0
yaml                      0.2.5                he774522_0
yaml-cpp                  0.7.0                hd77b12b_1
yapf                      0.31.0             pyhd3eb1b0_0
yarl                      1.8.1           py311h2bbff1b_0
ypy-websocket             0.8.2           py311haa95532_0
zeromq                    4.3.4                hd77b12b_0
zfp                       0.5.5                hd77b12b_6
zict                      2.2.0           py311haa95532_0
zipp                      3.11.0          py311haa95532_0
zlib                      1.2.13               h8cc25b3_0
zlib-ng                   2.0.7                h2bbff1b_0
zope                      1.0             py311haa95532_1
zope.interface            5.4.0           py311h2bbff1b_0
zstandard                 0.19.0          py311h2bbff1b_0
zstd                      1.5.5                hd43e919_0´´´

zm711, 2023-11-05:
@martuser,

I don't see phy. Is this your `(base)` or your `(phy2)` env? 

Could you do the same inside of whatever env has `phy` installed?

martuser, 2023-11-05:
Sorry, it was in base, here is un the env where phy is installed:
```# packages in environment at C:\Users\isaco\.conda\envs\phy2dev:
#
# Name                    Version                   Build  Channel
abseil-cpp                20211102.0           hd77b12b_0
arrow-cpp                 11.0.0               ha81ea56_2
asttokens                 2.0.5              pyhd3eb1b0_0
aws-c-common              0.6.8                h2bbff1b_1
aws-c-event-stream        0.1.6                hd77b12b_6
aws-checksums             0.1.11               h2bbff1b_2
aws-sdk-cpp               1.8.185              hd77b12b_1
backcall                  0.2.0              pyhd3eb1b0_0
blas                      1.0                         mkl
bokeh                     3.3.0           py311h746a85d_0
boost-cpp                 1.82.0               h59b6b97_2
bottleneck                1.3.5           py311h5bb9823_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py311h2bbff1b_1002
bzip2                     1.0.8                he774522_0
c-ares                    1.19.1               h2bbff1b_0
ca-certificates           2023.08.22           haa95532_0
certifi                   2023.7.22       py311haa95532_0
cffi                      1.15.1          py311h2bbff1b_3
charset-normalizer        2.0.4              pyhd3eb1b0_0
click                     8.0.4           py311haa95532_0
cloudpickle               2.2.1           py311haa95532_0
colorama                  0.4.6           py311haa95532_0
colorcet                  3.0.1                    pypi_0    pypi
comm                      0.1.2           py311haa95532_0
contourpy                 1.0.5           py311h59b6b97_0
cryptography              41.0.3          py311h89fc84f_0
cycler                    0.11.0             pyhd3eb1b0_0
cython                    3.0.0           py311h2bbff1b_0
cytoolz                   0.12.0          py311h2bbff1b_0
dask                      2023.6.0        py311haa95532_0
dask-core                 2023.6.0        py311haa95532_0
debugpy                   1.6.7           py311hd77b12b_0
decorator                 5.1.1              pyhd3eb1b0_0
distributed               2023.6.0        py311haa95532_0
executing                 0.8.3              pyhd3eb1b0_0
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.12.1               ha860e81_0
fsspec                    2023.9.2        py311haa95532_0
gflags                    2.2.2                ha925a31_0
ghp-import                2.1.0                    pypi_0    pypi
giflib                    5.2.1                h8cc25b3_3
git                       2.40.1               haa95532_1
glib                      2.69.1               h5dc1a3c_2
glog                      0.5.0                hd77b12b_0
grpc-cpp                  1.48.2               hfe90ff0_1
h5py                      3.9.0           py311h4e0e482_0
hdf5                      1.12.1               h51c971a_3
heapdict                  1.0.1              pyhd3eb1b0_0
icc_rt                    2022.1.0             h6049295_2
icu                       58.2                 ha925a31_3
idna                      3.4             py311haa95532_0
importlib-metadata        6.0.0           py311haa95532_0
iniconfig                 1.1.1              pyhd3eb1b0_0
intel-openmp              2023.1.0         h59b6b97_46319
ipykernel                 6.25.0          py311h746a85d_0
ipython                   8.15.0          py311haa95532_0
ipython_genutils          0.2.0              pyhd3eb1b0_1
jedi                      0.18.1          py311haa95532_1
jinja2                    3.1.2           py311haa95532_0
joblib                    1.2.0           py311haa95532_0
jpeg                      9e                   h2bbff1b_1
jupyter_client            8.1.0           py311haa95532_0
jupyter_core              5.3.0           py311haa95532_0
kiwisolver                1.4.4           py311hd77b12b_0
[... 105 lines cut ...]
tblib                     1.7.0              pyhd3eb1b0_0
threadpoolctl             2.2.0              pyh0d69192_0
tk                        8.6.12               h2bbff1b_0
toml                      0.10.2             pyhd3eb1b0_0
toolz                     0.12.0          py311haa95532_0
tornado                   6.3.3           py311h2bbff1b_0
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.7.1           py311haa95532_0
tzdata                    2023c                h04d1e81_0
urllib3                   1.26.16         py311haa95532_0
utf8proc                  2.6.1                h2bbff1b_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
watchdog                  3.0.0                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0
wheel                     0.41.2          py311haa95532_0
win_inet_pton             1.1.0           py311haa95532_0
xyzservices               2022.9.0        py311haa95532_1
xz                        5.4.2                h8cc25b3_0
yaml                      0.2.5                he774522_0
zeromq                    4.3.4                hd77b12b_0
zict                      3.0.0           py311haa95532_0
zipp                      3.11.0          py311haa95532_0
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.5                hd43e919_0´´´

zm711, 2023-11-05:
cool thanks Marta.

One more thing. Could you copy the exact error that you are getting from the terminal/cmd prompt/? It would help if you run with debug so:

```bash
phy template-gui params.py --debug
```

So I can see the exact place where the error is occurring.

As long as you tested installing with the yaml as in #1218, then I need to see the specific error. If you haven't tried installing from the yaml, I would start with that.

martuser, 2023-11-05:
This is the error using the environment that I currently have:

``` (phy2dev) F:\NP\LUX#75_NP-1_g0\kilosort2.5>phy template-gui params.py --debug
19:12:01.169 [D] __init__:68          Start capturing exceptions.
19:12:01.616 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
19:12:01.932 [D] model:619            Loading spike clusters.
19:12:02.272 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
19:12:02.592 [D] model:569            No channel shank file found.
19:12:02.593 [D] model:692            Loading templates.
19:12:02.676 [D] model:724            Templates are dense.
19:12:07.260 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
19:12:07.260 [D] model:730            Loading the whitening matrix.
19:12:07.264 [D] model:737            Loading the inverse of the whitening matrix.
19:12:07.284 [D] model:766            Loading features.
19:12:07.287 [D] model:781            Features are sparse.
19:12:07.288 [D] model:803            Loading template features.
19:12:07.290 [D] model:813            Template features are sparse.
19:12:07.292 [D] model:504            Load `cluster_Amplitude.tsv`.
19:12:07.298 [D] model:504            Load `cluster_ContamPct.tsv`.
19:12:07.394 [D] model:504            Load `cluster_group.tsv`.
19:12:07.395 [D] model:504            Load `cluster_KSLabel.tsv`.
19:12:08.062 [D] context:100          Initialize joblib cache dir at `F:\NP\LUX#75_NP-1_g0\kilosort2.5\.phy`.
19:12:08.062 [D] context:101          Reducing the size of the cache if needed.
19:12:08.194 [D] base:102             Add filter `high_pass`.
19:12:08.195 [D] config:31            Load config file `C:\Users\isaco\.phy\phy_config.py`.
19:12:08.196 [D] plugin:145           Loading 0 plugins.
19:12:08.196 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
19:12:08.223 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
19:12:08.248 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
19:12:08.250 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
19:12:08.252 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
19:12:08.266 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
19:12:08.282 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
19:12:08.296 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
19:12:08.299 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
19:12:08.302 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
19:12:08.318 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
19:12:08.338 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
19:12:08.341 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
19:12:08.343 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
19:12:08.609 [D] context:185          Save data to `F:\NP\LUX#75_NP-1_g0\kilosort2.5\.phy\spikes_per_cluster.pkl`.
19:12:10.037 [D] gui:463              Creating GUI.
19:12:10.052 [D] state:46             Load C:\Users\isaco\.phy\TemplateGUI\state.json for GUIState.
19:12:10.054 [D] state:46             Load F:\NP\LUX#75_NP-1_g0\kilosort2.5\.phy\state.json for GUIState.
19:12:10.612 [D] gui:718              Add view ClusterView to GUI.
19:12:10.624 [D] gui:718              Add view SimilarityView to GUI.
19:12:10.670 [D] gui:718              Add view WaveformView to GUI.
19:12:10.675 [D] base:337             Set state for WaveformView.
19:12:10.689 [D] gui:718              Add view CorrelogramView to GUI.
19:12:10.692 [D] base:337             Set state for CorrelogramView.
19:12:10.709 [D] gui:718              Add view FeatureView to GUI.
19:12:10.711 [D] base:337             Set state for FeatureView.
19:12:10.735 [D] gui:718              Add view AmplitudeView to GUI.
19:12:10.738 [D] base:337             Set state for AmplitudeView.
19:12:10.761 [D] gui:718              Add view FiringRateView to GUI.
19:12:10.764 [D] base:337             Set state for FiringRateView.
19:12:11.556 [D] gui:718              Add view TraceView to GUI.
19:12:11.559 [D] base:337             Set state for TraceView.
19:12:11.560 [D] base:494             Set color scheme to random.
19:12:18.564 [D] base:1626            Adding default color schemes to TraceView.
19:12:22.318 [D] gui:718              Add view TemplateFeatureView to GUI.
19:12:22.320 [D] base:337             Set state for TemplateFeatureView.
19:12:22.541 [D] gui:519              Load the geometry state.
QMimeDatabase: Error loading internal MIME data
An error has been encountered at line 1 of <internal MIME data>: Premature end of document.:
19:12:23.936 [D] supervisor:817       Clusters selected: [180] (315) ´´´

And this is the error using the environment from the yaml file in https://github.com/cortex-lab/phy/issues/1218:
``` (phy2) F:\NP\LUX#75_NP-1_g0\kilosort2.5>phy template-gui params.py --debug
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\isaco\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 4, in <module>
  File "C:\Users\isaco\.conda\envs\phy2\Lib\site-packages\phy\__init__.py", line 21, in <module>
    from .utils.config import load_master_config
  File "C:\Users\isaco\.conda\envs\phy2\Lib\site-packages\phy\utils\__init__.py", line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File "C:\Users\isaco\.conda\envs\phy2\Lib\site-packages\phy\utils\plugin.py", line 14, in <module>
    import imp
ModuleNotFoundError: No module named 'imp' ´´´

martuser, 2023-11-05:
And this is the error using the environment from the yaml file in https://github.com/cortex-lab/phy/issues/1218:
``` (phy2) F:\NP\LUX#75_NP-1_g0\kilosort2.5>phy template-gui params.py --debug
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\isaco\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 4, in <module>
  File "C:\Users\isaco\.conda\envs\phy2\Lib\site-packages\phy\__init__.py", line 21, in <module>
    from .utils.config import load_master_config
  File "C:\Users\isaco\.conda\envs\phy2\Lib\site-packages\phy\utils\__init__.py", line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File "C:\Users\isaco\.conda\envs\phy2\Lib\site-packages\phy\utils\plugin.py", line 14, in <module>
    import imp
ModuleNotFoundError: No module named 'imp' ´´´

zm711, 2023-11-05:
Cool. That we can fix. So the issue is Python 3.12 just came out where they removed `imp`. So what you need to do is edit the yaml and say python=3.11 instead. Then try again with the yaml. I have a PR open to fix this issue, but I'm waiting for it to be merged. Can you try that and let me know if that works? (Or let me know if you want more explanation on ending the yaml.)

martuser, 2023-11-05:
That's it! Thank you very much, Zach. I no longer have any errors, and the GUI is working! Here's what appears in debug mode:
``` 19:56:11.127 [D] __init__:68          Start capturing exceptions.
19:56:11.602 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
19:56:11.924 [D] model:619            Loading spike clusters.
19:56:12.310 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
19:56:12.641 [D] model:569            No channel shank file found.
19:56:12.644 [D] model:692            Loading templates.
19:56:12.743 [D] model:724            Templates are dense.
19:56:17.356 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
19:56:17.356 [D] model:730            Loading the whitening matrix.
19:56:17.358 [D] model:737            Loading the inverse of the whitening matrix.
19:56:17.365 [D] model:766            Loading features.
19:56:17.366 [D] model:781            Features are sparse.
19:56:17.367 [D] model:803            Loading template features.
19:56:17.371 [D] model:813            Template features are sparse.
19:56:17.374 [D] model:504            Load `cluster_Amplitude.tsv`.
19:56:17.385 [D] model:504            Load `cluster_ContamPct.tsv`.
19:56:17.395 [D] model:504            Load `cluster_group.tsv`.
19:56:17.396 [D] model:504            Load `cluster_KSLabel.tsv`.
19:56:17.909 [D] context:100          Initialize joblib cache dir at `F:\NP\LUX#75_NP-1_g0\kilosort2.5\.phy`.
19:56:17.909 [D] context:101          Reducing the size of the cache if needed.
19:56:18.048 [D] base:102             Add filter `high_pass`.
19:56:18.050 [D] config:31            Load config file `C:\Users\isaco\.phy\phy_config.py`.
19:56:18.050 [D] plugin:145           Loading 0 plugins.
19:56:18.051 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
19:56:18.099 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
19:56:18.111 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
19:56:18.112 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
19:56:18.112 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
19:56:18.113 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
19:56:18.114 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
19:56:18.115 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
19:56:18.116 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
19:56:18.116 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
19:56:18.117 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
19:56:18.120 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
19:56:18.120 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
19:56:18.121 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
19:56:18.695 [D] context:185          Save data to `F:\NP\LUX#75_NP-1_g0\kilosort2.5\.phy\spikes_per_cluster.pkl`.
19:56:19.668 [D] gui:463              Creating GUI.
19:56:19.680 [D] state:46             Load C:\Users\isaco\.phy\TemplateGUI\state.json for GUIState.
19:56:19.681 [D] state:46             Load F:\NP\LUX#75_NP-1_g0\kilosort2.5\.phy\state.json for GUIState.
19:56:20.168 [D] gui:718              Add view ClusterView to GUI.
19:56:20.189 [D] gui:718              Add view SimilarityView to GUI.
19:56:20.233 [D] gui:718              Add view WaveformView to GUI.
19:56:20.563 [D] base:337             Set state for WaveformView.
19:56:20.576 [D] gui:718              Add view CorrelogramView to GUI.
19:56:20.579 [D] base:337             Set state for CorrelogramView.
19:56:20.592 [D] gui:718              Add view FeatureView to GUI.
19:56:20.594 [D] base:337             Set state for FeatureView.
19:56:20.618 [D] gui:718              Add view AmplitudeView to GUI.
19:56:20.620 [D] base:337             Set state for AmplitudeView.
19:56:20.644 [D] gui:718              Add view FiringRateView to GUI.
19:56:20.646 [D] base:337             Set state for FiringRateView.
19:56:21.526 [D] gui:718              Add view TraceView to GUI.
19:56:21.529 [D] base:337             Set state for TraceView.
19:56:21.529 [D] base:494             Set color scheme to random.
19:56:28.463 [D] base:1626            Adding default color schemes to TraceView.
19:56:32.099 [D] gui:718              Add view TemplateFeatureView to GUI.
19:56:32.101 [D] base:337             Set state for TemplateFeatureView.
19:56:32.349 [D] gui:519              Load the geometry state.
19:56:33.711 [D] supervisor:817       Clusters selected: [180] (315)
19:58:50.172 [D] supervisor:817       Clusters selected: [41] (42) ´´´

zm711, 2023-11-05:
Happy it works. Best of luck Marta!

Cheers,
Zach

closed 2024-06-03 by zm711