cortex-lab/phy #1225: Couldn't open phy2 file from other device

Doomsday0910, 2023-10-20:
Hello!

I was trying to open phy2 on two different windows desktops (on a portable SSD). 
However, analyzed results from one computer wouldn't open in the other one, and I got error message as such. (The other way worked) Could anyone help figure out what's going wrong?
Thanks in advance!

[image]

zm711, 2023-11-01:
@Doomsday0910,

just to be clear. Phy still worked on the original computer with the ssd. I'm assuming some sort of absolute path got saved into one of the files causing Phy to only work on the computer that originally analyzed the data.

Doomsday0910, 2023-11-03:
@zm711 Thanks for your reply. Yes, it works on the original (1st) computer. However, data analyzed on the 2nd computer can be opened on the 1st computer.
Do you have any suggestions on sharing data to be worked on different computers?

zm711, 2023-11-03:
@Doomsday0910 

Couple more questions. Are both computers windows? Or are trying to move between computers?
And could you copy and paste whatever it says in the `dat_path` section of the `params.py` file here so I can see if that is part of the problem.

RunjieTomYi, 2023-12-08:
@zm711

Both computers are windows 10.

In params.py I have
dat_path = '../Concat.dat'
[image]

zm711, 2023-12-08:
And what spike sorter are you using to generate this params.py? For KS it usually doesn't save the dir_path, but lets Phy generate the dir_path itself (also how was Concat.dat generated -- you wrote the file itself or you had another program write it ?)

RunjieTomYi, 2023-12-08:
I was using Kilosort to generate this data, and my Concat.data comes from multiple consecutive recordings merged with [customized code](https://github.com/buzsakilab/buzcode/blob/0afa591e2b2e6b87d0e317d80225645d2e9e13c7/preprocessing/bz_ConcatenateDats.m#L4) in MATLAB.

zm711, 2023-12-08:
could you try one thing for me? Could you edit the the dat_path to be

```
dat_path = './Concat.dat'
```

instead. Normally `./` means search in my current relative directory whereas `../` means go up one folder and search the folder above. I'm curious if that is contributing.

Then if that doesn't work could you try:

```
dat_path =  'Concat.dat'
```

as well. I'm just seeing if there if a path based issue. 

finally if that doesn't work could you try just deleting the dir_path line for now.

RunjieTomYi, 2023-12-09:
I tried changing dat_path and deleting dir_path, they still returned the same error.
Testing path search returns True.
The file Concat.dat is stored in the folder above, where Kilosort output folder is stored.
[image]