cortex-lab/phy #1226: ImportError on macOS: library not loaded: @rpath/QtCore.framework/Versions/5/QtCore

anna-teruel, 2023-10-22:
Hey everyone!

I'm trying to install phy on my macOS M1 Sonoma 14.0, but I fail to run phy template-gui params.py. I am aware that phy is not yet supported for arm-64 architecture. I created an environment using miniconda macOS-14.0-x86_64-i386-64bit, and installed phy inside that environment, following the instructions provided on the Readme file. The error I get is the following: 

`(phy2) annateruel@MacBook-MacBook-Pro-de-Anna PyQt5 % phy template-gui params.py
Traceback (most recent call last):
  File "/Users/annateruel/minicondax86/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/phy/gui/qt.py", line 33, in <module>
    from PyQt5.QtCore import (Qt, QByteArray, QMetaObject, QObject,  # noqa
ImportError: dlopen(/Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so, 0x0002): Library not loaded: @rpath/QtCore.framework/Versions/5/QtCore
  Referenced from: <1164C5D2-6562-35CA-9546-488193326D7A> /Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so
  Reason: tried: '/Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/annateruel/minicondax86/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/annateruel/minicondax86/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/annateruel/minicondax86/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Library/Frameworks/QtCore.framework/Versions/5/QtCore' (no such file), '/System/Library/Frameworks/QtCore.framework/Versions/5/QtCore' (no such file, not in dyld cache)`

Indeed, the error is in PyQt5.QtCore, because somehow it doesn't find some paths. I have tried to reach those paths and I get a permission error. Only can enter if sudo -s. But still, some of the paths mentioned don't exist. Thus, I don't know what to do. Guess I am missing the installation of some packages. 

In my environment here's the list of packages installed and their corresponding version, just in case it is helpful. 
Package                       Version
----------------------------- ------------
appnope                       0.1.3
asttokens                     2.4.0
backcall                      0.2.0
backports.functools-lru-cache 1.6.5
bokeh                         3.3.0
Brotli                        1.1.0
cached-property               1.5.2
certifi                       2023.7.22
charset-normalizer            3.3.0
click                         8.1.7
cloudpickle                   3.0.0
colorama                      0.4.6
colorcet                      3.0.1
comm                          0.1.4
contourpy                     1.1.1
cycler                        0.12.1
Cython                        3.0.4
cytoolz                       0.12.2
dask                          2023.10.0
debugpy                       1.8.0
decorator                     5.1.1
distributed                   2023.10.0
exceptiongroup                1.1.3
executing                     1.2.0
fonttools                     4.43.1
fsspec                        2023.9.2
ghp-import                    2.1.0
h5py                          3.10.0
idna                          3.4
importlib-metadata            6.8.0
iniconfig                     2.0.0
ipykernel                     6.25.2
ipython                       8.16.1
ipython-genutils              0.2.0
jedi                          0.19.1
Jinja2                        3.1.2
joblib                        1.3.2
jupyter_client                8.4.0
jupyter_core                  5.4.0
kiwisolver                    1.4.5
klusta                        3.0.16
klustakwik2                   0.2.6
locket                        1.0.0
lz4                           4.3.2
Markdown                      3.5
MarkupSafe                    2.1.3
matplotlib                    3.8.0
matplotlib-inline             0.1.6
mergedeep                     1.3.4
mkdocs                        1.5.3
msgpack                       1.0.6
mtscomp                       1.0.2
munkres                       1.1.4
nest-asyncio                  1.5.8
numpy                         1.26.0
packaging                     23.2
pandas                        2.1.1
param                         1.13.0
parso                         0.8.3
partd                         1.4.1
pathspec                      0.11.2
pexpect                       4.8.0
phy                           2.0b5
phylib                        2.4.3
pickleshare                   0.7.5
Pillow                        9.4.0
pip                           23.3
platformdirs                  3.11.0
pluggy                        1.3.0
ply                           3.11
prompt-toolkit                3.0.39
psutil                        5.9.5
ptyprocess                    0.7.0
pure-eval                     0.2.2
pyarrow                       13.0.0
pyct                          0.5.0
Pygments                      2.16.1
PyOpenGL                      3.1.6
pyparsing                     3.1.1
PyQt5                         5.15.10
PyQt5-Qt5                     5.15.11
PyQt5-sip                     12.13.0
PyQtWebEngine                 5.15.6
PyQtWebEngine-Qt5             5.15.2
PySocks                       1.7.1
pytest                        7.4.2
python-dateutil               2.8.2
pytz                          2023.3.post1
PyYAML                        6.0.1
pyyaml_env_tag                0.1
pyzmq                         25.1.1
qtconsole                     5.4.4
QtPy                          2.4.0
requests                      2.31.0
responses                     0.23.2
scikit-learn                  1.3.1
scipy                         1.11.3
setuptools                    68.2.2
sip                           6.6.2
six                           1.16.0
sortedcontainers              2.4.0
stack-data                    0.6.2
tblib                         2.0.0
threadpoolctl                 3.2.0
toml                          0.10.2
tomli                         2.0.1
toolz                         0.12.0
tornado                       6.3.3
tqdm                          4.66.1
traitlets                     5.11.2
types-PyYAML                  6.0.12.12
typing_extensions             4.8.0
tzdata                        2023.3
urllib3                       2.0.7
watchdog                      3.0.0
wcwidth                       0.2.8
wheel                         0.41.2
xyzservices                   2023.10.0
zict                          3.0.0
zipp                          3.17.0

Any help would be more than welcome. I've tried to install phy on different macOs with intel architecture and the error I get is the same. Then I guess is not related to my macOS M1. If you need any further information I'll provide it. 

Thanks in advance!

zm711, 2023-11-01:
@anna-teruel, the instructions have been updated. Not sure if this will fix the M architecture issue, but the new installation instructions do not involve installing phy from pypi. Have you tried:

```bash
conda env create -f environment.yml
conda activate phy2
pip install git+https://github.com/cortex-lab/phy.git
```

This should make sure that all the packages are the correct up-to-date packages. If you have already done this, then this could be a mac permission issue, but you'd probably have to search around on PyQt5 to see how they suggestion dealing with M* chip Macs.

anna-teruel, 2023-11-02:
Thanks @zm711 !!! 👍 

It worked now, more or less. So with new config file I can successfully open the GUI. However everything looks blank (please find picture attached). So I cannot see the amplitude viewer, correlogram etc. 

Any idea what is missing? 
Thanks for your time!
[image]

zm711, 2023-11-02:
@anna-teruel,

Yeah that change just happened unfortunately (Python 3.12). I'll see if I can fix the `imp` for future users, and if not I will update the yaml.

Could you run
```bash
phy template-gui params.py --debug #edit for your file path for params.py
```

Then copy the output here. That will say what stage of the gui generation is failing so we can try to troubleshoot what is going wrong.

anna-teruel, 2023-11-03:
Thanks for the quick response @zm711 

The output I get is the following (sorry, it's long): 
 
`(phy2) annateruel@MacBook-MacBook-Pro-de-Anna registro-5 % phy template-gui params.py --debug
09:55:23.488 [D] __init__:68          Start capturing exceptions.
09:55:23.521 [D] model:619            Loading spike clusters.
09:55:23.535 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
09:55:23.549 [D] model:569            No channel shank file found.
09:55:23.549 [D] model:692            Loading templates.
09:55:23.560 [D] model:724            Templates are dense.
09:55:23.588 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
09:55:23.588 [D] model:730            Loading the whitening matrix.
09:55:23.589 [D] model:737            Loading the inverse of the whitening matrix.
09:55:23.594 [D] model:766            Loading features.
09:55:23.594 [D] model:781            Features are sparse.
09:55:23.595 [D] model:803            Loading template features.
09:55:23.595 [D] model:813            Template features are sparse.
09:55:23.596 [D] model:504            Load `cluster_Amplitude.tsv`.
09:55:23.597 [D] model:504            Load `cluster_group.tsv`.
09:55:23.662 [D] model:504            Load `cluster_ContamPct.tsv`.
09:55:23.664 [D] model:504            Load `cluster_KSLabel.tsv`.
09:55:24.472 [D] context:100          Initialize joblib cache dir at `/Users/annateruel/Desktop/registro-5/.phy`.
09:55:24.472 [D] context:101          Reducing the size of the cache if needed.
09:55:24.488 [D] base:102             Add filter `high_pass`.
09:55:24.489 [D] config:31            Load config file `/Users/annateruel/.phy/phy_config.py`.
09:55:24.489 [D] plugin:145           Loading 0 plugins.
09:55:24.490 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
09:55:24.491 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
09:55:24.491 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
09:55:24.492 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
09:55:24.493 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
09:55:24.494 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
09:55:24.494 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
09:55:24.494 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
09:55:24.495 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
09:55:24.497 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
09:55:24.497 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
09:55:24.498 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
09:55:24.498 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
09:55:24.498 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
09:55:24.499 [D] context:209          The file `/Users/annateruel/Desktop/registro-5/.phy/new_cluster_id.pkl` doesn't exist.
09:55:24.513 [D] context:185          Save data to `/Users/annateruel/Desktop/registro-5/.phy/spikes_per_cluster.pkl`.
09:55:24.540 [D] gui:463              Creating GUI.
09:55:24.818 [D] state:46             Load /Users/annateruel/.phy/TemplateGUI/state.json for GUIState.
09:55:24.819 [D] state:46             Load /Users/annateruel/Desktop/registro-5/.phy/state.json for GUIState.
09:55:25.222 [D] gui:718              Add view ClusterView to GUI.
09:55:25.276 [D] gui:718              Add view SimilarityView to GUI.
09:55:25.323 [D] gui:718              Add view WaveformView to GUI.
qt.qpa.fonts: Populating font family aliases took 107 ms. Replace uses of missing font family ".AppleSystemUIFont" with one that exists to avoid this cost. 
09:55:25.544 [D] base:337             Set state for WaveformView.
09:55:25.557 [D] gui:718              Add view CorrelogramView to GUI.
09:55:25.560 [D] base:337             Set state for CorrelogramView.
09:55:25.585 [D] gui:718              Add view ISIView to GUI.
09:55:25.588 [D] base:337             Set state for ISIView.
09:55:25.602 [D] gui:718              Add view FeatureView to GUI.
09:55:25.605 [D] base:337             Set state for FeatureView.
09:55:25.628 [D] gui:718              Add view AmplitudeView to GUI.
09:55:25.632 [D] base:337             Set state for AmplitudeView.
09:55:25.656 [D] gui:718              Add view FiringRateView to GUI.
09:55:25.659 [D] base:337             Set state for FiringRateView.
09:55:25.681 [D] gui:718              Add view ProbeView to GUI.
09:55:25.684 [D] base:337             Set state for ProbeView.
09:55:25.700 [D] gui:718              Add view TemplateFeatureView to GUI.
09:55:25.703 [D] base:337             Set state for TemplateFeatureView.
09:55:25.724 [D] gui:718              Add view ClusterScatterView to GUI.
09:55:25.727 [D] base:337             Set state for ClusterScatterView.
09:55:25.727 [D] base:494             Set color scheme to random.
09:55:25.729 [D] base:1626            Adding default color schemes to ClusterScatterView.
09:55:25.955 [D] gui:519              Load the geometry state.
[... 2196 lines cut ...]
 002 // --- Snippets code : end --- //
 ...

[... cut ...]
 -> '' :  #version required and missing.

 000 #define _GLUMPY__VERTEX_SHADER__
 001 // --- Snippets code : start --- //
 002 // --- Snippets code : end --- //
 003 varying vec2 v_pos_tmp;
 ...

Error in Vertex shader 2 (<string>)
 -> 'varying' : syntax error: syntax error

 ...
 001 // --- Snippets code : start --- //
 002 // --- Snippets code : end --- //
 003 varying vec2 v_pos_tmp;
 004 
 005 uniform vec2 u_window_size;
 ...

09:55:32.604 [D] base:702             Exception in paintGL: Shader compilation error
`

zm711, 2023-11-03:
@anna-teruel,

so bad news and this is what I was afraid of. openGL is deprecated on macs. I'm linking to a reddit discussion [here](https://www.reddit.com/r/opengl/comments/oc8ec1/opengl_on_mac/)

So unless you're willing to try to mess around with stuff and really play with the version it is probably actually best to use Phy on Windows or Linux. 

The gui itself is working (which is why you have empty boxes), but openGL is used to actually do the graphics rendering. The errors that you are getting are indicating that openGL is failing on your mac, which is a known issue. 

Sorry! But since macOS is switching over to Vulkan and this package is currently more maintenance rather than active development I don't think this will be fixed for Macs anytime soon.

closed 2024-06-07 by zm711