cortex-lab/phy #1227: SegFault error when trying to launch Phy2 GUI on MobaXterm

manuelvasconcelos, 2023-10-31:
Hello!

I have just installed a fresh version of phy2. I have used phy many times in the past but it has been a couple of months and I see now there is a phy2, which I just installed following the docs, making sure to also delete the status.json files as I previously had phy1 installed. 

Here is the issue:

When trying to run the GUI on some data (which previously ran with no issue) I encountered a SegFault error and have tried --debug, --clear-state, --clear-cache as well as manually deleting the .json files. I have attached a text file with my conda environment as well as debug from phy.

[phySegFaultError.txt](https://github.com/cortex-lab/phy/files/13222067/phySegFaultError.txt)

Please let me know if there's any other information that would help troubleshooting.

Thanks.

zm711, 2023-11-01:
@manuelvasconcelos,
You have phy from pypi. But the latest bug fixes have only been applied to the GitHub version (ie you need to install from git instead of from pypi). 

It might be easiest to just try:
```bash
conda env create -f environment.yml
```
from the repo since that auto-installs the dev version of phy. Otherwise you should follow the instructions on the readme to install with
```bash
pip install git+https://github.com/cortex-lab/phy.git
```
after generating your env.

manuelvasconcelos, 2023-11-02:
@zm711 
Thanks for the response. 

Creating the environment with `conda env create -f environment.yml` solved the issue! It took much longer to resolve the environment but the wait was worth it.

As a note, `conda list` still lists phy and phylib as from the pypi channel, but nonetheless this resolved the segfault error.

Thank you!

closed 2023-11-02 by manuelvasconcelos