cortex-lab/phy #1228: Incorrect waveforms in Phy

neuronzoo, 2023-11-01:
Hi, I've seen a version of this issue before (#1218), but I've tried the solution there and it didn't work for me.
I am using Windows 11 and KS3.
For some of the datasets that I am trying to load into phy, the gui comes up but the waveforms are off: they are presented, but clearly not the right ones - for a "good" unit, they're noisy and scrambled.
The following error appears when I run the command `phy template-gui params.py` from within the phy2 environment:

QWindowsWindow::setGeometry: Unable to set geometry 1946x600+122+86 (frame: 1962x639+114+55) on QWidgetWindow/"TemplateGUIWindow" on "\\.\DISPLAY1". Resulting geometry: 1924x600+122+86 (frame: 1940x639+114+55) margins: 8, 31, 8, 8 minimum size: 1894x115 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=1910,154 maxtrack=0,0)
QMimeDatabase: Error loading internal MIME data
An error has been encountered at line 1 of <internal MIME data>: Premature end of document.:

If I first try to extract the waveforms by using `phy extract-waveforms params.py`, this is what I get:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\phy\apps\__init__.py", line 245, in template_extract_waveforms
    model.save_spikes_subset_waveforms(
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 1390, in save_spikes_subset_waveforms
    export_waveforms(
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\phylib\io\traces.py", line 658, in export_waveforms
    writer = NpyWriter(path, shape, dtype)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Aviv Mizrahi-Kliger\anaconda3\envs\phy2\Lib\site-packages\phylib\io\traces.py", line 559, in __init__
    self.fp = open(path, 'wb')
              ^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: 'D:\\Aviv\\Analysis\\Sleep\\WT-R06\\Sleep\\POS\\2023-09-20\\Record Node 101\\experiment1\\recording1\\continuous\\Neuropix-PXI-100.ProbeA-AP\\_phy_spikes_subset.waveforms.npy'

I tried to do the following, with no success:
1. restarting
2. deleting and reinstalling the phy env
3. installing using the yaml file. Seems like I was missing a dependency named imp there..?

Thanks!
Aviv

neuronzoo, 2023-11-03:
Hey, thanks for the reply! I actually tried it yesterday after reading #1226 - the imp problem is gone, but WaveformView is still off - scrambled and weird spikes for all (good) units. the rest of the GUI opens fine.

Here is the output I get for `phy template-gui params.py --debug`:
[text1.txt](https://github.com/cortex-lab/phy/files/13252546/text1.txt)

And here is what I get for `phy extract-waveforms params.py`:
[text2.txt](https://github.com/cortex-lab/phy/files/13252548/text2.txt)

Maybe relevant - From the Anaconda prompt, I can't get to my KS folder using just `"cd path\to\ks\folder"` so I do `"cd /d path\to\ks\folder"`

Thanks!

zm711, 2023-11-03:
@neuronzoo 

> Maybe relevant - From the Anaconda prompt, I can't get to my KS folder using just "cd path\to\ks\folder" so I do "cd /d path\to\ks\folder"

This is a windows cd issue. The /d means switching drive (c to d etc.). So no worries there. Most instructions are not written for windows so they neglect that fact.

So could you send a tiny screen shot of one of the "failed" waveform views. Also maybe a screen shot of what files are actually in the folder you're trying to analyze (should be like `params.py`, 'raw_data.dat`, a bunch of numpy files.

And the contents of the params.py file. I'm trying to figure out if this is a file error or something else.  I'm wondering mostly about what the dat path says for that.

neuronzoo, 2023-11-03:
Hey, definitely->
Here is a screenshot for a relatively good cell, 18K spikes, good separation from noise judging by correlogram and amplitude views, but waveforms look messed up:
[image]

Kilosort folder content: 
[image]

And the params.py file:
```
dat_path = 'D:/.../Analysis/Sleep/WT-R06/Sleep/TAS/2023-09-20/Record Node 101/experiment1/recording1/continuous/Neuropix-PXI-100.ProbeA-AP/temp_wh.dat'
n_channels_dat = 383
dtype = 'int16'
offset = 0
sample_rate = 30000.
hp_filtered = True
```

zm711, 2023-11-03:
I see. One thing you could try is using the raw file rather than the whitened file. Your raw file is the `continous.dat` right? You could edit the dat path to end with `continous.dat` instead and use the raw waveforms rather than the whitened data. Could you try that and see if you think the waveforms look better.

neuronzoo, 2023-11-03:
Hey, still the same result - > 
for `phy template-gui params.py --debug`:
[text3.txt](https://github.com/cortex-lab/phy/files/13255035/text3.txt)

And for `phy extract-waveforms params.py`:
[text4.txt](https://github.com/cortex-lab/phy/files/13255033/text4.txt)

I tried a fresh new recording session from Kilosort, and it actually works well. Could it be that if I use phy to modify the original kilosort data, that by itself causes the waveform problem?

neuronzoo, 2023-11-13:
Hey, 
1. I really don't know - For the same unit, on the first time I ran Phy the waveforms were spike-like with very low noise, and then after doing some curation and reopening (>5 times) I always got low signal-to-noise, variable spikes. I do think there is something going on there..
2. How would you recommend to name the files to prevent escaping?

zm711, 2023-11-13:
@neuronzoo,

>I really don't know - For the same unit, on the first time I ran Phy the waveforms were spike-like with very low noise, and then after doing some curation and reopening (>5 times) I always got low signal-to-noise, variable spikes. I do think there is something going on there..

For 1). yeah, sometimes that happens. If things are weird. I restart my computer. If super weird, I clear the phy caches and restart the computer and see if that helps. But the waveforms themselves should not be overwritten since they are just pulled from the raw file. If you merged a noise cluster with a "good" cluster though then the next time you open it could pull the noise waveforms for that unit since it is suppose to randomly pull waveforms. What type of curation have you been doing-- (ie a lot of merging of clusters)?

> How would you recommend to name the files to prevent escaping?

2). The easiest solution is to feed a raw string into python. So `r'd:\...\'`, but unfortunately for the command line I don't believe this is an option. (Feel free to try it though!).  Otherwise, you would just need to google for escaping values in python (like \t means tab \n means return etc). 

The other potential problem that just hit me is that python has a limit to file name around like ~100 characters or so.  Your file path is ~138 characters. Could you actually try to shorten it by  making it a little more shallow? An easy test would be something like:

`D:/your name/ Analysis/Record Node 101/experiment1/recording1/continuous/npx`

This will allow us to test if you are just reaching the limits of file name length in python. (It's happened to me before and it's actually a pain to debug because the errors are often OS errors that give no indication that the file lengths are the problems.)

himahuja, 2024-02-14:
I have the same work flow, and the same error. But sadly the character limit is not the issue here, since my file location <100 characters long. 

```
File "C:\Users\ramirezlab\anaconda3\envs\phy2\Lib\site-packages\phylib\io\traces.py", line 559, in __init__
    self.fp = open(path, 'wb')
              ^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument: 'C:\\tmp\\kilosort3\\sorter_output\\_phy_spikes_subset.waveforms.npy'
```

zm711, 2024-02-14:
Why do you have `.waveforms.npy`. I'm wondering if the double `.` is causing your problem @himahuja ?

zm711, 2024-06-07:
I'll close since the original issue was fixed. Feel free to open new issues if something comes up.

closed 2024-06-07 by zm711