cortex-lab/phy #1234: Could not create view FeatureView

WWL606, 2023-11-18:
Hi,

I am using kilosort3.  I can not create FeatureView and TemplateFeatureView

```
model:603            Unreferenced clusters found in templates (generally not a problem)
model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
gui:671              Could not create view FeatureView.
gui:671              Could not create view TemplateFeatureView.
```

Also, why the Waveformview look different? 
[image]

How can I see the waveforms as follows:
[image]

Thanks,
Wenliang

zm711, 2023-11-20:
@WWL606,

this is a known problem. Kilosort3 decided to no longer write those .npy files so those views cannot be generated. Your options are to: 
1) downgrade to Kilo2.5
2) use spikeinterface which will generate those files from Kilosort3 data (although not the exact files, so it will look different)
3) just other views for curation.

Does that all make sense? The kilosort people have switched over to just working on KS4, so I really don't think this will be fixed on their end any time soon. Sorry!
Best,
Zach

WWL606, 2023-11-21:
Hi Zach, Thanks so much for your answer!
Best,
Wenliang Wang