cortex-lab/phy #1236: installation issues on macos ventura

alemessurier, 2023-11-21:
Hi, I've tried installing phy2 several ways on my macbook pro (running ventura) but I've run into multiple issues.

When I install using the instructions from the read me I get the issue described here: https://github.com/cortex-lab/phy/issues/1231

I then cloned the whole repository and made the environment from the .yml file and I get the openGL error described in that ^ issue, though I'm doing this locally, not on a cluster. 

I next tried just downloading the environment.yml file, making a new conda environment from the .yml, and running phy and I get this error, which seems to be due to phy2 being incompatible with python 3.12 (so maybe the environment.yml needs to be updated?): 

```
conda env create -f environment.yml
conda activate phy2
phy template-gui ./params.py --debug
```

```
Traceback (most recent call last):
  File "/Users/aml717/opt/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.12/site-packages/phy/__init__.py", line 21, in <module>
    from .utils.config import load_master_config
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.12/site-packages/phy/utils/__init__.py", line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.12/site-packages/phy/utils/plugin.py", line 14, in <module>
    import imp
ModuleNotFoundError: No module named 'imp'
```

here is the environment at this point:

`conda list

# packages in environment at /Users/aml717/opt/anaconda3/envs/phy2:
#
# Name                    Version                   Build  Channel
appnope                   0.1.3              pyhd8ed1ab_0    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
aws-c-auth                0.7.7                hc3630cc_0    conda-forge
aws-c-cal                 0.6.9                h49e9720_1    conda-forge
aws-c-common              0.9.8                h10d778d_0    conda-forge
aws-c-compression         0.2.17               hff1f2c8_6    conda-forge
aws-c-event-stream        0.3.2                hb6e475e_6    conda-forge
aws-c-http                0.7.14               h950a07a_1    conda-forge
aws-c-io                  0.13.35              hb98174f_8    conda-forge
aws-c-mqtt                0.9.9                h5e4a26e_0    conda-forge
aws-c-s3                  0.3.24               h4293da7_1    conda-forge
aws-c-sdkutils            0.1.12               hff1f2c8_5    conda-forge
aws-checksums             0.1.17               hff1f2c8_5    conda-forge
aws-crt-cpp               0.24.7               h0a75192_2    conda-forge
aws-sdk-cpp               1.11.182             h28d282b_7    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
bokeh                     3.3.1              pyhd8ed1ab_0    conda-forge
brotli                    1.1.0                    pypi_0    pypi
brotli-bin                1.1.0                h0dc2134_1    conda-forge
brotli-python             1.1.0           py312heafc425_1    conda-forge
bzip2                     1.0.8                h10d778d_5    conda-forge
c-ares                    1.22.1               h10d778d_0    conda-forge
ca-certificates           2023.11.17           h8857fd0_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
certifi                   2023.11.17         pyhd8ed1ab_0    conda-forge
charset-normalizer        3.3.2              pyhd8ed1ab_0    conda-forge
click                     8.1.7           unix_pyh707e725_0    conda-forge
cloudpickle               3.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
colorcet                  3.0.1                    pypi_0    pypi
comm                      0.1.4              pyhd8ed1ab_0    conda-forge
contourpy                 1.2.0                    pypi_0    pypi
curl                      8.4.0                h726d00d_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
[... 472 lines cut ...]
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
toolz                     0.12.0             pyhd8ed1ab_0    conda-forge
tornado                   6.3.3            py38h6c40b1e_0    anaconda
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.13.0             pyhd8ed1ab_0    conda-forge
types-pyyaml              6.0.12.12          pyhd8ed1ab_0    conda-forge
typing_extensions         4.8.0              pyha770c72_0    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
urllib3                   2.1.0              pyhd8ed1ab_0    conda-forge
watchdog                  3.0.0                    pypi_0    pypi
wcwidth                   0.2.10             pyhd8ed1ab_0    conda-forge
wheel                     0.41.3             pyhd8ed1ab_0    conda-forge
xorg-libxau               1.0.11               h0dc2134_0    conda-forge
xorg-libxdmcp             1.1.3                h35c211d_0    conda-forge
xyzservices               2023.10.1          pyhd8ed1ab_0    conda-forge
xz                        5.4.2                h6c40b1e_0    anaconda
yaml                      0.2.5                h0d85af4_2    conda-forge
zeromq                    4.3.5                h93d8f39_0    conda-forge
zict                      3.0.0              pyhd8ed1ab_0    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h4dc903c_0    anaconda
zstd                      1.5.5                hc035e20_0    anaconda`

Any advice on what to try next? thanks!

alemessurier, 2023-11-21:
I have also tried:

`$ conda create -n phy2 -y cython dask h5py joblib matplotlib numpy pillow pip pyopengl pyqt pyqtwebengine pytest python=3.8 qtconsole requests responses scikit-learn scipy traitlets
$ conda activate phy2
(phy2) $ pip install phy --pre --upgrade`

and I get this error:

`(phy2)$ phy template-gui ./params.py --debug
/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/plot/gloo/variable.py:86: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
  gl.GL_BOOL: (1, gl.GL_BOOL, np.bool),
Traceback (most recent call last):
  File "/Users/aml717/opt/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/__init__.py", line 25, in <module>
    from .base import (  # noqa
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 26, in <module>
    from phy.cluster._utils import RotatingProperty
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/__init__.py", line 9, in <module>
    from .views import *  # noqa
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/views/__init__.py", line 10, in <module>
    from .base import ManualClusteringView  # noqa
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/views/base.py", line 21, in <module>
    from phy.plot import PlotCanvas, NDC, extend_bounds
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/plot/__init__.py", line 17, in <module>
    from .base import BaseVisual, GLSLInserter, BaseCanvas, BaseLayout
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/plot/base.py", line 20, in <module>
    from . import gloo
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/plot/gloo/__init__.py", line 10, in <module>
    from .program import Program
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/plot/gloo/program.py", line 18, in <module>
    from .variable import Uniform, Attribute
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/plot/gloo/variable.py", line 86, in <module>
    gl.GL_BOOL: (1, gl.GL_BOOL, np.bool),
  File "/Users/aml717/opt/anaconda3/envs/phy2/lib/python3.8/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'bool'.
`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
`

Here is my conda environment: 
`(phy2) $ conda list
# packages in environment at /Users/aml717/opt/anaconda3/envs/phy2:
#
# Name                    Version                   Build  Channel
appnope                   0.1.2           py38hecd8cb5_1001  
asttokens                 2.0.5              pyhd3eb1b0_0  
backcall                  0.2.0              pyhd3eb1b0_0  
blas                      1.0                    openblas  
bokeh                     2.4.3            py38hecd8cb5_0  
bottleneck                1.3.5            py38h67323c0_0  
brotli                    1.0.9                hca72f7f_7  
brotli-bin                1.0.9                hca72f7f_7  
brotli-python             1.0.9            py38he9d5cce_7  
c-ares                    1.19.1               h6c40b1e_0  
ca-certificates           2023.08.22           hecd8cb5_0  
certifi                   2023.11.17       py38hecd8cb5_0  
cffi                      1.16.0           py38h6c40b1e_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
click                     8.1.7            py38hecd8cb5_0  
cloudpickle               2.2.1            py38hecd8cb5_0  
colorcet                  3.0.1                    pypi_0    pypi
comm                      0.1.2            py38hecd8cb5_0  
contourpy                 1.0.5            py38haf03e11_0  
cryptography              41.0.3           py38h30e54ef_0  
cycler                    0.11.0             pyhd3eb1b0_0  
cyrus-sasl                2.1.28               h3973f50_1  
cython                    3.0.0            py38h6c40b1e_0  
[... 134 lines cut ...]
sip                       6.7.12           py38hcec6c5f_0  
six                       1.16.0             pyhd3eb1b0_1  
sortedcontainers          2.4.0              pyhd3eb1b0_0  
sqlite                    3.41.2               h6c40b1e_0  
stack_data                0.2.0              pyhd3eb1b0_0  
tblib                     1.7.0              pyhd3eb1b0_0  
threadpoolctl             2.2.0              pyh0d69192_0  
tk                        8.6.12               h5d9f67b_0  
tomli                     2.0.1            py38hecd8cb5_0  
toolz                     0.12.0           py38hecd8cb5_0  
tornado                   6.3.3            py38h6c40b1e_0  
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.7.1            py38hecd8cb5_0  
typing_extensions         4.7.1            py38hecd8cb5_0  
[... cut ...]
wcwidth                   0.2.5              pyhd3eb1b0_0  
wheel                     0.41.2           py38hecd8cb5_0  
xz                        5.4.2                h6c40b1e_0  
yaml                      0.2.5                haf1e3a3_0  
zeromq                    4.3.4                h23ab428_0  
zict                      3.0.0            py38hecd8cb5_0  
zipp                      3.11.0           py38hecd8cb5_0  
zlib                      1.2.13               h4dc903c_0  
zstd                      1.5.5                hc035e20_0 `

alemessurier, 2023-11-21:
ok I just found this issue related to openGL being deprecated on macs, I wonder if this is the root of my issues too: https://github.com/cortex-lab/phy/issues/1226

Does this depend on which macOS version I'm running?

alemessurier, 2023-11-21:
Ok, I was finally able to open the gui by changing python in the environment.yml to python==3.11 and building the conda environment from the .yml. However, I end up with the empty plots issue from #1226 so I guess I'm giving up on using phy on my mac unless anyone has found a way around this!!

zm711, 2023-11-21:
@alemessurier,

I really don't think Phy will work on Macs. But yeah technically you might be able to use an older version of macOS and get it working since openGL was part of older Macs. I think an older Intel Mac might be your best bet (but in that case you might need older python etc). So it probably isn't worth the effort.

alemessurier, 2023-11-21:
@zm711 thanks, that's good to know! It might be nice to mention the issue with newer macs/macOS in the readme - I don't think I saw anything about this, just assumed phy was generally compatible with macs since my labmate uses it on an older macbook.

closed 2023-11-21 by alemessurier

zm711, 2023-11-21:
Thanks @alemessurier, I'm a contributor, but I'm not the main person, so I can't add it to the readme (PRs get merged about 1 or 2x / year), but Phy used to work on Macs fine. The switch over to their own Arm-based on chips/Metal instead is the main problem here. So when Phy was first written openGl was fine, but Apple has been slowly removing support to push people to program with Metal.

alemessurier, 2023-11-21:
got it, thanks!