cortex-lab/phy #1239: No suitable graphics backend found

nmtimme, 2023-11-27:
Hello

I'm attempting to run phy2 on a virtual desktop and I'm encountering an error. I've previously run phy2 on this same virtual desktop, so I know it is possible. However, I had to reinstall phy2 and some new update since the previous version must have produced problems. Specifically, I ran:

```
conda create -n phy2 -y cython dask h5py joblib matplotlib numpy pillow pip pyopengl pyqt pyqtwebengine pytest python=3.11 qtconsole requests responses scikit-learn scipy traitlets

conda activate phy2

pip install git+https://github.com/cortex-lab/phy.git
```

Note, I had to use python=3.11 to avoid the missing imp module in python 3.12 (see issue #1224  ). 

Then, when I run `phy template-gui params.py --debug`, I get:

```
12:00:52.894 [D] __init__:68          Start capturing exceptions.
12:00:52.897 [W] model:1408           File /N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/temp/sorter_output/temp_wh.dat does not exist.
12:00:52.919 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
12:00:52.928 [D] model:619            Loading spike clusters.
12:00:52.938 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
12:00:52.953 [D] model:569            No channel shank file found.
12:00:52.955 [D] model:692            Loading templates.
12:00:52.982 [D] model:724            Templates are dense.
12:00:52.983 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
12:00:52.983 [D] model:730            Loading the whitening matrix.
12:00:52.984 [D] model:737            Loading the inverse of the whitening matrix.
12:00:52.987 [W] traces:462           File /N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/temp/sorter_output/temp_wh.dat does not exist.
12:00:52.987 [W] traces:462           File /N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/temp/sorter_output/temp_wh.dat does not exist.
12:00:52.987 [D] model:766            Loading features.
12:00:52.987 [D] model:803            Loading template features.
12:00:52.990 [D] model:504            Load `cluster_ContamPct.tsv`.
12:00:52.992 [D] model:504            Load `cluster_KSLabel.tsv`.
12:00:52.994 [D] model:504            Load `cluster_Amplitude.tsv`.
12:00:52.996 [D] model:504            Load `cluster_group.tsv`.
12:00:54.355 [D] context:100          Initialize joblib cache dir at `/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/sorter_output/.phy`.
12:00:54.356 [D] context:101          Reducing the size of the cache if needed.
12:00:54.440 [D] base:102             Add filter `high_pass`.
12:00:54.445 [D] config:31            Load config file `/N/u/nmtimme/BigRed200/.phy/phy_config.py`.
12:00:54.446 [D] plugin:145           Loading 0 plugins.
12:00:54.446 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
12:00:54.447 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
12:00:54.448 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
12:00:54.450 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
12:00:54.451 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
12:00:54.451 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
12:00:54.452 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
12:00:54.453 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
12:00:54.454 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
12:00:54.455 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
12:00:54.456 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
12:00:54.457 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
12:00:54.458 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
12:00:54.459 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
12:00:54.461 [D] context:209          The file `/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/sorter_output/.phy/new_cluster_id.pkl` doesn't exist.
12:00:54.469 [D] context:185          Save data to `/N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/sorter_output/.phy/spikes_per_cluster.pkl`.
12:00:54.486 [D] gui:463              Creating GUI.
12:00:54.640 [D] state:46             Load /N/u/nmtimme/BigRed200/.phy/TemplateGUI/state.json for GUIState.
12:00:54.641 [D] state:46             Load /N/project/lapishLabWorkspace/SpikeInterfaceSpikeSorting/NeuropixelProbeSpikeSortingResults/2023-11-15_12-27-09/sorter_output/.phy/state.json for GUIState.
No suitable graphics backend found

Aborted (core dumped)
```

I'd appreciate any advice you can provide. Thank you!

~Nick

zm711, 2023-11-27:
@nmtimme,

Howdy Nick. Yeah there were a couple updates. Could you try to install with the conda yaml instead.  It is listed as `enviroment.yml` on the main page. As you noted you need to limit python to 3.11 (my PR to fix this is awaiting review #1229) in the yml so once you download it just edit python in the dependencies. 

Could I also get some info about what the virtual machine and actual machine are (OS-wise)?

zm711, 2023-11-27:
I wish all issues were this easy :). Happy to help!
Zach