cortex-lab/phy #1240: Cluster view and Similarity View blank

Coglet00, 2023-12-18:
i was trouble shooting this problem because both views are blank even when env QTWEBENGINE_CHROMIUM_FLAGS="--single-process is used before opening phy. This had worked for many others. i heard about the readCSVfunction (https://github.com/cortex-lab/spikes/blob/master/preprocessing/phyHelpers/readClusterGroupsCSV.m) as this might be useful for reading cluster_group.tsv file. 

i am a bit inexperienced so i  dont know how to exactly integrate this new fucntion in matlab. like  in what folder do i put it and where do i call it?

thanks for help!

zm711, 2024-01-08:
@Coglet00,

Could you post any errors your getting here. This can be done by running fine in debug mode like this:

```bash
phy template-gui path/to/parms.py --debug
```

Then copy the log info. I'm not quite sure what you're actually asking at the moment. the spikes repo that you are mentioning is for post processing of phy data not for use inside of Phy. But maybe you're asking a couple different things?

ashsrobbins, 2024-01-31:
@zm711 I have the same issue on Ubuntu right now:

Here is the log when I click on a cluster (I can click them, just not see them).

[image]

> 14:51:30.695 [D] __init__:68          Start capturing exceptions.
> 14:51:30.696 [W] model:1408           File traces_cached_seg0.raw does not exist.
> 14:51:30.741 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
> 14:51:30.771 [D] model:619            Loading spike clusters.
> 14:51:30.803 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
> 14:51:30.835 [D] model:569            No channel shank file found.
> 14:51:30.836 [D] model:692            Loading templates.
> 14:51:30.856 [D] model:724            Templates are dense.
> 14:51:30.857 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 14:51:30.857 [D] model:730            Loading the whitening matrix.
> 14:51:30.861 [D] model:737            Loading the inverse of the whitening matrix.
> 14:51:30.866 [W] traces:462           File /media/danser-lab/hippo1/cartpole/24-01-07_data/20217/exp1/results_KS2_stitch/sorter_output/traces_cached_seg0.raw does not exist.
> 14:51:30.866 [W] traces:462           File /media/danser-lab/hippo1/cartpole/24-01-07_data/20217/exp1/results_KS2_stitch/sorter_output/traces_cached_seg0.raw does not exist.
> 14:51:30.866 [D] model:766            Loading features.
> 14:51:30.866 [D] model:781            Features are sparse.
> 14:51:30.867 [D] model:803            Loading template features.
> 14:51:30.867 [D] model:813            Template features are sparse.
> 14:51:30.867 [D] model:504            Load `cluster_KSLabel.tsv`.
> 14:51:30.868 [D] model:504            Load `cluster_ContamPct.tsv`.
> 14:51:30.869 [D] model:504            Load `cluster_Amplitude.tsv`.
> 14:51:30.870 [D] model:504            Load `cluster_group.tsv`.
> 14:51:31.036 [D] context:100          Initialize joblib cache dir at `/media/danser-lab/hippo1/cartpole/24-01-07_data/20217/exp1/results_KS2_stitch/sorter_output/.phy`.
> 14:51:31.037 [D] context:101          Reducing the size of the cache if needed.
> 14:51:31.043 [D] base:102             Add filter `high_pass`.
> 14:51:31.043 [D] config:31            Load config file `/home/danser-lab/.phy/phy_config.py`.
> 14:51:31.043 [D] plugin:145           Loading 0 plugins.
> 14:51:31.043 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
> 14:51:31.044 [D] context:126          Load memcache for `phy.apps.base._get_mean_waveforms`.
> 14:51:31.044 [D] context:126          Load memcache for `phy.apps.base._get_template_waveforms`.
> 14:51:31.044 [D] context:126          Load memcache for `phy.apps.base.get_mean_spike_template_amplitudes`.
> 14:51:31.044 [D] context:126          Load memcache for `phy.apps.base.get_template_counts`.
> 14:51:31.044 [D] context:126          Load memcache for `phy.apps.base.get_template_for_cluster`.
> 14:51:31.045 [D] context:126          Load memcache for `phy.apps.template.gui.get_template_amplitude`.
> 14:51:31.046 [D] context:126          Load memcache for `phy.apps.base.get_cluster_amplitude`.
> 14:51:31.046 [D] context:126          Load memcache for `phy.apps.base.get_mean_firing_rate`.
> 14:51:31.046 [D] context:126          Load memcache for `phy.apps.base.get_best_channel`.
> 14:51:31.046 [D] context:126          Load memcache for `phy.apps.template.gui.get_best_channels`.
> 14:51:31.047 [D] context:126          Load memcache for `phy.apps.base.get_channel_shank`.
> 14:51:31.047 [D] context:126          Load memcache for `phy.apps.base.get_probe_depth`.
> 14:51:31.047 [D] context:126          Load memcache for `phy.apps.base.peak_channel_similarity`.
> 14:51:31.073 [D] context:185          Save data to `/media/danser-lab/hippo1/cartpole/24-01-07_data/20217/exp1/results_KS2_stitch/sorter_output/.phy/spikes_per_cluster.pkl`.
> 14:51:31.131 [D] gui:463              Creating GUI.
> 14:51:31.136 [D] state:46             Load /home/danser-lab/.phy/TemplateGUI/state.json for GUIState.
> 14:51:31.136 [D] state:46             Load /media/danser-lab/hippo1/cartpole/24-01-07_data/20217/exp1/results_KS2_stitch/sorter_output/.phy/state.json for GUIState.
> 14:51:31.265 [D] gui:718              Add view ClusterView to GUI.
> 14:51:31.275 [D] gui:718              Add view SimilarityView to GUI.
> 14:51:31.306 [D] gui:718              Add view WaveformView to GUI.
> 14:51:31.309 [D] base:337             Set state for WaveformView.
> 14:51:31.319 [D] gui:718              Add view CorrelogramView to GUI.
> 14:51:31.321 [D] base:337             Set state for CorrelogramView.
> 14:51:31.340 [D] gui:718              Add view ISIView to GUI.
> 14:51:31.341 [D] base:337             Set state for ISIView.
> 14:51:31.352 [D] gui:718              Add view FeatureView to GUI.
> 14:51:31.354 [D] base:337             Set state for FeatureView.
> 14:51:31.372 [D] gui:718              Add view AmplitudeView to GUI.
> 14:51:31.373 [D] base:337             Set state for AmplitudeView.
> 14:51:31.391 [D] gui:718              Add view FiringRateView to GUI.
> 14:51:31.393 [D] base:337             Set state for FiringRateView.
> 14:51:31.659 [D] gui:718              Add view ProbeView to GUI.
> 14:51:31.661 [D] base:337             Set state for ProbeView.
> 14:51:31.676 [D] gui:718              Add view ClusterScatterView to GUI.
[... cut ...]
> 14:51:31.715 [D] base:494             Set color scheme to random.
> 14:51:31.716 [D] base:1626            Adding default color schemes to ClusterScatterView (1).
> 14:51:31.758 [D] gui:519              Load the geometry state.
> 14:51:32.268 [D] supervisor:817       Clusters selected: [30] (36)
> 14:53:22.026 [D] supervisor:817       Clusters selected: [22] (37)
> 14:53:23.821 [D] supervisor:839       Similar clusters selected: [178] (164)
> 14:53:25.309 [D] supervisor:817       Clusters selected: [294] (170)

ashsrobbins, 2024-01-31:
Hey @zm711! 

Yes spikeinterface and kilosort2.

params.py:
```dat_path = 'traces_cached_seg0.raw'
n_channels_dat = 951
dtype = 'int16'
offset = 0
sample_rate = 20000.
hp_filtered = False
```

zm711, 2024-01-31:
Is there no `dat_path` given? (or did you delete it for privacy). We really should have spikeinterface indicate where the `.dat` is.

ashsrobbins, 2024-01-31:
Ahh, apologies, I accidentally removed it:
```
dat_path = 'traces_cached_seg0.raw'
n_channels_dat = 951
dtype = 'int16'
offset = 0
sample_rate = 20000.
hp_filtered = False
```
 I'll check with some of my old recordings, because I have opened phy before successfully with data sorted in this same way. However I just recently created a new conda env and followed the install directions on the main page, causing this error.

zm711, 2024-01-31:
There are some bugs with the main page install. We can try something to fix it:

Instead could you download the `environment.yaml` from the root directory in this repo. Then open it and edit for python<3.12 (there's an issue with 3.12 just waiting for my PR to be merged to fix it). Then recreate the environment that way by creating the environment with a file.

zm711, 2024-06-07:
With python 3.11 which is now enforced by the yaml this should be fixed.

closed 2024-06-07 by zm711