cortex-lab/phy #1241: Programmatically save cluster_info.tsv

chris-angeloni, 2023-12-19:
Is there a command to export the cluster info of a sorted recording to a .tsv file without opening the GUI and pressing ctrl-S?

abisi, 2024-04-29:
Hello, did you find a way by chance?

chris-angeloni, 2024-04-29:
Unfortunately no. Kilosort does create a "cluster_group.tsv" file though, so I load that and cross-reference it to the cluster ids in spike_clusters.npy to create a new "cluster_info.tsv" file. Unfortunately it does not have all of the useful information from phy though (depth, channel, etc), I compute that on my own and add it

abisi, 2024-04-29:
Hi! No worries, thanks a lot for your reply. 😄 
In the end I was too lazy to recreate a `cluster_info` myself so I run a subprocess that opens the Phy GUI, saves and quits. 😅 
The following works for me, using the `pyautogui` package:

```       
# Execute Phy to generate cluster_info table
command = 'conda activate phy2 && phy template-gui params.py && conda deactivate'
process = subprocess.Popen(command,  shell=True, cwd=os.path.join(probe_path, 'kilosort2'))
time.sleep(15) # wait for GUI to load
pyautogui.hotkey('ctrl', 's') # simulate CTRL+S key press to save the file
time.sleep(15) # wait for saving to complete
pyautogui.hotkey('ctrl', 'q') # close GUI
time.sleep(15)
process.terminate() # terminate process
```

The sleeping times are necessary it seems but could be shortened, potentially.

Have an excellent day,
Axel