cortex-lab/phy #1242: Missing Amplitudeview

hbirge06, 2023-12-21:
I recently updated my version of Phy to the most recent version using the 'pip install git+https://github.com/cortex-lab/phy.git' command, but now I am unable to see anything in the Amplitudeview box and the window is appearing but it is completely blank apart from the reference grid. It seems to be happening on all the data, including ones that showed the amplitudeview correctly when running Phy in the past (using the previous versions). I'm also not getting any error messages, so any advice on where to look for a solution would be appreciated! Thanks.

Antho2422, 2024-01-03:
Hi @hbirge06, 

Can you please send me the dat_path in your params.py file ? I've been experiencing some similar issues and it was due to the fact that the path to my phy folder was too long. 

Maybe your issue is similar !

Cheers, 
Anthony

EricKenjiLee, 2024-03-27:
I'm running into the same error but it doesn't seem to do with path name length. Below is my path for params.py and I still receive a blank AmplitudeView.
/home/chandlab/Kilosort4 Data/deleteme/params.py

gkBCCN, 2024-04-04:
Hey @EricKenjiLee.  I had a similar issue and your post helped me to fix it (partially).  So first of all, thanks for that!  It's actually so simple that maybe you overlooked it as I did:

1. Make sure that the refresh button is clicked in the AmplitudeView (looks like recycle circle).  It should be gray like the other refresh buttons in the other views.
2. Make sure that you are using template amplitude instead of feature amplitude.  Even if you have template selected, it won't show if you don't have refresh activated from the first step.  

In my case, I'm having some feature loading issues, so anything that is related to features won't work, including the feature amplitude.  That may not be your problem, but if so this will allow you to at least view the template amplitude.