cortex-lab/phy #1244: "zsh: segmentation fault  phy template-gui params.py"

syedtamim4nov, 2024-01-05:
The following error occurs when I run the line ' phy template-gui params.py' in terminal. I am an undergraduate trying to learn and master Phy and any help would be highly appreciated.

zm711, 2024-01-08:
@syedtamim4nov

zsh is the base shell for Macs. Since Phy relies heavily on openGL, it isn't super friendly with Macs any more.

Could you let us know your OS and copy the contents of a:

```python
conda list
```
inside your environment (if you're using conda--if using a python virtual env then the equivalent.

syedtamim4nov, 2024-01-08:
The MacOS is Sonoma 14.1

# packages in environment at /Users/syedtamimahmad/anaconda3/envs/phy2:
#
# Name                    Version                   Build  Channel
abseil-cpp                20230802.0           h313beb8_2  
appnope                   0.1.2           py311hca03da5_1001  
arrow-cpp                 11.0.0               hc7aafb3_2  
asttokens                 2.0.5              pyhd3eb1b0_0  
aws-c-common              0.6.8                h80987f9_1  
aws-c-event-stream        0.1.6                h313beb8_6  
aws-checksums             0.1.11               h80987f9_2  
aws-sdk-cpp               1.8.185              ha71a6ea_1  
backcall                  0.2.0              pyhd3eb1b0_0  
blas                      1.0                    openblas  
bokeh                     3.3.0           py311hb6e6a13_0  
boost-cpp                 1.82.0               h48ca7d4_2  
bottleneck                1.3.5           py311ha0d4635_0  
brotli                    1.0.9                h1a28f6b_7  
brotli-bin                1.0.9                h1a28f6b_7  
brotli-python             1.0.9           py311h313beb8_7  
bzip2                     1.0.8                h620ffc9_4  
c-ares                    1.19.1               h80987f9_0  
ca-certificates           2023.12.12           hca03da5_0  
certifi                   2023.11.17      py311hca03da5_0  
cffi                      1.16.0          py311h80987f9_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
click                     8.1.7           py311hca03da5_0  
cloudpickle               2.2.1           py311hca03da5_0  
colorcet                  3.0.1                    pypi_0    pypi
comm                      0.1.2           py311hca03da5_0  
contourpy                 1.2.0           py311h48ca7d4_0  
cryptography              41.0.7          py311hd4332d6_0  
cycler                    0.11.0             pyhd3eb1b0_0  
cyrus-sasl                2.1.28               h9131b1a_1  
cython                    3.0.6           py311h80987f9_0  
cytoolz                   0.12.2          py311h80987f9_0  
dask                      2023.11.0       py311hca03da5_0  
dask-core                 2023.11.0       py311hca03da5_0  
debugpy                   1.6.7           py311h313beb8_0  
decorator                 5.1.1              pyhd3eb1b0_0  
distributed               2023.11.0       py311hca03da5_0  
executing                 0.8.3              pyhd3eb1b0_0  
fonttools                 4.25.0             pyhd3eb1b0_0  
freetype                  2.12.1               h1192e45_0  
fsspec                    2023.10.0       py311hca03da5_0  
gettext                   0.21.0               h13f89a0_1  
gflags                    2.2.2                hc377ac9_0  
ghp-import                2.1.0                    pypi_0    pypi
giflib                    5.2.1                h80987f9_3  
glib                      2.69.1               h514c7bf_2  
glog                      0.5.0                hc377ac9_0  
grpc-cpp                  1.48.2               hc60591f_4  
gst-plugins-base          1.14.1               h313beb8_1  
gstreamer                 1.14.1               h80987f9_1  
gtest                     1.14.0               h48ca7d4_0  
h5py                      3.9.0           py311hba6ad2f_0  
hdf5                      1.12.1               h05c076b_3  
heapdict                  1.0.1              pyhd3eb1b0_0  
icu                       73.1                 h313beb8_0  
idna                      3.4             py311hca03da5_0  
importlib-metadata        6.0.0           py311hca03da5_0  
iniconfig                 1.1.1              pyhd3eb1b0_0  
ipykernel                 6.25.0          py311hb6e6a13_0  
ipython                   8.15.0          py311hca03da5_0  
jedi                      0.18.1          py311hca03da5_1  
jinja2                    3.1.2           py311hca03da5_0  
joblib                    1.2.0           py311hca03da5_0  
jpeg                      9e                   h80987f9_1  
jupyter_client            8.6.0           py311hca03da5_0  
[... 112 lines cut ...]
snappy                    1.1.10               h313beb8_1  
sortedcontainers          2.4.0              pyhd3eb1b0_0  
sqlite                    3.41.2               h80987f9_0  
stack_data                0.2.0              pyhd3eb1b0_0  
tblib                     1.7.0              pyhd3eb1b0_0  
threadpoolctl             2.2.0              pyh0d69192_0  
tk                        8.6.12               hb8d0fd4_0  
toolz                     0.12.0          py311hca03da5_0  
tornado                   6.3.3           py311h80987f9_0  
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.7.1           py311hca03da5_0  
tzdata                    2023c                h04d1e81_0  
urllib3                   1.26.18         py311hca03da5_0  
utf8proc                  2.6.1                h1a28f6b_0  
watchdog                  3.0.0                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0  
wheel                     0.41.2          py311hca03da5_0  
xyzservices               2022.9.0        py311hca03da5_1  
xz                        5.4.5                h80987f9_0  
yaml                      0.2.5                h1a28f6b_0  
zeromq                    4.3.4                hc377ac9_0  
zict                      3.0.0           py311hca03da5_0  
zipp                      3.11.0          py311hca03da5_0  
zlib                      1.2.13               h5a0b063_0  
zstd                      1.5.5                hd90d995_0

zm711, 2024-01-08:
So as I said unfortunately the current version of MacOS isn't really compatible with Phy any more and since phy isn't being actively developed at the moment it likely won't work. Your best bet is to use a Linux or Windows computer for this analysis.

syedtamim4nov, 2024-01-08:
@zm711 

I am using VS Code on MacOs Sonoma 14.1....... is there any way to rectify the issue? Thanks a lot for the input!

syedtamim4nov, 2024-01-08:
@zm711 Thanks a lot!

zm711, 2024-01-08:
Of course. If you run into any install issue on a different computer just let me know--happy to help.

closed 2024-06-04 by zm711