cortex-lab/phy #1245: IndexError

hugosmie, 2024-01-09:
[image]
I encountered this issue when I split the cluster in the feature space with ctl+left click. And after this error, the following operation would raise this issue again and again (like simply select a cluster and so on).
It didn't fix even after I reinstall the phy.

hugosmie, 2024-01-09:
[image]

zm711, 2024-01-12:
Could you post how you installed phy?

zm711, 2024-01-16:
Yeah different packages update different so that can lead to the problems. I've actually had more success just installing with the yaml file. Could you download:

`environment.yml` then edit it to limit python to 3.11 (ie python=3.11, I have PR to fix the 3.12 issue, but it is not merged yet). Try that and let me know how that goes.

zm711, 2024-06-20:
Closing for lack of response. Feel free to reopen if you try the suggestion and it doesn't work.

Note that you will need to edit numpy < 2.0 since we have not tested on 2.0.

closed 2024-06-20 by zm711