cortex-lab/phy #1246: Error due to 'Segmentation Fault' on Linux

paulweger, 2024-01-11:
Hey, I keep running into the issue of "Segmentation Fault". I am working on Linux and tried reinstalling the environment/phy. The following error messages appear:

(phy2) paul@spikesorter:/storage3/paul/data/1950_2/KS$ phy template-gui params.py --debug
10:54:03.701 [D] __init__:68          Start capturing exceptions.
10:54:03.895 [W] model:603            Unreferenced clusters found in templates (generally not a problem)
10:54:04.034 [D] model:619            Loading spike clusters.
10:54:04.181 [W] model:625            Unreferenced clusters found in spike_clusters (generally not a problem)
10:54:04.334 [D] model:569            No channel shank file found.
10:54:04.335 [D] model:692            Loading templates.
10:54:04.415 [D] model:724            Templates are dense.
10:54:04.420 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
10:54:04.420 [D] model:730            Loading the whitening matrix.
10:54:04.420 [D] model:434            Whitening matrix file not found.
10:54:04.421 [D] model:737            Loading the inverse of the whitening matrix.
10:54:04.430 [D] model:766            Loading features.
10:54:04.430 [D] model:803            Loading template features.
10:54:04.432 [D] model:504            Load `cluster_Amplitude.tsv`.
10:54:04.479 [D] model:504            Load `cluster_ContamPct.tsv`.
10:54:04.483 [D] model:504            Load `cluster_group.tsv`.
10:54:04.487 [D] model:504            Load `cluster_KSLabel.tsv`.
10:54:05.032 [D] context:100          Initialize joblib cache dir at `/storage3/paul/data/1950_2/KS/.phy`.
10:54:05.032 [D] context:101          Reducing the size of the cache if needed.
10:54:05.044 [D] base:102             Add filter `high_pass`.
10:54:05.045 [D] config:31            Load config file `/home/paul/.phy/phy_config.py`.
10:54:05.047 [D] plugin:145           Loading 0 plugins.
10:54:05.049 [D] context:209          The file `/storage3/paul/data/1950_2/KS/.phy/new_cluster_id.pkl` doesn't exist.
10:54:05.447 [D] context:185          Save data to `/storage3/paul/data/1950_2/KS/.phy/spikes_per_cluster.pkl`.
10:54:05.815 [D] gui:463              Creating GUI.
10:54:05.833 [D] state:46             Load /home/paul/.phy/TemplateGUI/state.json for GUIState.
Segmentation fault

Incase that helps somehow: when using another environment (without the respective libaries), Phy is opening, but showing blank (white) ClusterView and SimilarityView; other panels are black except of TraceView. Even the test-dataset provided doesn't work in scenarios.

Thank you for your help,
Paul

paulweger, 2024-01-11:
As mentioned in #1231, I installed phy via environment.yml and ran into the following problem with the imp libary. Did you find a solution yet?

(phy2) paul@spikesorter:~/Downloads/phy-data-master/template$ phy template-gui params.py --debug
Traceback (most recent call last):
  File "/home/paul/miniconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/home/paul/miniconda3/envs/phy2/lib/python3.12/site-packages/phy/__init__.py", line 21, in <module>
    from .utils.config import load_master_config
  File "/home/paul/miniconda3/envs/phy2/lib/python3.12/site-packages/phy/utils/__init__.py", line 8, in <module>
    from .plugin import IPlugin, attach_plugins
  File "/home/paul/miniconda3/envs/phy2/lib/python3.12/site-packages/phy/utils/plugin.py", line 14, in <module>
    import imp
ModuleNotFoundError: No module named 'imp'

paulweger, 2024-01-11:
Found a workaround using the base version environment and the command in #1143. Please inform me about any other solution if possible.

fbartsch, 2024-01-11:
What exactly was your workaround? I've had the same segfault issue on Ubuntu 22.04.
I just tried to run `env QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui params.py`
as recommended in #1143 in the regular phy2 environment, but still got the same error. I likely missed something, any help is greatly appreciated.

paulweger, 2024-01-11:
@fbartsch try the preexisting base environment, that somehow worked for me

fbartsch, 2024-01-11:
@paulweger still not working for me, but thanks for confirming - I'll keep looking.

zm711, 2024-01-12:
You should install with the environment yaml instead. But when you download it you need to edit it and limit it to python=3.11. You should never install in base. It can break your conda install. @fbartsch , @paulweger.

I have a PR open to fix the imp issue, but I'm waiting for review and merge.

fbartsch, 2024-01-12:
@zm711 Ha, that (and a quick reboot after reinstalling using the yaml) did it! Thanks so much!