cortex-lab/phy #1247: temp_wh.dat does not exist

Wydneuro, 2024-01-11:
In my file path, temp_wh.dat exists, but when running phy2, show “temp_wh.dat does not exist”，I don't know why？
[image]

zm711, 2024-01-12:
When I click on your link the picture is not opening. Could you post the contents of your params.py file? It chooses where to look for the `temp_wh.dat` based on the path given in that file.

zm711, 2024-02-07:
Could you install with the `environment.yaml` instead. You need to download it then edit it to say python<3.12. That should fix the mime issue.

Moyu0014, 2024-02-07:
Hi @zm711 ,

Thanks for your reply, can I download it on the Anaconda directly because I don't have access to python?

What about the issue "tmp_wh.dat does not exist", the location "C:\Users\HP-lab\Desktop\SORTING" is not the location where my params.py files is (The D:\22-01-2024\SingleUnit is the location that I want). I wonder whether this may be the cause of this issue.

Thanks

zm711, 2024-02-07:
@Moyu0014,

Yes you can do this through Anaconda (you actually have python through Anaconda :) ).

So first if you have git or github desktop access you can clone the repo. If you don't then you can download a zip of the files. Inside the files you'll see that there is a file called `environment.yaml`. You just open that and find the line where it says python and change that line to be `python<3.12`. 

then you need to open your anaconda prompt and type
```
conda env create -f environment.yaml
````

The file might be .yml (depends on the tag you gave so if it gives a file error )

```
conda env create -f environment.yml
```

As far as the file location you could edit the params.py to point it to the correct location. Just open that file and you'll see the 
```python
dat_path=xxx
```
You would just need to edit that path.

Moyu0014, 2024-02-07:
Hi @zm711,

Thanks for your reply! I just ran the code and it displayed a conda value error below:
 
CondaValueError: prefix already exists: C:\Users\my name\anaconda3.

What shall I do next?

Best wishes,

Moyu0014

zm711, 2024-02-07:
@Moyu0014 

First you have to run

```
conda remove --name phy2 --all
```

To remove the broken install of phy2. You have to be in base to do this so if you are in phy2 you would type
```
conda deactivate
```
first. If you're already in base then just do the one command I posted.

Moyu0014, 2024-02-09:
Hi @zm711 ,

I changed to another PC & this issue no longer happens, but I have another issue when I run Kilosort 3. Matlab was giving me the error Unrecognized field name "dmin", but I am pretty sure dmin exists Rez.ops. I am new to matlab, so can you let me know how to solve this in detail, please? Thanks very much!
 
[image]

Kind Regards,

Moyu0014

zm711, 2024-02-09:
@Moyu0014

That's a Kilosort question and not a Phy question. Unfortunately the Kilosort team is busy working on Kilosort4, so I'm not sure they will be able to answer. I don't have any helpful tips except try running a different dataset and see if this is a global problem for you or an individual problem. I have been able to get Kilosort3 to work for myself, so I can tell you it works. But just not sure.

closed 2024-06-04 by zm711