cortex-lab/phy #1250: PyQt5 issue with starting GUI

maycaj, 2024-02-01:
I am having an issue starting the PHY GUI. I consistently get an error with PyQt5 even after a reinstallation of anaconda and phy. Any suggestions would be greatly appreciated!
```
(phy2) NEIK2A72CM22B:kilosorting_laminar maycaj$ phy template-gui params.py
Traceback (most recent call last):
  File "/Users/maycaj/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/qt.py", line 33, in <module>
    from PyQt5.QtCore import (Qt, QByteArray, QMetaObject, QObject,  # noqa
ImportError: dlopen(/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so, 0x0002): Library not loaded: @rpath/QtCore.framework/Versions/5/QtCore
  Referenced from: <5DB832BD-87BC-3750-B686-5A0A711D09B3> /Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so
  Reason: tried: '/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file)
(phy2) NEIK2A72CM22B:kilosorting_laminar maycaj$ QTWEBENGINE_CHROMIUM_FLAGS="--single-process" phy template-gui params.py
Traceback (most recent call last):
  File "/Users/maycaj/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/qt.py", line 33, in <module>
    from PyQt5.QtCore import (Qt, QByteArray, QMetaObject, QObject,  # noqa
ImportError: dlopen(/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so, 0x0002): Library not loaded: @rpath/QtCore.framework/Versions/5/QtCore
  Referenced from: <5DB832BD-87BC-3750-B686-5A0A711D09B3> /Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so
  Reason: tried: '/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file)
(phy2) NEIK2A72CM22B:kilosorting_laminar maycaj$ 

```
My terminal output that includes 1) removing and re-installing anaconda 2) attempt at opening PHY
[Terminal Saved Output.txt](https://github.com/cortex-lab/phy/files/14130136/Terminal.Saved.Output.txt)

zm711, 2024-02-07:
Could you install with the `environment.yaml`. Just download it then edit it so that it says `python<3.12`.

zm711, 2024-02-07:
Could you do `phy template-gui params.py --debug` just so I can see the full output?

maycaj, 2024-02-07:
Looks like we are getting the same output again, but not sure:

```
(phy2) NEIK2A72CM22B:kilosorting_laminar maycaj$ phy template-gui params.py --debug
Traceback (most recent call last):
  File "/Users/maycaj/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/__init__.py", line 6, in <module>
    from .qt import (
  File "/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/gui/qt.py", line 33, in <module>
    from PyQt5.QtCore import (Qt, QByteArray, QMetaObject, QObject,  # noqa
ImportError: dlopen(/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so, 0x0002): Library not loaded: @rpath/QtCore.framework/Versions/5/QtCore
  Referenced from: <5DB832BD-87BC-3750-B686-5A0A711D09B3> /Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/QtCore.abi3.so
  Reason: tried: '/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file), '/Users/maycaj/anaconda3/envs/phy2/bin/../lib/QtCore.framework/Versions/5/QtCore' (no such file)
(phy2) NEIK2A72CM22B:kilosorting_laminar maycaj$ 
```

zm711, 2024-02-07:
That is very worth mentioning thanks. 

Macs deprecated the use of openGL, which unfortunately what is required for Phy. So at this time I highly doubt you'd be able to get it to work on a Mac. Sorry!

closed 2024-06-04 by zm711