cortex-lab/phy #1253: Unable to open Phy and visualize data

HoykenM, 2024-02-06:
Dear all,

I'm new to phy and seem not to get Phy running and visualization open. 
I've been going through similar issues here and my problems seem very similar to other, I already tried most of the proposed solution but non seem to work ( like installing and reinstally phy2, installing vispy).

Basically, there are 2 problems. First I started a conda environment and tried to run the gui (phy template-gui params.py). However, Phy won't open and I'm getting the error message below:

(phy2) NAME@MacBook-Pro-van-Michael phy_folder % phy template-gui params.py
objc[49388]: Class QMacAutoReleasePoolTracker is implemented in both /Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore (0x1271b9230) and /Users/NAME/anaconda3/envs/phy2/lib/libQt5Core.5.15.2.dylib (0x134011340). One of the two will be used. Which one is undefined.
objc[49388]: Class QT_ROOT_LEVEL_POOL__THESE_OBJECTS_WILL_BE_RELEASED_WHEN_QAPP_GOES_OUT_OF_SCOPE is implemented in both /Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore (0x1271b92a8) and /Users/NAME/anaconda3/envs/phy2/lib/libQt5Core.5.15.2.dylib (0x1340113b8). One of the two will be used. Which one is undefined.
objc[49388]: Class KeyValueObserver is implemented in both /Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore (0x1271b92d0) and /Users/NAME/anaconda3/envs/phy2/lib/libQt5Core.5.15.2.dylib (0x1340113e0). One of the two will be used. Which one is undefined.
objc[49388]: Class RunLoopModeTracker is implemented in both /Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore (0x1271b9320) and /Users/NAME/anaconda3/envs/phy2/lib/libQt5Core.5.15.2.dylib (0x134011430). One of the two will be used. Which one is undefined.
Traceback (most recent call last):
  File "/Users/NAMEanaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/__init__.py", line 25, in <module>
    from .base import (  # noqa
  File "/Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/phy/apps/base.py", line 22, in <module>
    from phylib.io.array import Selector, _flatten
ImportError: cannot import name 'Selector' from 'phylib.io.array' (/Users/NAME/anaconda3/envs/phy2/lib/python3.11/site-packages/phylib/io/array.py)

**Other people seem to have similar importError problems. I already tried reinstalling phy2 and upgrading Phylib(python3 -m pip install --upgrade phylib).** 

**Next, I tried just running the gui in python (not through anaconda). Whilest doing this, the gui opens, howevever I cannot visualise the data (see attached file). I already installed vispy without effect.**
[image]

Thank in advance
Michael

zm711, 2024-02-07:
@HoykenM,

How'd you install? I would advise two things. First install with the `environment.yaml` into conda environment. Second, once you download the yaml file you need to limit python<3.12 since their is an issue with import imp in 3.12.

zm711, 2024-02-07:
And now I just looked at your terminal and it appears you're on a Mac. Unfortunately Mac has deprecated openGL, so I haven't been able to get Phy to work on M-series Macs since this point. So unless you have a very old Mac I'm worried that you won't have the capability to use all of Phy's functionality.

closed 2024-08-08 by zm711

MarcoUniPr, 2025-03-25:
Hi, I have the similar problem by running Phy on Windows Server 2019. Do you have any advices?