cortex-lab/phy #1256: Waveform view selecting multiple clusters

msj220001, 2024-02-16:
I'm having a similar issue as this previous one: WaveformView issue while comparing clusters #947
Except I am using phy2 gui and not kwik gui.

I used the spykingcircus2 sorter (developed off of Spyking Circus by SpikeInterface).

When I select a cluster, waveform view selects clusters from other channels as well.
[image]

When I select a similar cluster, the same thing happens.
[image]

Is there a command or key that can deselect other clusters or is this a bug?

zm711, 2024-03-17:
Sorry I'm not sure I understand. You selected cluster 56 which appears on channels 2,1,3,0. Which makes sense most clusters will be sorted onto multiple channels. SC2 is probably using some sort of sparsity but even then it would likely appear on multiple channels. Could you reexplain the issue :)

msj220001, 2024-03-19:
I see, so cluster 56 is defined as having waveforms across multiple channels?
What I would like is for cluster 56 to be waveforms from only channel 1 and not from any other channel.

This is a low-density probe so waveforms and units that appear on each channel are considered independent of each other, so I was trying to find a way I could make the algorithm only define units for a single channel without considering other channels.

zm711, 2024-03-19:
I don't think you would do that in phy. But you could do it outside of phy. Spikeinterface has nice machinery for doing that. Otherwise you would just have to make a memmap of your raw file, iterate through each spike_time in `spike_times.npy` and see which channel has the max amplitude and take the waveforms only from that channel. But phy will highlight all possible channels and show waveforms values even if there are no waveforms on the channel. 

since you are using SC2 you must be using SpikeInterface so I would just use their machinery. You could open an issue over there and see what they recommend for this. Unless you were just hoping to visualize the results.