cortex-lab/phy #1259: phy template-gui  /home/huxiaoyi/PycharmProjects/spikeinterface/si_env/phy_KS25/params.py 20:40:27.761 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed. 段错误

huxiaoyi520, 2024-03-09:
This error always appears when I run it.
But when I run "trace-gui" it runs successfully.
How to solve this? Thank you so much

huxiaoyi520, 2024-03-09:
我来说下这个的安装，这个教程的安装总是出错。这都是我的血泪教训，搞了好久好久。首先，git clone git@github.com:cortex-lab/phy.git  下载，然后cd phy-master进入到这个文件夹。然后将environment.yml里面的python后加上=3.11。最后直接conda env create -f environment.yml安装所有。然后conda activate phy2和env QTWEBENGINE_CHROMIUM_FLAGS="--single-process" 。最后再phy template-gui params.py。  然后GUI就出来了。

zm711, 2024-03-17:
@huxiaoyi520 sorry a bunch of your message appears to be in Chinese. Is there any way you could display it in English? In general skipping spike waveforms that do not exist is not a problem. You have a bunch of "cached" files with data, but if you feed phy the raw data (a `.bin` or `.dat`) it can extract things that it is missing from the cached files.

zm711, 2024-06-07:
I'm going to close this for lack of activity. But let us know if you have more issues.

closed 2024-06-07 by zm711